gokrazy/gokrazy #158: gokr-packer doesn't seem to accept local packages

eliasnaur, 2022-11-20:
## Observed behavior

```
% mkdir local-module
% cd local-module
% go mod init hello.com
go: creating new go.mod: module hello.com
a@macbog local-module % cat >> main.go
package main

import (
	"fmt"
	"log"
	"os"
)

func main() {
	log.Println("Hello, world!")
	fmt.Println("Hello, world")
	console, err := os.Open("/dev/console")
	if err != nil {
		log.Fatal(err)
	}
	defer console.Close()
	fmt.Fprintln(console, "Hello World, console")
}

% gokr-packer -overwrite=/dev/disk4 .
gokrazy packer v0.0.0-20221008090041-c87c806562c3 on GOARCH=arm64 GOOS=darwin

Updating gokrazy installation on http://gokrazy

Build target: CGO_ENABLED=0 GOARCH=arm64 GOOS=linux
Build timestamp: 2022-11-20T09:05:52-06:00
Loading system CA certificates from bundled Mozilla CA list
Building 1 Go packages:

  .

[building (go compiler)]2022/11/20 09:05:52 Migrated go.mod to builddir/github.com/gokrazy/kernel/go.mod, see https://gokrazy.org/development/modules/
go: finding module for package github.com/gokrazy/kernel
2022/11/20 09:05:52 Migrated go.mod to builddir/github.com/gokrazy/firmware/go.mod, see https://gokrazy.org/development/modules/
go: finding module for package github.com/gokrazy/firmware
2022/11/20 09:05:53 Migrated go.mod to builddir/github.com/gokrazy/rpi-eeprom/go.mod, see https://gokrazy.org/development/modules/
go: finding module for package github.com/gokrazy/rpi-eeprom
2022/11/20 09:05:53 Migrated go.mod to builddir/github.com/gokrazy/gokrazy/cmd/dhcp/go.mod, see https://gokrazy.org/development/modules/
go: finding module for package github.com/gokrazy/gokrazy/cmd/dhcp
2022/11/20 09:05:53 Migrated go.mod to builddir/github.com/gokrazy/gokrazy/cmd/ntp/go.mod, see https://gokrazy.org/development/modules/
go: finding module for package github.com/gokrazy/gokrazy/cmd/ntp
2022/11/20 09:05:54 Migrated go.mod to builddir/github.com/gokrazy/gokrazy/cmd/randomd/go.mod, see https://gokrazy.org/development/modules/
go: finding module for package github.com/gokrazy/gokrazy/cmd/randomd
2022/11/20 09:05:54 Migrated go.mod to builddir/go.mod, see https://gokrazy.org/development/modules/
2022/11/20 09:05:54 getting incomplete packages [.]
go: .: no package to get in current directory
[done] in 2.08s
2022/11/20 09:05:54 [go get .]: exit status 1
a@macbog local-module %
```

## Expected behavior

I had expected local modules and package paths to work.

I suppose this issue is more a question: how do I develop gokrazy programs without uploading to a repository?

stapelberg, 2022-11-20:
Use `hello.com` instead of `.` to refer to your package. Then, ensure you have a `replace` directive in the corresponding go.mod file (`builddir/hello.com/go.mod`).

We should probably document this better. I also want to add a `gok add` subcommand as part of issue #147, which will work with `.`

eliasnaur, 2022-11-20:
Great, the replace directive trick worked. Thank you; I'll also try out the new `gok` command, although it seems to require a gokrazy device on WiFi :)

closed 2022-11-20 by eliasnaur

stapelberg, 2022-11-20:
> EDIT: `gok` works with SD cards through the https://docs.google.com/document/d/1AUDmc8lygLR59O29Oc4ckv-Buk2R9zrvPDO9JbPOts8/edit proposal.

Yes. If you want to give that (work in progress) a shot, you’ll need the config-redesign branch of the `internal` and `tools` repository:

* https://github.com/gokrazy/internal/commits/config-redesign
* https://github.com/gokrazy/tools/tree/config-redesign