gokrazy/gokrazy #159: [Question] Is it possible to write to the HDMI console?

eliasnaur, 2022-11-20:
I tried

```
package main

import (
	"fmt"
	"log"
	"os"
)

func main() {
	log.Println("Hello, world!")
	fmt.Println("Hello, world")
	console, err := os.Open("/dev/console")
	if err != nil {
		log.Fatal(err)
	}
	defer console.Close()
	fmt.Fprintln(console, "Hello World, console")
}
```

but nothing appears on the HDMI console:

[image]

stapelberg, 2022-11-20:
Are you running `gokr-packer` with `-serial_console=disabled`? If no, that explains the behavior (output goes to serial).

Writing to `/dev/console` works in principle — I’m doing that in https://router7.org, too.

eliasnaur, 2022-11-20:
Thank you, `-serial_console=disabled` helped: now the device shows gokrazy messages, but no world hellos:

[image]

I also tried `github.com/gokrazy/hello`; no hellos either. Perhaps its an issue with the community kernel? The full command line was:

```
% GOARCH=arm gokr-packer -serial_console disabled -kernel_package=github.com/gokrazy-community/kernel-rpi-os-32/dist -overwrite /dev/disk6 github.com/gokrazy/hello
```

stapelberg, 2022-11-20:
> I also tried `github.com/gokrazy/hello`; no hellos either

That package prints to stdout, which goes to gokrazy’s ring buffer that’s exposed in the web interface. It doesn’t log to the console.

> console, err := os.Open("/dev/console")

I think `os.Open` is not sufficient, as it opens the file read-only. If you add error checking to your write calls, you’ll probably get an error message.

Here’s what I’m using: https://github.com/rtr7/router7/blob/b39b137e20a71f3afd52ee353b873111129706f0/internal/teelogger/teelogger.go#L48

eliasnaur, 2022-11-20:
> > I also tried `github.com/gokrazy/hello`; no hellos either
> 
> That package prints to stdout, which goes to gokrazy’s ring buffer that’s exposed in the web interface. It doesn’t log to the console.
>

Gotcha!
 
> > console, err := os.Open("/dev/console")
> 
> I think `os.Open` is not sufficient, as it opens the file read-only. If you add error checking to your write calls, you’ll probably get an error message.
> 

Duh! Opening `/dev/console` read-write worked! Thank you very much for your help, I think I have enough to start developing. Great project, btw.

stapelberg, 2022-11-20:
> Thank you very much for your help, I think I have enough to start developing. Great project, btw.

Thanks! Let us know what comes out of your development, or if you get stuck somewhere :)

closed 2022-11-20 by stapelberg