gokrazy/gokrazy #164: Support passing flags to `gok run`

markdrayton, 2023-01-26:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

`gok run` doesn't support passing flags to the binary it runs.

## Expected behavior

`cd ~/go/src/github.com/stapelberg/scan2drive/cmd/scan2drive && gok run -autocert_host_list=scan2drive.example.com` would run `scan2drive autocert_host_list=scan2drive.example.com` on the target instance.

I had a quick look at how to do this but adding it to the current diversion approach doesn't seem straightforward. I suspect support for starting a new service (as mentioned at https://github.com/gokrazy/gokrazy/issues/88#issuecomment-974845682) is needed. I could potentially take a look at this if you could provide some high-level tips on where to wire things up.

stapelberg, 2023-01-26:
Hey, thanks for having a look at this improvement :)

With the instance-centric config now having landed, this should be feasible: `gok run` should take the current flags from the instance config, append the flags specified on the command line (if any), and include the entire list of flags in the divert HTTP request.

Because sending a slice of strings via HTTP form parameters is awkward, it might be easiest to define a DivertRequest struct that is then decoded/encoded using JSON, and send that as the divert request body.

The divert handler itself would need to be modified to either include the flags on the `cmd` when calling `NewService` (in case of a program that has not been included in the gokrazy instance yet), or call `setCmd` (which doesn’t exist yet) when diverting an existing service.

In terms of syntax, it would not be `gok run -autocert_host_list=scan2drive.example.com`, but `gok run -- -autocert_host_list=scan2drive.example.com` so that the flags are not confused with flags for `gok run` itself.

Does that roughly make sense? Let me know if you have detail questions.

markdrayton, 2023-01-26:
This sounds pretty reasonable, thanks! I'll take a look when I get a moment.

> In terms of syntax, it would not be gok run -autocert_host_list=scan2drive.example.com, but gok run -- -autocert_host_list=scan2drive.example.com so that the flags are not confused with flags for gok run itself.

Ah, yes, I was doing this in testing but forgot to report it accurately when I opened the issue. I assume cobra will handle this but I'll check either way.

markdrayton, 2023-01-27:
(This is probably a broader question so once I work it out I'll update the wiki or website with the answers, unless you want to do it first.)

What is the recommendation for managing gokrazy's multiple modules in a development environment? The work in this issue touches several different modules (gokrazy, tools, and updater) and I haven't hit upon the right set of `replace` directives to ensure that all the pieces are built with my locally modified code. In particular, I have succeeded in building `gok` with local changes but I haven't managed to get it to build and update a disk image with the local changes (this is needed because `gok` will send a different type of request to `/divert`, so the receiving handler needs to know about it).

@stapelberg, can you describe how you'd set up you development environment for a change like this? Where do you keep your checked-out repos, how do you configure `replace`s, do you use Go workspaces, etc? If it helps I can describe the various combinations I tried but  it's likely easier to ask someone who already has it working first ;-)

stapelberg, 2023-01-28:
> In particular, I have succeeded in building `gok` with local changes but I haven't managed to get it to build and update a disk image with the local changes (this is needed because `gok` will send a different type of request to `/divert`, so the receiving handler needs to know about it).

I put `replace github.com/gokrazy/gokrazy => /home/michael/go/src/github.com/gokrazy/gokrazy` into my `~/gokrazy/hello/builddir/github.com/gokrazy/gokrazy/go.mod`. This is sufficient for the resulting disk image to contain local changes to the gokrazy init system itself. See also https://gokrazy.org/development/modules/ for more details.

I don’t currently use Go workspaces because they never seem useful for my workflow. But perhaps this is just a matter of preference…

markdrayton, 2023-01-28:
Hm, I'm not having much luck with this approach. I tried a minimal repro on another machine and saw the same issue (transcript at https://gist.github.com/markdrayton/a053da6075d120d82d41954ed53b283e). Do I need to `go mod tidy` or anything after updating the `go.mod` file?

stapelberg, 2023-01-29:
What you’re doing looks correct to me.

Can you open the web interface and click on /gokrazy/dhcp, then look at the module info at the bottom? If the replace directive is effective, you should see something like this:

```
go	go1.19.4
path	github.com/gokrazy/gokrazy/cmd/dhcp
mod	github.com/gokrazy/gokrazy	v0.0.0-20221113114523-dd415c9ee654
=>	/home/michael/go/src/github.com/gokrazy/gokrazy	(devel)
```

(I’m suspecting that your code change *was* effective but doesn’t manifest itself in a loud-enough failure to be visible.)

(You might need to delete `builddir/github.com/gokrazy/gokrazy/cmd` so that the go.mod from `builddir/github.com/gokrazy/gokrazy` (one level above) is picked up for the dhcp program, too.)

Let me know if that doesn’t help and I’ll try to reproduce from scratch on another machine.

stapelberg, 2023-01-29:
Oh, sorry, I think I figured out the problem! The builddir for the init program is `builddir/init`, despite the source living in `builddir/github.com/gokrazy/gokrazy`, so the replace directive needs to go into `builddir/init/go.mod`.

markdrayton, 2023-01-29:
Aha, yes, that seems to have done it! Thanks.

(It would neat if there was a way to dump the module info for init as well as the other userland binaries.)

stapelberg, 2023-01-29:
> (It would neat if there was a way to dump the module info for init as well as the other userland binaries.)

Agreed. I’m not entirely sure where it should go, but it would probably be good to re-think the current UI a little anyway — that the module info is hidden away below the log output isn’t great.

stapelberg, 2023-02-06:
This is all merged now. Thank you for your contribution!

closed 2023-02-06 by stapelberg