gokrazy/gokrazy #167: gokrazy Update Service (GUS)

stapelberg, 2023-02-05:
related to https://github.com/gokrazy/gokrazy/issues/25

@damdo and I have been working on a little project called gokrazy Update Service (GUS).

You can find the design document at https://docs.google.com/document/d/1q2TbB3DlO01_1EO7sQqFH4eMJE21XzniMs5hCqfqM8A/edit?usp=sharing

This issue tracks overall progress. Feel free to leave a comment or ask a question if you have any feedback!

* [x] [`gok sbom` subcommand](https://github.com/gokrazy/tools/commit/ca01203cadf49d3f7d773470bbd7340bf0f8852c)
* [ ] github.com/gokrazy/gokrazy/cmd/heartbeat program
  * [ ] add a public method to fetch machine-id (and fallback to hostname)
* [ ] github.com/gokrazy/gus/cmd/gus-server
  * [ ] SQLite database
  * [ ] Heartbeat API
  * [ ] UI 
  * [ ] Ingest API
  * [ ] GetUpdate API + AttemptUpdate API
  * [ ] [later: registry plugin as a reverse proxy]
* [x] GAF archive format writing
  * [ ] gok overwrite
* [ ] gok push
  * [ ] gok push to GUS local disk
  * [ ] gok push using a registry plugin
* [ ] gok gus
  * [ ] gok gus diff
  * [ ] gok gus set/unset
* [ ] github.com/gokrazy/selfupdate client
  * [ ] selfupdate using a registry plugin
* [ ] registry plugin(s)
  * [ ] github.com/gokrazy/gokplugins/oci (separate go.mod for each plugin) 
  * [ ] [later: github.com/gokrazy/gokplugins/s3]

jshield, 2023-02-09:
a potential enhancement to this would be being able to serve root/kernel/initrd/firmware over the network via nbd/nfs/styx/9p to do test boots for bakery validation, I'd imagine as part of early init the device could ask the update server if it should boot over the network.

stapelberg, 2023-02-09:
> a potential enhancement to this would be being able to serve root/kernel/initrd/firmware over the network via nbd/nfs/styx/9p to do test boots for bakery validation, I'd imagine as part of early init the device could ask the update server if it should boot over the network.

That is certainly possible, and I have considered netbooting in the past.

The big issue however is that I really want the “boot from SD card” workflow to work 100% of the time, and the only way to achieve that is to exercise exactly this flow in our CI setup. 

This might sound over-the-top, but I have seen behavioral differences on the Raspberry Pi depending on which boot mechanism was used: rebooting just hangs indefinitely when my Pi 4 is (was?) booted from a USB stick instead of an SD card. That’s a shame, because writing images to the USB stick is much faster, but if I get a different set of bugs, it doesn’t make sense to develop with that setup.

I have also diagnosed Linux kernel bugs that only happened when you run the Pi headless, without an HDMI monitor connected.

Hence, my current stance is that we should use a standard setup as much as possible, that is as close to the recommended setup (boot from SD card, with/without a monitor connected) as possible.

jshield, 2023-02-09:
No that makes complete sense to me, SD boot should be the guaranteed boot mode and thus tested out of the box.

I'd still see utility in it for devices that don't have much local storage as a primary boot mechanism, I was considering trying to do something with some of the devices that I have that are currently running OpenWRT today, I have an old Wireless Access Point that has a decent amount of RAM (128MB) but little flash (16MB), the smallest I could get the core services in gokrazy was about 9MB by additionally stripping the binaries as part of the build, but storing it on the flash is impractical, I was able to copy the squashed root to tmpfs, mount and pivot however.

On the otherhand I could also potentially just write a service that would configure it over SSH and UCI and deploy that onto one of the other gokrazy devices.

also as an aside given you have the interfaces.json definition for gokrazy/rtr7, have you seen http://netjson.org/ it appears to be an attempt to standardize something similar and have features for fleet management that might be a useful addition to this as well.

philhug, 2023-02-21:
would it be possible to make the update service more generic to allow it to also return config files as well (or is this already part of the sbom?) and maybe allow other commands like reboot, exec, ...
this would make it possible for user story 5 to completely disable the web interface on the gokrazy instance.

also I think the Api calls could be reduced to one, by just returning the command payload (update_firmware, update_config, reboot, .. in the heartbeat response and putting the attempt_update payload in the heartbeat message.

stapelberg, 2023-02-21:
> I'd still see utility in it for devices that don't have much local storage as a primary boot mechanism, I was considering trying to do something with some of the devices that I have that are [...]

Impressive! 

> On the otherhand I could also potentially just write a service that would configure it over SSH and UCI and deploy that onto one of the other gokrazy devices.

Yep, that seems like a good architecture.

> also as an aside given you have the interfaces.json definition for gokrazy/rtr7, have you seen http://netjson.org/ it appears to be an attempt to standardize something similar and have features for fleet management that might be a useful addition to this as well.

Thanks for the pointer!

stapelberg, 2023-02-21:
> would it be possible to make the update service more generic to allow it to also return config files as well (or is this already part of the sbom?) 

You can run `gok sbom` to see what’s part of the SBOM and what isn’t. For each file, the path and content hash goes into the SBOM, but not file contents. Conceptually, the SBOM is a bill of what went into the build, not instructions for how to build something.

I think you’re talking about the gokrazy [instance config](https://gokrazy.org/userguide/instance-config/). Distributing this file alone is not very useful — to build a gokrazy image, you currently need:

1. the instance config
1. the go.mod/go.sum files in `builddir`, and any locally modified working copies
1. any referenced extrafiles

You’re welcome to store all of these files in a private git repository and distribute that in a way that’s convenient for you.

I don’t think this is a useful addition to GUS in the general case, though.

> and maybe allow other commands like reboot, exec, ... this would make it possible for user story 5 to completely disable the web interface on the gokrazy instance.

You always need the web interface to do updates, even when said update is done through the selfupdate program.

If you want to interactively control your device, just [install tailscale](https://gokrazy.org/packages/tailscale/) to make the web interface available. This doesn’t need to go through GUS.

> also I think the Api calls could be reduced to one, by just returning the command payload (update_firmware, update_config, reboot, .. in the heartbeat response and putting the attempt_update payload in the heartbeat message.

While that’s technically true, I don’t think it would be good design. I don’t see a reason to deviate from small, targeted requests/responses (see https://github.com/gokrazy/gokapi for an up-to-date description and docs browser).

elgohr, 2023-11-26:
Thank you for the effort!
Maybe you could have a look at https://www.nerves-hub.org/ for inspiration - if you didn't already 🙂

mpl, 2024-12-19:
It seems you're already quite far along in the overall progress.
Is there a prototype we can try somewhere?
I'm investigating gokrazy as a solution for rpi5 fleet management (VS things like balena, mender, etc), and it's a cool project, but it seems without GUS there's too much indeed I'd have to slap on top of it myself.

mpl, 2025-01-07:
Hello,

following the design doc, I've been able to try out most of the steps: running GUS, generating a GAF, pushing it to GUS, trying gokrazy/selfupdate on the client side.
But now I'm stuck, since `gok gus set` does not exist yet, so I can't tell GUS to ingest the GAF I've pushed, and therefore the client gets no response when asking for an update.
Is there any code I've missed somewhere that could do that? Or should I try to manually POST to the ingestion endpoint of GUS? Would that work in theory? Is that the only step missing for the whole setup, so that a client can automatically pull updates?

mpl, 2025-01-09:
> Hey @mpl

Hello,

> Happy you are excited about this feature. Yes we are still working on it, and I'm still planning on finishing the missing bits. 

Out of curiosity, looking at the commits activity, I didn't see much happening in general for gokrazy or GUS itself in 2024 (and even in second half of 2023, but maybe I didn't look everywhere). And given that I'm evaluating for future production use (small/medium fleet of rpis 5) I'm wondering if there's still much impetus behind it?
So far gokrazy is the most exciting one I've seen on a personal level, and it's very fun to tinker with it, but I also need to be serious for that work project, and I'll have to balance its pros and cons VS alternatives like balenaOS (which I've already used and was not completely satisfied with).

> Yes you got it right, there isn't a way to set things yet via `gus set`, but you can manually write a POST to ingest it. I wrote a rudimental e2e test that does that exact thing to cover for the missing command, you can find it here: [gokrazy/selfupdate@0dfe1ef#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52R45](https://github.com/gokrazy/selfupdate/commit/0dfe1ef33c11b4b33d9b42ac02b666eeb723c2d8#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52R45) Let me know if it works for you.

Thanks, that was super useful! That first allowed me to realize that the heartbeat on the client needed to be configured (imho it's a bit weird that it's included by default, even though the user is not aware that it should be configured).
And then I was able to indeed perform the ingestion step that I was missing.
So my rpi5 with selfupdate eventually did automatically update itself from the GUS server. Very cool to see. :)

> If you are interested here is the PR with the missing functionality that I need to get back to: [gokrazy/tools#62](https://github.com/gokrazy/tools/pull/62)

stapelberg, 2025-01-09:
> Out of curiosity, looking at the commits activity, I didn't see much happening in general for gokrazy or GUS itself in 2024 (and even in second half of 2023, but maybe I didn't look everywhere). And [...]

To be clear, gokrazy is a hobby project. There is no company behind it, and no guaranteed support of any kind.

That said, I do see gokrazy as an active project. Some development might not be obvious from the outside, for example:

* https://michael.stapelberg.ch/posts/2024-02-11-minimal-linux-bootloader-debugging-story/ was a big investigation / bugfix.
* The approach outlined in https://michael.stapelberg.ch/posts/2024-11-19-testing-with-go-and-postgresql-ephemeral-dbs/ was developed for GUS (among other, similar projects).
* [Support for the Raspberry Pi 5](https://github.com/gokrazy/gokrazy/issues/223) was added by making new kernels available: [kernel.rpi](https://github.com/gokrazy/gokrazy/issues/264) as well as [generic kernels](https://github.com/gokrazy/gokrazy/issues/263) that are also useful for virtualized environments.

I have a number of Raspberry Pis and PC Engines machines deployed in my home and other locations, so I am highly motivated to fix serious issues.

mpl, 2025-01-09:
> To be clear, gokrazy is a hobby project. There is no company behind it, and no guaranteed support of any kind.

yes, of course.
That hasn't scared me in the past (especially with a Go open-source project), but I understand the disclaimer :)

> That said, I do see gokrazy as an active project. Some development might not be obvious from the outside, for example:
> 

yes, I know better than only looking at commit numbers, hence my question, thanks. ;)

> * https://michael.stapelberg.ch/posts/2024-02-11-minimal-linux-bootloader-debugging-story/ was a big investigation / bugfix.
> * The approach outlined in [...]

Interesting. So I gather it is a lot of work to support a new/different board each time?

Would you have an estimate of how much of a priority GUS is, in the grand of scheme of things, for the gokrazy project as a whole?
I can understand that for many people with a few devices, it is not important at all. In my biased opinion it is of course important (since I need it for work). But also, having had a quick look at the ecosystem, I haven't found many enticing alternatives to be honest, so I think that's a domain where GUS could shine and take the prize.

> I have a number of Raspberry Pis and PC Engines machines deployed in my home and other locations, so I am highly motivated to fix serious issues.

Good to know ;)

damdo, 2025-01-09:
I also actively run ~20 gokrazy instances scattered across various devices, including rpi1, rpi2, rpi4 x86 VMs and x86 bare metal.

I use it in my personal homelab and also as the core platform for the fleet used for a week long yearly festival with thousands of people. We rely on it for infrastructure services and a POS system.

@stapelberg and I designed the GUS system to cover this and many more use cases. So I am still personally interested in seeing this as an official opt-in service within the gokrazy ecosystem.

Unfortunately the time I could dedicate to it was very scarce in the last year. This year should be better on that regard. But to echo what @stapelberg said, it is a hobby also for me, so no guarantees :)

stapelberg, 2025-01-10:
> Interesting. So I gather it is a lot of work to support a new/different board each time?

That depends on the device. With the Pi 4, we needed to mainly figure out when upstream kernel support was good enough. For the Pi 5, the upstream Linux kernel does not have support (and it’s unclear when that will happen), so we had to provide a new kernel variant.

Providing a new kernel variant was some work in the beginning (to generalize the process), but is much easier now that we have a few of them.

> Would you have an estimate of how much of a priority GUS is, in the grand of scheme of things, for the gokrazy project as a whole?

No, I don’t do roadmap planning or similar for hobby projects. I can say that I still intend to make progress on GUS, and that there are no other competing (or more important) efforts within the gokrazy project.

ben-auo, 2026-10-16:
I'm interested as well in a gaf/GUS solution. Would you be open to PRs if I revisit it?