gokrazy/gokrazy #168: Add gokrazy.WaitForNetwork()

stapelberg, 2023-02-05:
Currently, we use gokrazy.WaitForClock() as a catch-all for clock and network.

This works well on the Raspberry Pi, which doesn’t have an RTC chip on board, and hence needs the network before the clock can be set.

But, in qemu (and other devices that have a functioning clock on every boot), this doesn’t work. 

We should add some sort of connectivity checking.

oliverpool, 2023-03-20:
My 2 cents: call it `gokrazy.WaitForClockAndNetwork`, so that it can be used for https requests (if the clock is wrong, the certificate verification should fail).

vilhalmer, 2024-12-07:
Also having one which is separate from waiting for NTP would be ideal for running a DNS server like blocky on gokrazy, since NTP servers/pools are often addressed by DNS name and this creates a bootstrapping loop if the whole network has been down after a power outage or similar.

stapelberg, 2025-02-22:
I think it would be best to allow waiting for a combination of different things. 

Commit https://github.com/gokrazy/gokrazy/commit/d4899e240e7c0a79680316de90719873634356ea implements such a function and documents which events I think are broadly useful. Let me know if you think there are others that should be added.

`WaitFor("net-route")` and `WaitFor("net-online")` are not currently implemented. If anyone is eager to take a stab at an implementation and wants to send a PR, let me know :)

stapelberg, 2025-02-23:
Update: I implemented `WaitFor("net-route")` and successfully tested it in a VM:

```
2025/02/23 10:32:13 running gokrazy.WaitFor('net-route')
2025/02/23 10:32:13 waiting for net-route
2025/02/23 10:32:13 waiting for net-route: RouteGet(193.0.14.129): network is unreachable
2025/02/23 10:32:14 waiting for net-route: RouteGet(193.0.14.129): network is unreachable
2025/02/23 10:32:15 done waiting after 2.001390114s
```

Next up is WaitFor("net-online")

stapelberg, 2025-02-23:
All done with commit https://github.com/gokrazy/gokrazy/commit/267d2ac6403c0717026b685ce604489624d19471

closed 2025-02-23 by stapelberg

damdo, 2025-02-23:
Awesome! Thank you!