gokrazy/gokrazy #170: gok tool creates GPT instead of MBR

n8225, 2023-02-05:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64
* [x] custom WIP image lepotato, tested with odroid xu4

## Observed behavior
Writing an SD card image for the odroid HC1 target or a custom target with a deviceconfig including "MBROnlyWithoutGPT: true," results in a GPT filesystem. Despite console log output of gok overwrite --full /dev/sda stating:

> Feature summary:
  use GPT: false
  use PARTUUID: true
  use GPT PARTUUID: false
2023/02/05 13:18:00 partitioning /dev/sda (no GPT, only MBR)
2023/02/05 13:18:00 Using sudo to gain permission to format /dev/sda

## Expected behavior
An SD card that contains a bootable MBR filesystem.

stapelberg, 2023-02-05:
Thanks for the report.

Can you include a minimal config.json to trigger the problem?

Does the problem also reproduce if you have write permission to the device node (`/dev/sda`) so that gok does not need to use `sudo`?

n8225, 2023-02-05:
> Thanks for the report.
> 
> Can you include a minimal config.json to trigger the problem?
```json
config.json

{
    "Hostname": "hello",
    "DeviceType": "odroidhc1",
    "Update": {
        "HTTPPassword": "somepassword"
    },
    "Packages": [
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/hello",
        "github.com/gokrazy/serial-busybox"
    ],
    "KernelPackage": "github.com/anupcshan/gokrazy-odroidxu4-kernel",
    "EEPROMPackage": "",
    "PackageConfig": {
        "github.com/gokrazy/gokrazy/cmd/randomd": {
            "ExtraFileContents": {
                "/etc/machine-id": "randommachineid\n"
            }
        }
    },
    "SerialConsole": "disabled"
}
```

> Does the problem also reproduce if you have write permission to the device node (`/dev/sda`) so that gok does not need to use `sudo`?

Running as root does not reproduce and does result in an MBR formatted SD card.

stapelberg, 2023-02-06:
Thanks, I could reproduce the problem. This should be fixed with commit https://github.com/gokrazy/tools/commit/05762f796b7f3132e4b48e55d15ae0331e63ddfd now

closed 2023-02-06 by stapelberg