gokrazy/gokrazy #176: Proper way to poweroff

oliverpool, 2023-03-01:
## Platform

* [x] Raspberry Pi Model B Rev 2 (https://github.com/gokrazy-community/kernel-rpi-os-32)

## Observed behavior

There is no way to shut down the raspberry pi.
I only found a way to reboot, by POSTing to `/reboot`.

## Expected behavior

I would like to shutdown my raspberry pi properly (to switch the SD card for instance) by issuing a dedicated request (POST `poweroff` or `halt` or `shutdown`?)

Even better would be to have a button on the https interface for both operations (rebooting, shutting down).

## Context

I had to replace my SD (with was defective) and I want to preserve the new SD card as much as possible :)

I would be ready to draft a PR (I guess [LINUX_REBOOT_CMD_POWER_OFF](https://man7.org/linux/man-pages/man2/reboot.2.html) is the right syscall)

stapelberg, 2023-03-02:
Hey @oliverpool 

Thanks for filing this. There already is a similar issue to do a shutdown when pressing the power button: https://github.com/gokrazy/gokrazy/issues/142

Maybe both can be fixed at once.

The overall stages of a shutdown should probably be:
1. Killing processes nicely
2. Terminating remaining processes that have not shut down after a certain time
3. Sync data to disk
4. Power off

Thanks in advance for sending a PR :)

stapelberg, 2023-03-05:
It turns out I now have an urgent need for proper shutdown, both on updates and when pressing the power button, because I’m now running a VM on my router that I would like to properly shut down whenever possible.

Have you started work on this yet? Otherwise I might just take a stab at this myself in the next few days.

Thanks

oliverpool, 2023-03-05:
I have a working poc for a `/poweroff` endpoint. Will open a PR later on.

PR #177 (2023-03-07): implement Poweroff (and refactor around killSupervisedServices)
Fixes #176 

I took the liberty of refacoring `killSupervisedServices` on the way:
- it sends SIGTERM to all running processes
- waits up to 15 seconds (`=signalDelay`)
- sends SIGKILL to the remaining processes
- waits up to 15 seconds (`=signalDelay`)
- returns

So that we get a higher chance of all processes having been terminated before unmount `/perm`.

(I initially replaced `syscall.Kill` with `process.Signal` and proper error checking, however rebooting always took 30s. After I while, I figured that `process.Signal` does no [... cut]
closed 2023-03-07 by stapelberg