gokrazy/gokrazy #183: Broken version of gokrazy/firmware prevents boot on certain raspberry pi

damdo, 2023-03-13:
Just an FYI https://github.com/gokrazy/firmware/pull/300 pulled in a problematic firmware for certain raspberry pis (seen on pi 3B+, pi 1B and 1B+), which prevents their boot.

The upstream version believed to have issues is: https://github.com/raspberrypi/firmware/commit/379d5bfa60bf3b0afff413a18344aaaf3bde0083
Upstream issue tracking this is: https://github.com/raspberrypi/firmware/issues/1791

stapelberg, 2023-03-13:
Thanks for the heads-up.

This seems to only manifest itself when running a 32-bit kernel, right? That’s what I can gather from the linked issue, and it would explain why the kernel made it through our CI.

From the issue, there should be a new version tomorrow, so I think I’ll just wait for automation to pull that in.

damdo, 2023-03-16:
I think we can close this now that the fix has been brought in by https://github.com/gokrazy/firmware/pull/301
Thanks.

closed 2023-03-16 by damdo