gokrazy/gokrazy #186: suggestion: add a cmd/dmesg program

oliverpool, 2023-03-20:
I have created a small package which reads the kernel messages from `/dev/kmsg`.

Do you think it might be useful to add it here? Or maybe in some `https://github.com/gokrazy-community/` repo?

```go
package main

import (
	"bufio"
	"fmt"
	"log"
	"os"
	"strconv"
	"strings"
	"time"
)

func main() {
	if err := run(); err != nil {
		log.Fatal(err)
	}
}

// https://www.kernel.org/doc/Documentation/ABI/testing/dev-kmsg
func run() error {
	f, err := os.Open("/dev/kmsg")
	if err != nil {
		return err
	}
	defer f.Close()

	scanner := bufio.NewScanner(f)
	for scanner.Scan() {
		line := scanner.Text()
		if len(line) == 0 {
			continue
		}
		if line[0] == ' ' {
			// discard continuation line
			continue
		}
		metadata, msg, _ := strings.Cut(line, ";")

		d := extractTimestamp(metadata)
		fmt.Printf("[%8s] %s\n", roundTime(d), msg)
	}
	return scanner.Err()
}

func extractTimestamp(s string) time.Duration {
	parts := strings.Split(s, ",")
	if len(parts) < 3 {
		return 0
	}
	d, _ := strconv.ParseInt(parts[2], 10, 64)
	return time.Duration(d) * time.Microsecond
}

func roundTime(d time.Duration) time.Duration {
	if d > time.Minute {
		return d.Round(time.Second)
	}
	if d > time.Second {
		return d.Round(time.Second / 100)
	}
	if d > time.Millisecond {
		return d.Round(time.Millisecond / 100)
	}
	if d > time.Microsecond {
		return d.Round(time.Microsecond / 100)
	}
	return d
}
```

Sample output (note the format of the duration info):
```
[      0s] ftrace: allocated 99 pages with 4 groups
[      0s] trace event string verifier disabled
[      0s] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[     6µs] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    78µs] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[   173µs] bcm2835: system timer (irq = 27)
[   975µs] Console: colour dummy device 80x30
[  1.56ms] printk: console [tty1] enabled
[  1.64ms] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[ 60.32ms] pid_max: default: 32768 minimum: 301
```

stapelberg, 2023-03-21:
I don’t think this is worth putting into the main gokrazy repository, as it would be used so rarely that it doesn’t justify the space consumption of an extra binary.

Why not just run dmesg when logged into breakglass with busybox? 

We could probably put this into gokrazy-community/dmesg

oliverpool, 2023-03-24:
I don't have a good name for now. Maybe if a good idea comes up I might open a new gokrazy-community repo :)

closed 2023-03-24 by oliverpool