gokrazy/gokrazy #188: Document state of Raspberry Pi 400

bradfitz, 2023-04-01:
## Platform

I’m using:

* [x] gokrazy/rpi3b
* [x] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [x] .... and also gokrazy/rpi400

## Observed behavior

The Raspberry Pi 400 seems to mostly work. Everything except wifi.

I moved one of my SD cards from a Raspberry Pi 3B+ to a RPi 400 and it booted fine, but only got DHCP on wired ethernet, not wifi. Then moved it to another RPi 400 (because I'd previously taken apart the first 400 and broke parts of its keyboard), to confirm it wasn't a hardware problem, and same on the second RPi 400: ethernet works, but no wifi.

## Expected behavior

Documentation on https://gokrazy.org/platforms/ about the state of Raspberry Pi 400 (ethernet works, no wifi)

Extra bonus: fix wifi on 400 too:)

I'm willing to dedicate one of my three RPi 400s as a CI machine if needed.

bradfitz, 2023-04-01:
I guess the 4B and 400 are nearly identical, except when it comes to the wifi.

The 4B has BCM43455: "2.4 GHz and 5.0 GHz IEEE 802.11ac"
The 400 has BCM43456 (I think?): "2.4 GHz and 5.0 GHz IEEE 802.11b/g/n/ac" 

When using wifi on the rpi400 with Debian, I see:

```
[    5.316574] brcmfmac: F1 signature read @0x18000000=0x15294345
...
[    5.341573] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
...
[    5.569214] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    5.569365] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    5.582772] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: May 14 2020 17:26:08 version 7.84.17.1 (r871554) FWID 01-3d9e1d87
```

stapelberg, 2023-04-02:
> When using wifi on the rpi400 with Debian, I see:

Just to note: Raspbian is not the same as Debian, in particular they have different WiFi drivers.

I don’t have a Pi 400 to test, so you’ll need to try out the WiFi firmware files yourself. But yeah, maybe getting the right files in place with the right name is all you need :)

bradfitz, 2023-04-02:
Well, my naive attempt (https://github.com/bradfitz/gokrazy-wifi/commit/46162e718da9e5697a3280f0a675ea732af98b4a) didn't work.

Maybe this means something to you:

lsmod says:

```
brcmfmac_wcc 16384 0 - Live 0xffffa2a99fc4c000
brcmfmac 262144 1 brcmfmac_wcc, Live 0xffffa2a99fbfa000
brcmutil 20480 1 brcmfmac, Live 0xffffa2a99fbf4000
```

And `dmesg` says:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.2.9 (gokrazy@docker) (aarch64-linux-gnu-gcc (Debian 8.3.0-2) 8.3.0, GNU ld (GNU Binutils for Debian) 2.31.1) #1 SMP PREEMPT Wed Mar  1 20:57:29 UTC 2017
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 400 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xfb817a00-0xfb819fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 21 pages/cpu s46184 r8192 d31640 u86016
[    0.000000] pcpu-alloc: s46184 r8192 d31640 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Policy zone: DMA32
[    0.000000] Kernel command line: video=HDMI-A-1:1600x900M@60 dma.dmachans=0x37f5 bcm2709.boardrev=0xc03130 bcm2709.serial=0xc19a486c bcm2709.uart_clock=48000000 smsc95xx.macaddr=DC:A6:32:CD:86:A7 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=60c24cc1-f3f9-427a-8199-c55898220001/PARTNROFF=1 init=/gokrazy/init rootwait panic=10 oops=panic
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000033400000-0x0000000037400000] (64MB)
[    0.000000] Memory: 3791340K/4050944K available (20416K kernel code, 4506K rwdata, 12460K rodata, 9408K init, 630K bss, 194068K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[... 325 lines cut ...]
Codename:	buster

pi@pi400-brokenkeys:~ $ cat /etc/apt/sources.list.d/raspi.list 
deb http://archive.raspberrypi.org/debian/ buster main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ buster main

pi@pi400-brokenkeys:~ $ dmesg | grep brcm
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 10)
[    1.282835] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.285023] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.287257] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.289412] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.356153] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.358588] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    5.241661] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    5.245689] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
[... cut ...]
[    5.421970] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    5.424228] usbcore: registered new interface driver brcmfmac
[    5.643637] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    5.643804] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    5.655761] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: May 14 2020 17:26:08 version 7.84.17.1 (r871554) FWID 01-3d9e1d87
[    8.139567] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

jayaddison, 2023-05-08:
@bradfitz I've been attempting to identify the parts of the plain Debian Installer that don't quite work with the RPi 400, and the `brcmfmac43456` question in particular has landed me here.

To attempt to answer your question:

> How's it thinking this is `raspberrypi,4-model-b`?

I'm learning this stuff, but I think it'd be worth figuring out the 'flattened device tree' (.dtb) file that's being loaded by `gokrazy` at boot to determine this.

Roughly speaking: Devicetree provides an alternative to ACPI to describe the names and configuration properties of hardware, and the kernel frequently uses those properties when building firmware filepaths to request.

In the case of the `brcmfmac` driver, and the `bcm2711` chip that exists on both the RPi4 and RPi400, the kernel module uses the 'compatible' field from Devicetree where available:

- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/bcm2711-rpi-4-b.dts?h=v6.2#n10
- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/bcm2711-rpi-400.dts?h=v6.2#n6

Those `dts` source files are compiled into `.dtb` files provided at boot-time, so my best-guess would be that for some reason you are booting with the `rpi-4` `.dtb` instead of the `rpi-400`.

As for the `brcmfmac43456` question (the reason I arrived here, because I'm trying to find out more about it) - the ideal would be for the firmware files to arrive in the upstream `linux-firmware.git`.  That would make it easier for many different distributions to package and deliver working wifi/bluetooth for relevant systems.  Synaptics seem to be the owners of the intellectual property and licensing for that device, so I think it's a question of their open source org structure and awareness of `linux-firmware.git`.

As of recently there _is_ a license and firmware available, as [distributed](https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/) (license in the `debian/copyright` file in `firmware-nonfree_20221012-1~bpo11+1+rpt1.debian.tar.xz`, firmware in the `firmware-brcm80211_20221012-1~bpo11+1+rpt1_all.deb` package) by RPi Foundation.

Separately, I'm also trying to figure out whether the BCM43455 and BCM43456 are in fact the same device with different regulatory/calibration parameters.. but I'm not particularly sure how to go about doing that.

jayaddison, 2023-05-09:
Thank you, @stapelberg - I'll follow up on a separate issue thread there (it may take me a while to get around to that and to write the question, but will do).

bradfitz, 2024-06-02:
Perhaps this is now addressed via #264 (the Pi-specific kernels)? I'll report back when I've tested.