gokrazy/gokrazy #191: Feature request: GOARCH in config.json

bradfitz, 2023-04-03:
I know the docs advocate using Makefiles, but I'd like to be able to set GOARCH in `config.json` rather than an environment variable.

I just forgot it and bricked an x86 appliance with an arm64 userspace.

Or maybe can we at least have some validation checks in the remote updating case that the image being written matches the existing image's GOARCH?

stapelberg, 2023-04-03:
Yeah, I thought about putting GOARCH into config.json as a special case (perhaps as part of a GoEnv struct or something), also so that you could clone new gokrazy instances easily (think `gok new`, but seeded from a GitHub repo).

Would you want to send a PR? Otherwise I can probably get around to it some time…

bradfitz, 2023-04-03:
I can probably handle it. I'll ping this bug if/when I start on it. (I'm on paternity leave and my free time is random at best.)

But rather than more config to learn & document, maybe we should just auto-infer it from the `KernelPackage`? If can tell from the `vmlinuz` there, no? ...

```
$ file gokrazy-kernel/vmlinuz 
gokrazy-kernel/vmlinuz: Linux kernel ARM64 boot executable Image, little-endian, 4K pages

$ file rtr7-kernel/vmlinuz 
rtr7-kernel/vmlinuz: Linux kernel x86 boot executable bzImage, version 6.2.9 (gokrazy@docker) #1 SMP PREEMPT_DYNAMIC Wed Mar  1 20:57:29 UTC 2017, RO-rootFS, swap_dev 0XD, Normal VGA

bradfitz, 2023-04-04:
I sent https://github.com/gokrazy/tools/pull/54

stapelberg, 2023-04-06:
> But rather than more config to learn & document, maybe we should just auto-infer it from the `KernelPackage`? If can tell from the `vmlinuz` there, no? ...

Hmm, yeah, I think that could work. At the very least, we can try inferring it and revisit later, should this approach turn out to not be sufficient/ideal for some reason.

Thanks :)

liwde, 2024-09-09:
> I thought about putting GOARCH into config.json as a special case (perhaps as part of a GoEnv struct or something)

It would also be great to have this as a generic solution to also support setting GOARM via the config.json (required for Raspberry Pi 1, see https://github.com/gokrazy/gokrazy/issues/239). Not sure if that can be determinded as easily as for the GOARCH...

stapelberg, 2025-01-26:
Sorry for sitting on this for so long. Having to specify `GOARCH=amd64` when using `gok run` has finally annoyed me enough to add an `Environment` setting to config.json: https://gokrazy.org/userguide/instance-config/#environment

(This seems straight-forward enough compared to auto-detection based on the kernel.)

closed 2025-01-26 by stapelberg