gokrazy/gokrazy #197: Unable to update executable because root filesystem is read-only

brutella, 2023-04-12:
When using [go-github-selfupdate](https://github.com/rhysd/go-github-selfupdate) to update my app by replacing the old executable with a new one, I get an error that the root filesystem is read only. I know that this is expected behaviour because SquashFS is readonly. But do you see any way to get a workaround until the update mechanism in gokrazy is fully implemented?

christian-heusel, 2023-04-12:
You could store your application on the `/perm` partition and replace it there and reexec 🤔 But I guess other people here can give better tips 😄

brutella, 2023-04-12:
> It might not be well-documented, but it should be possible.

I couldn't find enough documentation on the website and #167 still has some tasks unfinished, so I assumed that it's not done yet.

> Yeah, having a small Go program that only runs a program on /perm (which you then manage yourself, be it with go-github-selfupdate, or something else entirely) sounds like a good workaround.

How would I install on /perm?

christian-heusel, 2023-04-12:
On 23/04/12 11:08AM, Matthias wrote:
> How would I install on /perm?

You first have to create the perm partition, the easiest way imo is via
[`gokrazy/mkfs`](https://github.com/gokrazy/mkfs), its described
[here in the userguide](https://gokrazy.org/userguide/permanent-data/)
how one can use it to setup the `/perm` partition.

Then you can use something like
[`gokrazy/breakglass`](https://github.com/gokrazy/breakglass) to ship
your binary.

Maybe Michael has some more tips here! :)

stapelberg, 2023-04-14:
> I couldn't find enough documentation on the website and #167 still has some tasks unfinished, so I assumed it's not done yet.

Maybe @damdo has some time to put together a rough first draft for documentation on how to set up GUS? :)

brutella, 2023-04-14:
That would be great. 

In the meantime, do you have any ideas on how to configure a gokrazy instance, which installs my executable on the perm partition and executes it? I guess I'll need a helper program, which is installed on the root partition, which then executes my executable from the perm partition, right?

stapelberg, 2023-04-14:
Yes. @christian-heusel already outlined the steps for working with the perm partition in https://github.com/gokrazy/gokrazy/issues/197#issuecomment-1505849325

brutella, 2023-04-17:
I'm playing around with the `ExtraFilePaths` option and found that you can only copy files to the root file system. Is there a specific reason for that? Can I specify a path to a different file system, for example `/perm`?

stapelberg, 2023-04-17:
The reason is that gokrazy only manages the boot and root file system; the /perm file system is under your control and ignored entirely by gokrazy.

I’m not sure if /perm should be managed by gokrazy and to what extent. So far, we’ve always considered it out of scope.

brutella, 2023-04-18:
Thanks for clarifying that. In my case I would use it to copy an executable onto the /perm file system using the `ExtraFilePaths` option.

I was confused that it didn't work because I didn't get an error when executing `gok overwrite`. Other paths did work though.

stapelberg, 2023-04-27:
> In my case I would use it to copy an executable onto the /perm file system using the `ExtraFilePaths` option.

Yeah, you currently need to do that in your own Go code (just overwrite the copy on /perm on every start).

> I was confused that it didn't work because I didn't get an error when executing `gok overwrite`. Other paths did work though.

That’s fair. We should add an error message for this case.

brutella, 2023-04-27:
> Yeah, you currently need to do that in your own Go code (just overwrite the copy on /perm on every start).

I don't think that would work because the copy on /perm might get updated using [go-github-selfupdate](https://github.com/rhysd/go-github-selfupdate).

> That’s fair. We should add an error message for this case.

An error message would be great. 👍

How much effort would it take to support other file systems in `ExtraFilePaths`?
Any ideas how I would implement that myself?

brutella, 2023-05-10:
I've resolved the issue by adding support for re-running my executable at a different location.

If someone is curious how I did it: I've added a command line flag, which allows me to specify a location on the file system to which the app should copy itself – ex. `--copy=/perm/bin/<my-executable>`. Then it re-runs itself at the specified location using the [exec.Command](https://pkg.go.dev/os/exec#Command) API.

Works great so far. 👍