gokrazy/gokrazy #203: Importing existing configuration

BrunoTeixeira1996, 2023-07-10:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [X] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

This is more a question than a issue but I think it might be cool to update the docs with this information.
I am using gokrazy on a pi4b and that is working great but lets say I switch computers or lose the information about my pi config.
What should I do to get back the config and related files/folders?
The only option is to backup the instance folder to an external drive?

Thanks

stapelberg, 2023-07-10:
You can use `gok sbom` to print all the files and directories that go into your instance’s build. The `config.json` is certainly most important, but as soon as you make local modifications to the software you’re installing, those locally modified program source directories become important, too.

In general, I would recommend having a daily backup for your entire machine (or at least home directory). But if you want to be more targeted, `gok sbom` prints all the files you need.

Another aspect to consider: the `/perm` partition on the Raspberry Pi. You can mirror its contents from the Pi to your PC using `rsync`: https://gokrazy.org/userguide/rsync-backups/

Does that make sense? I think it’s a good idea to add it to the documentation somewhere, but I’m not yet sure where the best place is.

BrunoTeixeira1996, 2023-07-10:
It does makes sense!
I think the best approach here is what you said about using some type of backup.
For my case I think the best thing to do is to backup my gokrazy instance (that resides on my laptop) and the`/perm` partition on the Pi so then I can use the gokrazy instance folder in another pc and then copy the `/perm` to the pi if needed!

I