gokrazy/gokrazy #204: tailscale: Setup advertise-routes does not work in config.json

BrunoTeixeira1996, 2023-07-11:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [X] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

When using `advertise-routes` from tailscale in the `config.json` I got a syntax error in gokrazy stderr web interface.

```json
	"tailscale.com/cmd/tailscale": {
            "CommandLineFlags": [
                "up --advertise-routes=192.168.30.0/24"
            ]
        }
```

Using the command line directly in gokrazy breakglass I am able to use this command so I think this might be a issue when extracting the flags because in this case we are using `--`

## Expected behavior

Setup advertise-routes in the configuration file to be able to advertise routes and make gokrazy act like a subnet router

## Note

- I could be doing this wrong and maybe the issue here is me, anyway I think it would be good to have another option on the docs about this use case

BrunoTeixeira1996, 2023-07-11:
Thanks @stapelberg  that worked like a charm.
Do you mind if I create that use case on the docs?

stapelberg, 2023-07-11:
There already is an example for how to format multiple command-line flags in the docs: https://gokrazy.org/packages/tailscale/#step-2-set-command-line-flags

The config file docs also contain an example already: https://gokrazy.org/userguide/instance-config/#packagecommandlineflags