gokrazy/gokrazy #205: /gokrazy/ntp: permission denied

Syphdias, 2023-07-13:
## Platform
I’m using: gokrazy/rpi4b

## Observed behavior
Apparently ntp does not work. And is stuck in restarting.
```
1970/01/01 01:00:10 gokrazy: attempt 9, starting ["/gokrazy/ntp"]
1970/01/01 01:00:10 privdrop.go:85: fork/exec /gokrazy/ntp: permission denied
1970/01/01 01:00:10 gokrazy: exit status 1
1970/01/01 01:00:11 gokrazy: attempt 10, starting ["/gokrazy/ntp"]
1970/01/01 01:00:11 privdrop.go:85: fork/exec /gokrazy/ntp: permission denied
1970/01/01 01:00:11 gokrazy: exit status 1
1970/01/01 01:00:12 gokrazy: attempt 11, starting ["/gokrazy/ntp"]
1970/01/01 01:00:12 privdrop.go:85: fork/exec /gokrazy/ntp: permission denied
1970/01/01 01:00:12 gokrazy: exit status 1
```

## Expected behavior
ntp to work

## What I tried to fix it
I tried to start over:
```
sudo wipefs -a /dev/sde
sudo dd if=/dev/zero of=/dev/sde bs=4M status=progress
sync
# replug SD card
gok -i gokrazy overwrite --full /dev/sde
```

christian-heusel, 2023-07-13:
So I also started from a clean SD card but cannot reproduce the issue 🤔

Syphdias, 2023-07-13:
I'm pretty sure there is something I am doing wrong. Or maybe it is a hardware issue. Both RPi Model B 4GB and SD card (SanDisk Extreme microSDXC UHS-I memory card 128 GB) are new. Maybe I bought the wrong SD format?
I did not consider it being a network issue. I don't think I block anything outgoing.
Everything(\*) else seems to work so far.

I might be able to try another SD card but I  don't have a spare RPi laying around (took long enough to get this one).

In case I did something wrong in the config:
```
❯ cat ~/gokrazy/gokrazy/config.json
{
    "Hostname": "gokrazy",
    "Update": {
        "HTTPPassword": "XXXXXXXXXXXXXXX"
    },
    "Packages": [
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/hello",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/iptables",
        "github.com/gokrazy/nsenter",
        "github.com/gokrazy/podman",
        "github.com/greenpau/cni-plugins/cmd/cni-nftables-portmap",
        "github.com/greenpau/cni-plugins/cmd/cni-nftables-firewall",
        "github.com/gokrazy/wifi",
        "github.com/gokrazy/mkfs"
    ],
    "PackageConfig": {
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "/home/syphdias/gokrazy/gokrazy/breakglass.authorized_keys"
            }
        },
        "github.com/gokrazy/gokrazy/cmd/randomd": {
            "ExtraFileContents": {
                "/etc/machine-id": "1bfe4ba3d20668c4608cd4585f9bf6e1\n"
            }
        },
        "github.com/gokrazy/wifi": {
            "ExtraFilePaths": {
                "/etc/wifi.json": "wifi.json"
            }
        }
    },
    "SerialConsole": "disabled",
    "InternalCompatibilityFlags": {}
}
```

(\*) Since the time is off, this prevents stuff like pulling docker images or anything that requires certificates I assume. fbstatus was restarting until I connected a monitor (will probably remove this package together with hello). And iptables which I added in preparation for podman. I also tried without iptables, nsenter, podman, and cni packages. No change.

Syphdias, 2023-07-13:
Unfortunately, I don't have a spare MicroSD card, only a big one :/

stapelberg, 2023-07-13:
This is really strange. I don’t think it’s a hardware or SD card issue. And it shouldn’t be possible to configure things wrong in this regard — NTP is expected to always work.

Are you by any chance on an extremely old kernel version for some reason? Can you share a screenshot of the web interface?

What does `gok sbom` print for your instance? 

What’s the output of `gok overwrite --root /tmp/root.squashfs` and `unsquashfs -ll /tmp/root.squashfs`?

Syphdias, 2023-07-13:
I continued to try some stuff and updated the running system a few times. So here is the current config is:
```
{
    "Hostname": "gokrazy",
    "Update": {
        "HTTPPassword": "XXXXXXXXXXXXXXX"
    },
    "Packages": [
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/wifi",
        "github.com/restic/rest-server"
    ],
    "PackageConfig": {
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "/home/syphdias/gokrazy/gokrazy/breakglass.authorized_keys"
            }
        },
        "github.com/gokrazy/gokrazy/cmd/randomd": {
            "ExtraFileContents": {
                "/etc/machine-id": "1bfe4ba3d20668c4608cd4585f9bf6e1\n"
            }
        },
        "github.com/gokrazy/wifi": {
            "ExtraFilePaths": {
                "/etc/wifi.json": "wifi.json"
            }
        }
    },
    "SerialConsole": "disabled",
    "InternalCompatibilityFlags": {}
}
```

> I have no clue, but what are the permissions of the ntp executable?

Interesting, mine does not have read and execution permissions for other. 
```
/tmp/breakglass2664188945 # ls -l /gokrazy/ntp 
-rwxr-x---    1 0        0          3077622 Jul 13  2023 /gokrazy/ntp
/tmp/breakglass2664188945 # sha256sum /gokrazy/ntp
abd55dff3cffffc8287772c1e20236a75115416bd24cc86edf99f5019fc11a0d  /gokrazy/ntp
```

> What does gok sbom print for your instance?
```
❯ gok -i gokrazy sbom
{
    "sbom_hash": "9f397af99732fe4f01f1926eeccf9f1cd1e5a18d7c75804b4d6e6ea92a5b7aeb",
    "sbom": {
        "config_hash": {
            "path": "/home/syphdias/gokrazy/gokrazy/config.json",
            "hash": "ee555a1fe1f92fcb70e17d1fa1f30d63c004e3c788528c2cf7d9bd24d9b3d867"
        },
        "go_mod_hashes": [
            {
                "path": "/home/syphdias/gokrazy/gokrazy/builddir/github.com/gokrazy/breakglass/go.mod",
                "hash": "d6c65c121d526e8a300bf3ec371aa7cc6935791d00da3d991e522e0f96a46561"
            },
            {
                "path": "/home/syphdias/gokrazy/gokrazy/builddir/github.com/gokrazy/firmware/go.mod",
                "hash": "82c5ded2f2ef53cba241fcb063fccb1fc67b0c2e48caf925424544aac5737e45"
            },
            {
                "path": "/home/syphdias/gokrazy/gokrazy/builddir/github.com/gokrazy/gokrazy/cmd/dhcp/go.mod",
                "hash": "380f8e6e53758be67d27a3f9496418cdb2b1b945debe80a62b44a83b267b5905"
[... 237 lines cut ...]
dr-xr-xr-x root/root                35 2023-07-13 17:14 squashfs-root/lib/modules/6.4.3/kernel/net/bluetooth
-r--r----- root/root           6588800 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/kernel/net/bluetooth/bluetooth.ko
-r--r----- root/root             11540 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.alias
-r--r----- root/root             11609 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.alias.bin
-r--r----- root/root             58511 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.builtin
-r--r----- root/root             62220 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.builtin.bin
-r--r----- root/root            322080 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.builtin.modinfo
-r--r----- root/root              2394 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.dep
-r--r----- root/root              3405 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.dep.bin
-r--r----- root/root                 0 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.devname
-r--r----- root/root               921 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.order
-r--r----- root/root                55 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.softdep
-r--r----- root/root              8800 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.symbols
-r--r----- root/root             11374 2023-07-13 10:31 squashfs-root/lib/modules/6.4.3/modules.symbols.bin
dr-xr-xr-x root/root                 3 2023-07-13 17:14 squashfs-root/perm
dr-xr-xr-x root/root                 3 2023-07-13 17:14 squashfs-root/proc
dr-xr-xr-x root/root                 3 2023-07-13 17:14 squashfs-root/run
dr-xr-xr-x root/root                 3 2023-07-13 17:14 squashfs-root/sys
dr-xr-xr-x root/root                 3 2023-07-13 17:14 squashfs-root/tmp
dr-xr-xr-x root/root                67 2023-07-13 17:14 squashfs-root/user
-rwxr-x--- root/root           9387953 2023-07-13 17:14 squashfs-root/user/breakglass
-rwxr-x--- root/root           6264042 2023-07-13 17:14 squashfs-root/user/serial-busybox
-rwxr-x--- root/root           5716337 2023-07-13 17:14 squashfs-root/user/wifi
dr-xr-xr-x root/root                 3 2023-07-13 17:14 squashfs-root/var
```

stapelberg, 2023-07-13:
Oh! I think it’s your umask setting. Try running `umask 0022` before running `gok`. I’ll try and submit a fix for this.

Syphdias, 2023-07-13:
@christian-heusel thanks for the hin.
I usually run umask of `0027`. I set it to 0022 and then ran `gok upgrade`. This fixed it

Syphdias, 2023-07-13:
@stapelberg Yes, exactly it. I just found out.

stapelberg, 2023-07-13:
Should be fixed now if you update your gok cli from https://github.com/gokrazy/tools

closed 2023-07-13 by stapelberg

Syphdias, 2023-07-13:
I can confirm this fixes it. Thank you!