gokrazy/gokrazy #209: Update gokrazy+tailscale docs for nftables support

stapelberg, 2023-07-23:
tailscale added experimental support for nftables in v1.46. 

As soon as the support becomes available without a flag, we should update the tailscale userguide. 

Side note: I also noticed that the resolv.conf updating code in tailscale does not work on gokrazy. Not sure how best to fix that, but we probably should fix that as well so that name resolution works out of the box.

bradfitz, 2023-07-24:
> Also, I noticed that Tailscale SSH does not work on gokrazy, as the system user lookup fails.

It used to!

I added code specifically for that. Did it regress, I wonder?  (we don't have gokrazy integration tests 🤷‍♂️ )

stapelberg, 2023-07-25:
I’m getting this panic:

<details>
<summary>panic with stack trace</summary>

<pre>
2023/07/25 18:09:59 ssh-conn-20230725T160959-49bd27820b: handling conn: 100.101.67.78:33752->michael@100.104.9.132:22
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x7a502c]

goroutine 563 [running]:
tailscale.com/ssh/tailssh.userLookup({0x40034c2fa0, 0x7})
	/home/michael/go/src/tailscale.com/ssh/tailssh/user.go:62 +0x6c
tailscale.com/ssh/tailssh.(*conn).doPolicyAuth(0x4003908000, {0xc0f170, 0x400357e738}, {0x0, 0x0})
	/home/michael/go/src/tailscale.com/ssh/tailssh/tailssh.go:381 +0x2c8
tailscale.com/ssh/tailssh.(*conn).NoClientAuthCallback(0x4003908000, {0xc0f170, 0x400357e738})
	/home/michael/go/src/tailscale.com/ssh/tailssh/tailssh.go:295 +0x40
tailscale.com/tempfork/gliderlabs/ssh.(*Server).config.func4({0xc09160?, 0x4003536d80?})
	/home/michael/go/src/tailscale.com/tempfork/gliderlabs/ssh/server.go:167 +0x64
github.com/tailscale/golang-x-crypto/ssh.(*connection).serverAuthenticate(0x4003536d80, 0x4002fec2a0)
	/home/michael/go/pkg/mod/github.com/tailscale/golang-x-crypto@v0.0.0-20230713185742-f0b76a10a08e/ssh/server.go:502 +0x694
github.com/tailscale/golang-x-crypto/ssh.(*connection).serverHandshake(0x4003536d80, 0x4002fec2a0)
	/home/michael/go/pkg/mod/github.com/tailscale/golang-x-crypto@v0.0.0-20230713185742-f0b76a10a08e/ssh/server.go:298 +0x484
github.com/tailscale/golang-x-crypto/ssh.NewServerConn({0xc0c030, 0x400350ad00}, 0x4002fec1c0)
	/home/michael/go/pkg/mod/github.com/tailscale/golang-x-crypto@v0.0.0-20230713185742-f0b76a10a08e/ssh/server.go:226 +0x1a8
tailscale.com/tempfork/gliderlabs/ssh.(*Server).HandleConn(0x40034bc5a0, {0xc0ba00, 0x40032f39e0})
	/home/michael/go/src/tailscale.com/tempfork/gliderlabs/ssh/server.go:291 +0x1a4
tailscale.com/ssh/tailssh.(*server).HandleSSHConn(0x4000221810, {0xc0ba00, 0x40032f39e0})
	/home/michael/go/src/tailscale.com/ssh/tailssh/tailssh.go:150 +0xf8
tailscale.com/ipn/ipnlocal.(*LocalBackend).handleSSHConn(0x40032f11f0?, {0xc0ba00, 0x40032f39e0})
	/home/michael/go/src/tailscale.com/ipn/ipnlocal/local.go:4723 +0x6c
tailscale.com/wgengine/netstack.(*Impl).acceptTCP(0x4000140d10, 0x400350ac40)
	/home/michael/go/src/tailscale.com/wgengine/netstack/netstack.go:893 +0x5fc
created by gvisor.dev/gvisor/pkg/tcpip/transport/tcp.(*Forwarder).HandlePacket
	/home/michael/go/pkg/mod/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/tcpip/transport/tcp/forwarder.go:97 +0x290
</pre>
</details>

bradfitz, 2023-08-15:
@stapelberg, whoops. 🤦‍♂️ Fixed.

I'd love to get Gokrazy qemu integration tests into our CI somehow but I don't have time.

closed 2023-12-23 by stapelberg