gokrazy/gokrazy #210: Upgrading version of gokrazy

BrunoTeixeira1996, 2023-07-27:
## Platform

I’m using:

* [x] gokrazy/rpi4b

## Observed behavior

When following the [docs about automation and upgrade](https://gokrazy.org/userguide/automation/) my gokrazy version does not get updated.
Even after going into `<gokrazy instance>/builddir/github.com/gokrazy/gokrazy` and performing `gok get -u` I can see that other packages get updated (i.e tailscale) but not gokrazy

```
2023/07/27 09:30:14 updating package 1 of 22: [go get github.com/gokrazy/gokrazy/cmd/dhcp]
2023/07/27 09:30:14   in builddir/github.com/gokrazy/gokrazy/cmd/dhcp
2023/07/27 09:30:16 updating package 2 of 22: [go get github.com/gokrazy/gokrazy/cmd/ntp]
2023/07/27 09:30:16   in builddir/github.com/gokrazy/gokrazy/cmd/ntp
2023/07/27 09:30:18 updating package 3 of 22: [go get github.com/gokrazy/gokrazy/cmd/randomd]
2023/07/27 09:30:18   in builddir/github.com/gokrazy/gokrazy/cmd/randomd
2023/07/27 09:30:20 updating package 4 of 22: [go get github.com/gokrazy/gokrazy/cmd/heartbeat]
2023/07/27 09:30:20   in builddir/github.com/gokrazy/gokrazy/cmd/heartbeat
2023/07/27 09:30:23 updating package 5 of 22: [go get github.com/gokrazy/gokrazy]
2023/07/27 09:30:23   in builddir/github.com/gokrazy/gokrazy
2023/07/27 09:30:23 updating package 6 of 22: [go get github.com/gokrazy/kernel]
2023/07/27 09:30:23   in builddir/github.com/gokrazy/kernel
2023/07/27 09:30:23 updating package 7 of 22: [go get github.com/gokrazy/firmware]
2023/07/27 09:30:23   in builddir/github.com/gokrazy/firmware
2023/07/27 09:30:24 updating package 8 of 22: [go get github.com/gokrazy/rpi-eeprom]
2023/07/27 09:30:24   in builddir/github.com/gokrazy/rpi-eeprom
2023/07/27 09:30:24 updating package 9 of 22: [go get github.com/gokrazy/fbstatus]
2023/07/27 09:30:24   in builddir/github.com/gokrazy/fbstatus
2023/07/27 09:30:25 updating package 10 of 22: [go get github.com/gokrazy/hello]
2023/07/27 09:30:25   in builddir/github.com/gokrazy/hello
2023/07/27 09:30:25 updating package 11 of 22: [go get github.com/gokrazy/serial-busybox]
2023/07/27 09:30:25   in builddir/github.com/gokrazy/serial-busybox
2023/07/27 09:30:26 updating package 12 of 22: [go get github.com/gokrazy/breakglass]
2023/07/27 09:30:26   in builddir/github.com/gokrazy/breakglass
2023/07/27 09:30:26 updating package 13 of 22: [go get github.com/gokrazy/wifi]
2023/07/27 09:30:26   in builddir/github.com/gokrazy/wifi
2023/07/27 09:30:27 updating package 14 of 22: [go get github.com/gokrazy/stat/cmd/gokr-webstat]
2023/07/27 09:30:27   in builddir/github.com/gokrazy/stat/cmd/gokr-webstat
2023/07/27 09:30:29 updating package 15 of 22: [go get github.com/gokrazy/stat/cmd/gokr-stat]
2023/07/27 09:30:29   in builddir/github.com/gokrazy/stat/cmd/gokr-stat
2023/07/27 09:30:31 updating package 16 of 22: [go get github.com/gokrazy/mkfs]
2023/07/27 09:30:31   in builddir/github.com/gokrazy/mkfs
2023/07/27 09:30:32 updating package 17 of 22: [go get tailscale.com/cmd/tailscaled]
2023/07/27 09:30:32   in builddir/tailscale.com
go: downloading tailscale.com v1.46.1
go: upgraded tailscale.com v1.46.0 => v1.46.1
2023/07/27 09:30:36 updating package 18 of 22: [go get tailscale.com/cmd/tailscale]
2023/07/27 09:30:36   in builddir/tailscale.com
```

after checking the `go.mod` file I still see an old file

```
$ cat go.mod
module gokrazy/build/brun0-pi

go 1.19

require (
        github.com/gokrazy/gokrazy v0.0.0-20230723153554-331c98011aad // indirect
        github.com/gokrazy/internal v0.0.0-20220129150711-9ed298107648 // indirect
        github.com/google/renameio/v2 v2.0.0 // indirect
        github.com/kenshaw/evdev v0.1.0 // indirect
        github.com/mdlayher/watchdog v0.0.0-20201005150459-8bdc4f41966b // indirect
        golang.org/x/sys v0.5.0 // indirect
)
```

## Expected behavior

Performing `gok -i <instance> get --update_all` should update all old packages to newer versions.

Syphdias, 2023-07-27:
I believe this only updates the local configuration not the device itself. What you want to do after you updated the packages, is to run `gok -i <instance> update` to update your device.

This is why the documentations says:
>For example, I use the following Makefile. I run `make get && make update` from [cron](https://en.wikipedia.org/wiki/Cron) every day.

BrunoTeixeira1996, 2023-07-27:
@Syphdias Yep I know that. The `get --update_all` should update localy and then I do `gok -i <instance> update` but the version is still outdated.
Btw, doing `get --update_all` should already tell me I have an outdated version of gokrazy and it does not show me that.

stapelberg, 2023-07-28:
This should be fixed with https://github.com/gokrazy/tools/commit/c5e8689f03bdb7007425642bdefaa2eafff2ed91 

(There was a mismatch for the init process where it used `builddir/init` instead of `builddir/github.com/gokrazy/gokrazy` as expected.)

closed 2023-07-28 by stapelberg

BrunoTeixeira1996, 2023-07-28:
@stapelberg after updating `gok` with the latest version and then doing `gok get --update_all` I still don't get the update

## gok version

```
gok  v0.0.0-20230728064552-c5e8689f03bd      up-to-date
```

## version of gokrazy after update

```
brun0@tb:~/gokrazy/brun0-pi/builddir/github.com/gokrazy/gokrazy
$ cat go.mod
module gokrazy/build/brun0-pi

go 1.19

require (
        github.com/gokrazy/gokrazy v0.0.0-20230723153554-331c98011aad // indirect
        github.com/gokrazy/internal v0.0.0-20220129150711-9ed298107648 // indirect
        github.com/google/renameio/v2 v2.0.0 // indirect
        github.com/kenshaw/evdev v0.1.0 // indirect
        github.com/mdlayher/watchdog v0.0.0-20201005150459-8bdc4f41966b // indirect
        golang.org/x/sys v0.5.0 // indirect
)
```

But going into `/gokrazy/brun0-pi/builddir/init` I can see a different version 

```
$ cat go.mod 
module gokrazy/build/brun0-pi

go 1.20

require github.com/gokrazy/gokrazy v0.0.0-20230701125004-edba3d33ffa9 // indirect
```

What should I do in this situation? Can I hardcoded this version on the `go.mod` file present in `/gokrazy/brun0-pi/builddir/github.com/gokrazy/gokrazy` ?

stapelberg, 2023-07-28:
> github.com/gokrazy/gokrazy v0.0.0-20230723153554-331c98011aad

This already is the latest version, there is nothing newer.

Just do a `gok update` and it should pick up builddir/github.com/gokrazy/gokrazy to build the init program.

BrunoTeixeira1996, 2023-07-28:
@stapelberg I can confirm this fixed!
Thanks