gokrazy/gokrazy #213: Using different port does not work

BrunoTeixeira1996, 2023-08-23:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

By changing `HTTPPort` in `config.json` to `9999` I am having an error.
Note that i was doing this while connecting with tailscale since I was not on the same network

```
2023/08/23 21:40:59 probing url for https: Get "http://brun0-pi:9999": dial tcp 192.168.30.12:9999: connect: connection refused
```

## Expected behavior

Listening on this port should be not refused and should work as intended.

stapelberg, 2023-08-23:
I think this is the same problem as https://github.com/gokrazy/gokrazy/discussions/200 — you’ll need to temporarily override the full update URL to deploy the changed image (or use `gok overwrite` after inserting the SD card, of course).

BrunoTeixeira1996, 2023-08-23:
Yes it is. Doing the following got this working without the full overwrite

```
    "Update": {
        "HTTPPassword": "<your_password>",
	"HTTPPort": "9999"
    },
    "InternalCompatibilityFlags": {
        "Update": "http://gokrazy:<your_password>@brun0-pi"
    },
 ```

BrunoTeixeira1996, 2023-08-23:
@stapelberg I then tried to delete these modifications (in order to use port 80 again) but I am having another error

```
2023/08/23 23:45:44 checking target partuuid support: Get "https://gokrazy:***@brun0-pi/update/features": dial tcp 192.168.30.12:443: connect: connection refused
```

Is this known? 
The only way now is to use the overwrite?

BrunoTeixeira1996, 2023-08-23:
I've managed to make this work again by deleting the `HTTPPort` then `gok update` then delete the full update URL from `InternalCompatibilityFlags` and `gok update`. But I think this should be done differently.
Maybe using flags in the `gok` command instead of using stuff in config.json for this case

closed 2023-08-25 by BrunoTeixeira1996