gokrazy/gokrazy #216: Loading device tree overlays

stv0g, 2023-09-12:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Description

I would like to load device tree overlays via the Raspberry Pi's bootloader (`start.elf`).
From my understanding gokrazy is using the stock Raspberry firmware from `gokrazy/firmware`.
Meaning we should be able to pass additional `dtoverlay` lines in the `config.txt`.
During boot the bootloader should then read these `.dtbo` files and merge them into a flattened `.dtb` which is passed of to the kernel.

My question: Is it possible to use the device tree overlays from the Raspberry Pi Foundations kernel with the gokrazy kernel? Has this been attempted already? Or are there better approaches for customizing the device tree used by gokrazy?

stapelberg, 2023-09-13:
> My question: Is it possible to use the device tree overlays from the Raspberry Pi Foundations kernel with the gokrazy kernel? Has this been attempted already? 

I think I vaguely remember someone using a device tree overlay at some point.

For testing, you could copy the files of the boot partition, re-create said partition (mkfs.vfat), copy back your files with any modifications you need.

For making gokrazy pick up dtbs, you need to change (at least) what gokrazy includes in the boot partition: https://github.com/gokrazy/tools/blob/23cde3b0d858497a63c21e93ad30859bf197995f/internal/packer/write.go#L151

> Or are there better approaches for customizing the device tree used by gokrazy?

No, this use-case isn’t well served right now. But I’m willing to accept changes to make it better :)

Doridian, 2023-10-05:
Poking in here because I need the ability to load DT overlays and wanting to contribute an enhancement to add as much.

@stapelberg What would your idea be of where these go? Firmware or kernel?
Firmware makes semantically more sense to be but kernel already has the `.dtb` file, which the `.dtbo` would overlay over, and keeping them in separate repos doesn't make that much sense to me.

stapelberg, 2023-10-05:
The devicetree is part of the kernel, so device tree overlays should go into the kernel repository, I think.

Doridian, 2023-10-17:
@stapelberg Opened a PR for this https://github.com/gokrazy/tools/pull/63 let me know what you think. This just adds the ability for the kernel package to contain `overlay/*.dtbo` and those to be included.

stapelberg, 2023-12-12:
As part of looking at https://github.com/gokrazy/kernel/pull/475/files, I came across the device tree overlay issue again.

I think what needs to be done is:

- [ ] Change the gokrazy/firmware auto-update setup to also pull in the boot/overlays directory from the raspberrypi/firmware repository
    - [ ] Change gokrazy/tools/internal/packer/write.go to include "overlays/*.dtbo" in firmwareGlobs
- [ ] Make the config.txt on the boot partition configurable through gokrazy’s instance config.json, most likely just with a config_addendum mechanism where the user-supplied config is appended to what’s already in gokrazy/kernel/config.txt

Not all overlays will necessarily work depending on how specific they are to the Raspberry Pi OS kernel. But chances are that some just work.

Let me know if anyone is interested in contributing support for this.

Doridian, 2023-12-12:
I'll see if I can get the first bit done to pull in the blobs (I also just now only realized you merged my previous PR, thanks for merging it).
The second one I started looking at somewhat when initially investigating that PR's change and I'll see if I can get some time to actually look at it later this week.

If someone else wants to poke in, they are more than welcome of course :)

stapelberg, 2023-12-15:
OK, we have the overlays directory included in the gokrazy/firmware repository now, and the packer includes them in the boot partition. 

What’s left is introducing the config_addendum mechanism in gokrazy/internal/config, using it in gokrazy/tools and documenting it in gokrazy/gokrazy/website.

stapelberg, 2023-12-15:
I had a few minutes of downtime, so I introduced the BootloaderExtraLines JSON array in config.json and made the packer write those lines to config.txt.

Still left to do: documentation

If anyone wants to give this a shot and report back, success/failure reports are welcome!

dme, 2023-12-16:
After updating all of the relevant pieces, then updating `config.json` with:

```javascript
    "BootloaderExtraLines": [
	"dtoverlay=w1-gpio"
    ]
```

I rebuilt (ensuring to use the default kernel package, not my own) to get:

```
/perm/home # ls /sys/bus/w1/devices
28-4adf871e64ff  28-89ca871e64ff  28-a48c071e64ff  w1_bus_master1
/perm/home # 
```

Which I believe means that this all works :+1:

closed 2023-12-18 by stapelberg

stapelberg, 2024-06-23:
FYI, with commit https://github.com/gokrazy/firmware/commit/a1d981711399c325a831c1bc729cf301140c19a0, I moved the overlays files out of the gokrazy/firmware repository and into the new gokrazy/kernel.rpi repository, where they belong :)