gokrazy/gokrazy #220: Huge cpu and mem load on gok update

BrunoTeixeira1996, 2023-09-21:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I am using a LXC responsible for managing my gokrazy instance. I have a small container that contains the gokrazy config so every time I need to update and uprade I use that machine instead of my personal laptop.
Initialy the container had 1 gb ram and 1 vCpu but then i tried to `gok update` and the containers just crashed.
Then I decided to increse to 4gb ram and 2 vCpus and the update worked but looking at proxmox graph I've noticed a huge load both on cpu and on ram.

[image]

[image]

Is this known?

## Expected behavior

Updating should not require a huge ammount of ram and vCpu

stapelberg, 2023-09-21:
Can you provide your config file? I wonder which processes are causing this resource usage.

Maybe it’s the Go compiler when you’re compiling lots of software? Then the next question would be why the cache isn’t used more.

If it’s the gok tool itself, that would be unexpected.

Maybe you can run top, or atop, or some other kind of system activity tool to find out what’s happening in more detail?

BrunoTeixeira1996, 2023-09-21:
Here is my config file.

```json
{
    "Hostname": "_____",
    "Update": {
        "HTTPPassword": "_____"
    },
    "Packages": [
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/wifi",
        "github.com/gokrazy/mkfs",
        "tailscale.com/cmd/tailscaled",
        "tailscale.com/cmd/tailscale",
        "github.com/BrunoTeixeira1996/waiw/cmd/waiw",
        "github.com/prometheus/node_exporter",
        "github.com/gokrazy/rsync/cmd/gokr-rsyncd",
        "github.com/TheMightyGit/rssole/cmd/rssole",
        "github.com/gokrazy/stat/cmd/gokr-webstat",
        "github.com/gokrazy/stat/cmd/gokr-stat"
    ],
    "PackageConfig": {
        "github.com/BrunoTeixeira1996/waiw/cmd/waiw": {
            "CommandLineFlags": [
                "-gokrazy",
                "-ip='____'",
                "-user='____'",
                "-password='____'",
                "-dbname='____'"
            ],
            "ExtraFilePaths": {
                "/etc/waiw/assets": "/root/personal/waiw/assets",
                "/etc/waiw/templates": "/root/personal/waiw/templates"
            }
        },
        "github.com/TheMightyGit/rssole/cmd/rssole": {
            "CommandLineFlags": [
                "-gokrazy",
                "-c=/perm/home/rssole/feeds.json",
                "-r=/perm/home/rssole/readcache.json"
            ],
            "ExtraFilePaths": {
                "/etc/rssole/feeds.json": "/root/gokrazy/brun0-pi/rssole/feeds.json",
                "/etc/rssole/readcache.json": "/root/gokrazy/brun0-pi/rssole/readcache.json"
            }
        },
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "/root/gokrazy/brun0-pi/breakglass.authorized_keys"
            }
        },
        "github.com/gokrazy/gokrazy/cmd/randomd": {
            "ExtraFileContents": {
                "/etc/machine-id": "7ac73ed61b3b48b694e7cb5584e08890\n"
            }
        },
        "github.com/gokrazy/rsync/cmd/gokr-rsyncd": {
            "CommandLineFlags": [
                "--gokr.config=/etc/gokr-rsyncd.toml",
                "--daemon"
            ],
            "ExtraFilePaths": {
                "/etc/gokr-rsyncd.authorized_keys": "gokr-rsyncd.authorized_keys",
                "/etc/gokr-rsyncd.toml": "gokr-rsyncd.toml"
            }
        },
        "github.com/gokrazy/stat/cmd/gokr-webstat": {
            "CommandLineFlags": [
                "-modules=cpu,disk,sys,net,mem,thermal"
            ]
        },
        "github.com/gokrazy/wifi": {
            "ExtraFilePaths": {
                "/etc/wifi.json": "wifi.json"
            }
        },
        "tailscale.com/cmd/tailscale": {
            "CommandLineFlags": [
                "up",
                "--advertise-routes=192.168.30.0/24"
            ]
        }
    },
    "SerialConsole": "disabled",
    "InternalCompatibilityFlags": {}
}
```

I tried using gok update again and then capture the top output

[image]

And when uploading

[image]

BrunoTeixeira1996, 2023-09-21:
Note that i am only running gok update on this container. Nothing is happening in the background too. Ive noticed this when using gok update on my laptop but i had more staff running on the background so I ignored it. But this LXC is empty:/

stapelberg, 2023-09-23:
Hmm. The screenshots you posted show an idle system (top shows the CPU is 99% idle and plenty of memory is available). Which program is showing you the high CPU and memory usage?

I tried using `gok overwrite` with your config, and everything seems to work as expected on first glance. The `gok` tool itself doesn’t show up in `top` on my machine, just a bunch of `go` compiler processes. 

Can you see if running `gok sbom` uses too many resources as well? 

Can you see if the problem is reproducible on your machine with a fresh new gokrazy instance? That should show us whether the issue is related to the machine setup or to the gokrazy config.

BrunoTeixeira1996, 2023-09-23:
@stapelberg can you guide me on how to make a fresh install without loosing stuff on gokrazy? So I can test what you asked?
Thanks

stapelberg, 2023-09-23:
> @stapelberg can you guide me on how to make a fresh install without loosing stuff on gokrazy? So I can test what you asked?
> Or you want me to clean the config.json and gok update to see if that is slow too?

Just create another instance: `gok new -i test` (will create ~/gokrazy/test/config.json) and then try it with that instance. No need to modify your current config.

BrunoTeixeira1996, 2023-09-23:
> Hmm. The screenshots you posted show an idle system (top shows the CPU is 99% idle and plenty of memory is available). Which program is showing you the high CPU and memory usage?

The program that is showing me the high CPU and memory usage is part of the Proxmox dashboard.

> Can you see if running gok sbom uses too many resources as well?

`gok sbom` prints a json object in 1 second so but after checking the proxmox tool i have the following output on the CPU

[image]

I've tracked the memory usage and it was fine, but the CPU got a huge peak.

> Just create another instance: gok new -i test (will create ~/gokrazy/test/config.json) and then try it with that instance. No need to modify your current config.

This failed but still I think you can see the CPU usage going up and down and this happens on the normal gokrazy instance

https://github.com/gokrazy/gokrazy/assets/12052283/03addd3e-a9a3-4d19-ab44-6cb48de1a365

stapelberg, 2023-09-23:
> `gok sbom` prints a json object in 1 second so but after checking the proxmox tool i have the following output on the CPU

Okay. This tells us that gokrazy’s configuration evaluation (e.g. walking through all `ExtraFiles`) is not the culprit.

> This failed but still I think you can see the CPU usage going up and down and this happens on the normal gokrazy instance

Yes, in your video we can see the `go` tool (compiler/linker) consuming CPU.

This is expected. 

Fundamentally, compiling Go code requires work, and there’s nothing we can do to get rid of this work. So the total amount of CPU time consumed will stay fixed. 

If you say the spike in CPU usage is undesired in some way, I would recommend you limit the maximum CPU usage somehow, e.g. by only allocating a certain number of cores in proxmox.

As I said, this won’t change the total CPU time, but it can change the shape of the graph — you can turn the high peaks into lower CPU usage but for a longer time.

BrunoTeixeira1996, 2023-09-23:
Yes that makes total sense @stapelberg .
Just for the sake of knowledge. When you update your gokrazy instace how much time it takes? Mine i usualy 1/2 minutes

stapelberg, 2023-09-23:
> Just for the sake of knowledge. When you update your gokrazy instace how much time it takes? Mine i usualy 1/2 minutes

My largest instance takes 12 seconds to build, and about half a minute to update (including rebooting the machine). But that’s on a pretty fast PC: https://michael.stapelberg.ch/posts/2022-01-15-high-end-linux-pc/

---

Now that we established that there is no unnecessary work we can just remove, maybe it makes more sense to talk about the original prompt: your containers crashing. 

If the CPU usage is the problem, you can run the `gok` tool through `nice`, which allows you to change the priority, meaning you should still be able to log in and run commands, even when `gok` consumes all the available CPU.

If the RAM usage is the problem, you can trade off update speed for lower peak RAM usage by limiting the compilation parallelism. Recompile the `gok` tool with a change like this one:

```diff
--- i/packer/gotool.go
+++ w/packer/gotool.go
@@ -268,6 +268,7 @@ func (be *BuildEnv) Build(bindir string, packages []string, packageBuildFlags, p
 	incompletePkgs = append(incompletePkgs, noBuildPackages...)
 
 	var eg errgroup.Group
+	eg.SetLimit(1)
 	for _, incompleteNoBuildPkg := range noBuildPackages {
 		buildDir, err := be.BuildDir(incompleteNoBuildPkg)
 		if err != nil {
```

…and then you should never see more RAM usage than is required for compiling each individual go program you’re including in your gokrazy instance.

BrunoTeixeira1996, 2023-09-23:
Oh ok, ye your pc is fast so I assume that helps too.
Thanks for your help and advice on how to fix this. Since this is not a gok problem i think we can close this issue
Thanks once again

closed 2023-09-23 by stapelberg