gokrazy/gokrazy #221: Windows not supported? (gok CLI)

MadSpindel, 2023-09-22:
Really excited to try this project out! However, I'm using Windows 11 and it looks like the gok CLI doesn't support it? Couldn't find any info about supported platforms for the gok CLI, so not sure if this is a bug or expected?

## Platform
go version go1.21.1 windows/amd64 (installed with winget install GoLang.Go)

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior
~> go install github.com/gokrazy/tools/cmd/gok@main
```
# github.com/gokrazy/tools/packer
go\pkg\mod\github.com\gokrazy\tools@v0.0.0-20230921171328-231dea4b5704\packer\packer.go:458:7: undefined: unix.Sync
go\pkg\mod\github.com\gokrazy\tools@v0.0.0-20230921171328-231dea4b5704\packer\packer.go:464:7: undefined: unix.Sync
# github.com/gokrazy/internal/squashfs
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:466:16: undefined: unix.S_IRUSR
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:466:31: undefined: unix.S_IXUSR
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:467:11: undefined: unix.S_IRGRP
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:467:26: undefined: unix.S_IXGRP
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:468:11: undefined: unix.S_IROTH
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:468:26: undefined: unix.S_IXOTH
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:490:16: undefined: unix.S_IRUSR
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:490:31: undefined: unix.S_IXUSR
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:491:11: undefined: unix.S_IRGRP
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:491:26: undefined: unix.S_IXGRP
go\pkg\mod\github.com\gokrazy\internal@v0.0.0-20230313074923-5f469e7488b0\squashfs\writer.go:491:26: too many errors
```

## Expected behavior
Installation just works

stapelberg, 2023-09-22:
Thanks for the report. I don’t use Windows, so I have no idea in what shape things are on Windows.

If you (or someone) want to contribute changes for better Windows support, I’d be happy to review the PRs. Thanks for your understanding.

MadSpindel, 2023-09-23:
@BrunoTeixeira1996 Yeah, wsl2 is quite awesome, I use it daily with Alpine Linux. I'll use it for this in the short term, but I believe it will still have some issues with for example overwrite Gokrazy to SD card (attaching USB drives to wsl2 is not supported out-of-the-box). And I guess it could be this overwrite function that's not supported on Windows? I don't have the time to debug now, maybe later in the future.

stapelberg, 2023-09-23:
If the `overwrite` feature doesn’t work with devices on Windows, you can use `overwrite --full` to write to a file, then copy said file to the SD card with a third-party tool. After the first SD card writing, updates are done over the network, so this is only cumbersome once.