gokrazy/gokrazy #223: Add support for the Raspberry Pi 5

stapelberg, 2023-09-28:
This issue tracks the progress. Currently, the Pi 5 is not in stock.

tobru, 2023-11-20:
I'm happy to help testing on my freshly arrived Pi 5 once you're starting to work on it.

stapelberg, 2023-11-28:
My first Pi 5 arrived 🎉

When I just try to boot a current gokrazy installation, I get the following error message by the bootloader:

[image]

When I set `os_check=0`, I don’t get any HDMI output.

I would have checked the serial port, but I don’t get any output on the GPIO header. I ordered a Raspberry Pi Debug Probe to use the JST port.

damdo, 2023-11-28:
Exciting! Happy hacking, let us know how it goes :)

stapelberg, 2023-11-30:
The Debug Probe arrived, so now we can see the serial output of the entire boot process.

For the attempt with `os_check=0` in `config.txt`, output just stops before the Linux kernel ever prints a line.

I then tried dropping https://github.com/raspberrypi/firmware/blob/master/boot/bcm2712-rpi-5-b.dtb into the kernel directory.

This made it go a whole lot further, in that the bootloader recognizes the installation as compatible, we see Linux kernel output, but then the boot hangs when it tries to find the root device:

<details>
<summary>full serial output</summary>

```

RPi: BOOTSYS release VERSION:30de0ba5 DATE: 2023/10/30 TIME: 16:45:10
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1698684310 serial d69878eb boardrev c04170 stc 3031485
AON_RESET: 00000003 PM_RSTS 00001000
RP1_BOOT chip ID: 0x20001927
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
PMIC reset-event 00000000 rtc 00000002 alarm 00000000 enabled 0
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 4267
DDR 4267 1 0 32 152
RP1_BOOT chip ID: 0x20001927

RP1_BOOT chip ID: 0x20001927
RP1_BOOT: fw size 25968
PCI2 init
PCI2 reset
PCIe scan 00001de4:00000001
RP1_CHIP_INFO 20001927

RPi: BOOTLOADER release VERSION:30de0ba5 DATE: 2023/10/30 TIME: 16:45:10
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1698684310 serial d69878eb boardrev c04170 stc 5918974
AON_RESET: 00000003 PM_RSTS 00001000
usb_pd_init status 3
USB_PD CONFIG 0 41
Boot mode: SD (01) order f4
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8000 [174]
CID: 000353445343313647801cd98aba0143
CSD: 400e00325b59000076b27f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358443 0x00000000
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  204800 type: 0x0c
MBR: 0x00000001,    8191 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
USB-PD: src-cap PDO object1 0x0a0191f4
Current 5000 mA
Voltage 5000 mV
USB-PD: src-cap PDO object2 0x0002d12c
Current 3000 mA
Voltage 9000 mV
USB-PD: src-cap PDO object3 0x0003c0e1
Current 2250 mA
Voltage 12000 mV
USB-PD: src-cap PDO object4 0x0004b0b4
Current 1800 mA
Voltage 15000 mV
Trying partition: 0
type: 16 lba: 8192 'gokrazy!' '            ' clusters 51161 (4)
rsc 4 fat-sectors 145 root dir cluster 1 sectors 5 entries 80
FAT16 clusters 51161
[sdcard] autoboot.txt not found
Trying partition: 0
type: 16 lba: 8192 'gokrazy!' '            ' clusters 51161 (4)
rsc 4 fat-sectors 145 root dir cluster 1 sectors 5 entries 80
FAT16 clusters 51161
[... 287 lines cut ...]
[    1.750376] cfg80211: failed to load regulatory.db
[    1.750385] ALSA device list:
[    1.758171]   No soundcards found.
[    1.761634] uart-pl011 107d001000.serial: no DMA platform data
[    1.767528] Waiting for root device PARTUUID=60c24cc1-f3f9-427a-8199-2e18c40c0001/PARTNROFF=1...
[   11.867307] platform cam1_reg: deferred probe pending
[   11.872388] platform 107d50c000.serial: deferred probe pending
[   11.878246] platform 107d004000.spi: deferred probe pending
[   11.883840] platform leds: deferred probe pending
[   11.888560] platform wl_on_reg: deferred probe pending
[   11.893717] platform cam0_reg: deferred probe pending
[   11.898788] platform pwr_button: deferred probe pending
[   32.088607] vcc-sd: disabling

```

</details>

So, my guess is that we need to sort out the following issues:
* [ ] upstream the device tree changes so that Linux builds a (working) bcm2712-rpi-5-b.dtb
* [ ] make sure the mmc driver works. 
    * The Raspberry Pi kernel seems to use bcm2712-sdhci, but that’s probably not in the upstream kernel yet. 
    * Maybe the driver currently in upstream Linux works as well, or maybe the other one needs upstreaming.
* [ ] make sure the ethernet driver works
* [ ] make sure the USB driver works

stapelberg, 2024-03-13:
Status update: 6by9, the person working on upstream support for the Pi 5, reports that it will take quite a long time because of a DMA-related long-standing misunderstanding in the kernel that’s hard to fix: https://forums.raspberrypi.com/viewtopic.php?p=2202581&sid=d2fbe43c5b4e9f133bf7842210a19d7d#p2202581

damdo, 2024-03-13:
Thanks for the update @stapelberg 
Hopefully they'll be able to figure it out 🤞

stapelberg, 2024-06-03:
Quick status update: over in https://github.com/gokrazy/gokrazy/issues/264, we now have an experimental Raspberry Pi kernel (not yet auto-updated), and I can successfully boot my Pi 5 with it, including fan control and encrypted WiFi.

If anyone is eager to try it out, feel free to give it a go.

stapelberg, 2024-06-10:
Got a chance to verify the crypto acceleration is working (using Go’s `crypto/cipher` standard library benchmark):

Pi 4:
```
goos: linux
goarch: arm64
pkg: crypto/cipher
BenchmarkAESGCM/Open-128-64-4         	  221560	      5386 ns/op	  11.88 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Seal-128-64-4         	  222818	      5454 ns/op	  11.73 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Open-256-64-4         	  191671	      6242 ns/op	  10.25 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Seal-256-64-4         	  194470	      8807 ns/op	   7.27 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Open-128-1350-4       	    9578	    121547 ns/op	  11.11 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Seal-128-1350-4       	    9826	    122123 ns/op	  11.05 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Open-256-1350-4       	   15530	     77287 ns/op	  17.47 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Seal-256-1350-4       	   15514	     77644 ns/op	  17.39 MB/s	      48 B/op	       3 allocs/op
BenchmarkAESGCM/Open-128-8192-4       	    2704	    441286 ns/op	  18.56 MB/s	      51 B/op	       3 allocs/op
BenchmarkAESGCM/Seal-128-8192-4       	    2707	    443302 ns/op	  18.48 MB/s	      51 B/op	       3 allocs/op
BenchmarkAESGCM/Open-256-8192-4       	    2377	    518658 ns/op	  15.79 MB/s	      51 B/op	       3 allocs/op
BenchmarkAESGCM/Seal-256-8192-4       	    3032	    346575 ns/op	  23.64 MB/s	      51 B/op	       3 allocs/op
PASS
```

Pi 5:

```
goos: linux
goarch: arm64
pkg: crypto/cipher
BenchmarkAESGCM/Open-128-64-4         	 5133783	       236.0 ns/op	 271.24 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Seal-128-64-4         	 5399284	       222.4 ns/op	 287.81 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Open-256-64-4         	 4742012	       252.8 ns/op	 253.21 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Seal-256-64-4         	 4764456	       249.3 ns/op	 256.75 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Open-128-1350-4       	  957399	      1257 ns/op	1073.62 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Seal-128-1350-4       	  959366	      1249 ns/op	1081.25 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Open-256-1350-4       	  815275	      1474 ns/op	 916.07 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Seal-256-1350-4       	  806194	      1488 ns/op	 907.16 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Open-128-8192-4       	  185481	      6471 ns/op	1265.96 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Seal-128-8192-4       	  185899	      6452 ns/op	1269.70 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Open-256-8192-4       	  154554	      7766 ns/op	1054.85 MB/s	       0 B/op	       0 allocs/op
BenchmarkAESGCM/Seal-256-8192-4       	  154186	      7787 ns/op	1052.00 MB/s	       0 B/op	       0 allocs/op
PASS
```

At this point, once https://github.com/gokrazy/gokrazy/issues/264 is fixed, we can call the Pi 5 supported.

stapelberg, 2024-06-23:
Alright, I think all remaining work items have been completed by now. Give the https://github.com/gokrazy/kernel.rpi kernel (now the default) a shot if you have a Raspberry Pi 5 :) Please open new issues if something doesn’t work.

closed 2024-06-23 by stapelberg

damdo, 2025-08-01:
Sounds good! Let's see :)

damdo, 2026-09-09:
I saw there has been some recent movement on this both here: https://gitlab.com/linux-kernel/linux-next/-/tree/master/arch/arm64/boot/dts/broadcom?ref_type=heads and here: https://github.com/u-boot/u-boot/tree/main/dts/upstream/src/arm64/broadcom

@stapelberg do you think it might warrant trying it out again on an rpi5? (I still don't have one, so I can't 😅)

stapelberg, 2026-09-13:
Good news! With the config changes in commit https://github.com/gokrazy/kernel/commit/238e3f63fad5f84a7ba844caf505adfd74a90801, I can boot the upstream Linux kernel on a Pi 5, with working serial console (needed to set `"SerialConsole": "ttyAMA10,115200",`), Ethernet, WiFI (!), USB and SD card 🥳

Notably, there does not seem to be a driver for the Pi fan yet, so it will stay off when using the upstream kernel. A patch series for the PWM parts is WIP (https://ratatoskr.run/linux-pwm/2026/09/17531114/t), but the thermal zones also need driver support. 

So, the upstream kernel now seems ready, provided you don’t need a fan for your setup.

damdo, 2026-09-13:
That's awesome news! 🎉
Thanks a lot for trying it and confirming!

damdo, 2026-09-13:
I've opened a couple of PRs to add this to the docs:
- https://github.com/gokrazy/gokrazy/pull/391
- https://github.com/gokrazy/kernel/pull/725