gokrazy/gokrazy #230: command not found

overtube, 2023-11-04:
## Platform

I’m using:

* [ ] gokrazy/x86-64

## Observed behavior
`go install github.com/gokrazy/tools/cmd/gok@main`
`go: downloading github.com/donovanhide/eventsource v0.0.0-20210830082556-c59027999da0`
`go: downloading github.com/google/renameio/v2 v2.0.0`
:~ gok
bash: gok: command not found`

damdo, 2023-11-04:
I think you might have an issue with your environment, it works in a clean environment in docker for example.

```
[~] $ docker run --rm -it golang:1.20-bookworm /bin/bash

root@047560fee4ba:/go#
root@047560fee4ba:/go# go install github.com/gokrazy/tools/cmd/gok@main
go: downloading github.com/gokrazy/tools v0.0.0-20231031183210-dceb56ee3943
go: downloading github.com/donovanhide/eventsource v0.0.0-20210830082556-c59027999da0
go: downloading github.com/gokrazy/internal v0.0.0-20231031182944-a645001f8b93
go: downloading github.com/google/renameio/v2 v2.0.0
go: downloading github.com/spf13/pflag v1.0.5
go: downloading github.com/gokrazy/updater v0.0.0-20230215172637-813ccc7f21e2
go: downloading github.com/spf13/cobra v1.6.1
go: downloading golang.org/x/mod v0.11.0
go: downloading golang.org/x/sync v0.1.0
go: downloading github.com/breml/rootcerts v0.2.10
go: downloading golang.org/x/sys v0.5.0
root@047560fee4ba:/go# gok
The gok tool is your main entrypoint to gokrazy and allows you to:

1. Create new gokrazy instances (gok new),
2. Deploy gokrazy instances to storage devices like SD cards (gok overwrite),
3. Update gokrazy instances over the network (gok update),
4. (For development) Run individual programs on a running gokrazy instance (gok run).

If you are unfamiliar with gokrazy, please follow:
https://gokrazy.org/quickstart/

Usage:
  gok [flags]
  gok [command]

Commands to create and edit a gokrazy instance:
  add         Add a Go package to a gokrazy instance
  edit        Edit a gokrazy instance configuration interactively
  get         Update the version of your Go program(s) using `go get`
  new         Create a new gokrazy instance

Commands to deploy and update a gokrazy instance:
  overwrite   Build and deploy a gokrazy instance to a storage device
  sbom        Print the Software Bill Of Materials of a gokrazy instance
  update      Build a gokrazy instance and update over the network

Commands to work with a running gokrazy instance:
  logs        Stream logs from a running gokrazy service
  run         `go install` and run on a running gokrazy instance

Commands to work with a remote GUS server:
  push        Push a gokrazy image to a remote GUS server

Additional Commands:
  completion  Generate the autocompletion script for the specified shell
  help        Help about any command
  version     Print gok version

Flags:
  -h, --help                help for gok
  -i, --instance string     instance, identified by hostname (default "hello")
      --parent_dir string   parent directory: contains one subdirectory per instance (default "/root/gokrazy")
      --version             print gok version

Use "gok [command] --help" for more information about a command.
```

Is `$GOPATH/bin` in your shell's `$PATH`?
i.e.
```
 echo $PATH | grep go
/go/bin:/usr/local/go/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
```

overtube, 2023-11-04:
```
 echo $PATH | grep go
/home/ui/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/opt/android-sdk/emulator:/opt/android-ndk:/opt/android-sdk/platform-tools:/opt/android-sdk/tools:/opt/android-sdk/tools/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/bin:/home/ui/.local/bin
```

stapelberg, 2023-11-05:
What’s the output of `go env`? Check `GOPATH` and `GOBIN` in particular. The default location for compiled programs is `$GOPATH/bin`, which is typically `$HOME/go/bin`.

It looks like `/home/ui/go/bin` is not in your `PATH` environment variable.

In general, gokrazy is not a good first project to get started with Go. The target audience is programmers who are already at least a little familiar with Go and want to deploy bare-metal on a Raspberry Pi.

overtube, 2023-11-07:
`go env`
```
GO111MODULE=''
GOARCH='amd64'
GOBIN=''
GOCACHE='/home/ui/.cache/go-build'
GOENV='/home/ui/.config/go/env'
GOEXE=''
GOEXPERIMENT=''
GOFLAGS=''
GOHOSTARCH='amd64'
GOHOSTOS='linux'
GOINSECURE=''
GOMODCACHE='/home/ui/Templates/@/lang/go/pkg/mod'
GONOPROXY=''
GONOSUMDB=''
GOOS='linux'
GOPATH='/home/ui/Templates/@/lang/go'
GOPRIVATE=''
GOPROXY='https://proxy.golang.org,direct'
GOROOT='/usr/lib/go'
GOSUMDB='sum.golang.org'
GOTMPDIR=''
GOTOOLCHAIN='auto'
GOTOOLDIR='/usr/lib/go/pkg/tool/linux_amd64'
GOVCS=''
GOVERSION='go1.21.3'
GCCGO='gccgo'
GOAMD64='v1'
AR='ar'
CC='gcc'
CXX='g++'
CGO_ENABLED='1'
GOMOD='/dev/null'
GOWORK=''
CGO_CFLAGS='-O3 -g'
CGO_CPPFLAGS=''
CGO_CXXFLAGS='-O3 -g'
CGO_FFLAGS='-O3 -g'
CGO_LDFLAGS='-O3 -g'
PKG_CONFIG='pkg-config'
GOGCCFLAGS='-fPIC -m64 -pthread -Wl,--no-gc-sections -fmessage-length=0 -ffile-prefix-map=/tmp/go-build3882417276=/tmp/go-build -gno-record-gcc-switches'
```

stapelberg, 2023-11-07:
> GOPATH='/home/ui/Templates/@/lang/go'

This means you need to add /home/ui/Templates/@/lang/go/bin to your $PATH.

This is also documented at https://go.dev/doc/install

closed 2023-11-07 by stapelberg

overtube, 2023-11-07:
Glad, Thank!