gokrazy/gokrazy #231: Executing on demand

BrunoTeixeira1996, 2023-11-05:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [X] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

I was wondering if it is possible to have a go appliance Run on demand. I dont want to run the program on a loop, however i want to define a time for the run.
Is it possible to do this without any external dependency on the go appliance it self?
Something like a command in gokrazy config

stapelberg, 2023-11-06:
Not sure I understand exactly what you mean, but if you want your program to do something at a certain time of the day, you’ll need to write that in Go code. 

Here’s an example of how I implemented running backups each weekday at 10, for example: https://github.com/stapelberg/zkj-nas-tools/blob/afbaf1cd767186d9244d93e4eaef87c48b30afee/dornroeschen/main.go#L66-L91

BrunoTeixeira1996, 2023-11-06:
@stapelberg I was usin a cronjob lib but I decided to go with your example approach. Anyway I am having some trouble finding what I am doing wrong using your approach but since this is not related with gokrazy Ill close this.

closed 2023-11-06 by BrunoTeixeira1996