gokrazy/gokrazy #233: GPIO / periph.io Edge detection / sysfs  not working

eripa, 2023-12-10:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [x] gokrazy/rpiz2w
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

I've followed the [GPIO guide](https://gokrazy.org/development/gpio/) and was able to run the example (Low, High, Low etc).

I then proceeded to configure my GPIO Pin for Input, which works fine if  I use no rising edge:

```go
		if err := p.In(gpio.PullUp, gpio.NoEdge); err != nil {
			log.Fatal(err)
		}
		fmt.Printf("%s is %s\n", p, p.Read())
```

However, since I want my program to react on inputs/events, I was hoping to be able to use edge detection as per the [periph.io button docs](https://periph.io/device/button/), unfortunately there seem to be some issue setting up/accessing the `/sys/class/gpio` path. I've used breakglass and there is no such path.

I can see that sysfs is enabled in the gokrazy kernel, so I'm a bit at a loss here what to do.

Found this comment on this depedency:
https://github.com/periph/host/blob/f673315ab3a9ebe15db798b450d193bc7a9a7f1c/bcm283x/gpio.go#L373-L380

## Observed behavior

```shell
2023/12/09 20:22:26 Loading periph.io drivers
2023/12/09 20:22:26 foo: bcm283x-gpio (GPIO4): pin 4 is not exported by sysfs
2023/12/09 20:22:26 gokrazy: exit status 1
```

## Expected behavior

Able to use GPIO edge detection as per `periph.io` docs.

stapelberg, 2023-12-10:
> I can see that sysfs is enabled in the gokrazy kernel, so I'm a bit at a loss here what to do.

The difference is most likely that Raspberry Pi OS uses a custom kernel driver (or custom patches), whereas gokrazy uses the regular upstream Linux kernel (without Raspberry Pi-specific patches).

I’m not sure if edge detection is generally not available with upstream Linux on the Raspberry Pi, or whether a different method needs to be implemented in the periph package.

eripa, 2023-12-10:
Sounds plausible, thanks for the reply! I suppose a can drive a little loop polling the status continuously, if it works out I can send a PR to the docs repo with an example.

closed 2023-12-12 by stapelberg

fischerman, 2024-09-23:
> I’m not sure if edge detection is generally not available with upstream Linux on the Raspberry Pi, or whether a different method needs to be implemented in the periph package.

Periph uses an outdated sysfs interface which is no longer available in Raspbian Bookworm.

See https://github.com/periph/host/issues/62

PR #234 (2023-12-12): docs: adds GPIO input example
Feel free to update the Go code as I'm not sure if this is the best approach, but I've confirmed that it's working for my use-case and is performing reasonably well.

This PR also updates to latest `v3` `periph.io` module structure for the GPIO output example. 

Resolves #233