gokrazy/gokrazy #236: support for mounting additional devices (as /perm, or separately)

stapelberg, 2023-12-18:
There has been some interest in mounting external disks as `/perm` (permanent data partition) or on arbitrary mountpoints:

* https://github.com/gokrazy/gokrazy/discussions/235
* https://github.com/gokrazy/gokrazy/issues/101

I’m not opposed to adding support for mounting disks, but we need to carefully think through which features we want to support and in which way.

For example, we don’t use udev, so we don’t have stable symlinks to the device nodes in `/dev`. Especially once you have more than one external disk, the order is not necessarily stable on boot, so I think it will be vital to have some sort of stable identifier to use. See also https://michael.stapelberg.ch/posts/2020-01-21-initramfs-from-scratch-golang/#3-block-device-identification 

Here’s a strawman proposal for how the configuration could look like:

```json
"MountDevices": [
 {
    "Device": "/dev/sdb",
    "Type": "ext4",
    "Mountpoint": "/perm/archival",
    "Options": "relatime,noexec"
  } 
]
```

(And aside from `Device`, one or more ways to specify a stable identifier should be supported.)

FiloSottile, 2024-01-27:
I would also use this, definitely with a stable identifier and probably mounting to an otherwise read-only mount point, so that if mounting fails data is not written anyway somewhere it doesn't belong.

I assume the mounts would happen before applications are started, but what if the filesystem needs checking, which can take a long time? Maybe there should be `gokrazy.WaitForFilesystem()`?

Alternatively, if there was a way to specify a dependency order between applications, this could just be an application configured with an ExtraFile, which the applications that need disks wait for.

stapelberg, 2024-04-01:
> I would also use this, definitely with a stable identifier and probably mounting to an otherwise read-only mount point, so that if mounting fails data is not written anyway somewhere it doesn't belong.

That’s a good point. We should probably introduce a `/mnt` directory on the (read-only) root file system that can be used for such mounts. 

> I assume the mounts would happen before applications are started, but what if the filesystem needs checking, which can take a long time? Maybe there should be `gokrazy.WaitForFilesystem()`?

Generally, applications have to declare what they are waiting for, gokrazy doesn’t sequence dependencies or delay application startup.

In the early days of gokrazy, this was only possible by modifying the source to call e.g. `gokrazy.WaitForClock()`, but then we introduced a configuration parameter: https://gokrazy.org/userguide/instance-config/#packagewaitforclock

I like your suggestion of `gokrazy.WaitForFilesystem(path)`, which would wait for a file system to be mounted at path. We could have a corresponding configuration parameter for applications where modifying the source code is inconvenient.

---

In https://github.com/gokrazy/gokrazy/discussions/235#discussioncomment-8812519, @thirdeyenick links to the start of an implementation. I’m not sure if we need the diskfs dependency, as gokrazy already has its own code to deal with partition tables that might be good to extend for this purpose (to avoid any dependencies we don’t absolutely need). But aside from that detail, it looks like a good start :)

stapelberg, 2024-05-10:
Alright, I implemented a first version of this feature (see the attached commits). 

Here’s what I included in my instance configuration: 

```json
    "MountDevices": [
	{
	    "Source": "PARTUUID=c995c848-1d29-4f08-a34d-cd811e84a763",
	    "Type": "ext4",
	    "Target": "/mnt/logs",
	    "Options": ""
	}
    ]

```

This is what my Raspberry Pi printed on the serial console on boot (note how the first mount attempt failed before the USB memory stick was enumerated, but gokrazy keeps trying):

```
1970/01/01 01:00:04 mount.go:238: mounting PARTUUID=c995c848-1d29-4f08-a34d-cd811e84a763: PARTUUID=c995c848-1d29-4f08-a34d-cd811e84a763 not found
[    4.797156] scsi 0:0:0:0: Direct-Access     SanDisk  Extreme Pro      0    PQ: 0 ANSI: 6
[    4.807694] sd 0:0:0:0: [sda] 250085376 512-byte logical blocks: (128 GB/119 GiB)
[    4.816030] sd 0:0:0:0: [sda] Write Protect is off
[    4.821572] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.843760]  sda: sda1 sda2 sda3 sda4
[    4.848368] sd 0:0:0:0: [sda] Attached SCSI removable disk
1970/01/01 01:00:05 mount.go:205: mounting /dev/sda4 on /mnt/logs

2024/05/10 19:01:00 gokrazy.go:359: starting shell /tmp/serial-busybox/ash upon input on serial console
/ # ls -l /mnt/logs
total 24
drwxr-xr-x    4 0        0             4096 May  8 18:01 install-logs-2024-05-08.0
drwxr-xr-x    4 0        0             4096 May  8 21:12 install-logs-2024-05-08.1
drwx------    2 0        0            16384 May  8 18:01 lost+found
/ # df -h /mnt/logs
Filesystem                Size      Used Available Use% Mounted on
/dev/sda4               111.2G      5.1M    105.5G   0% /mnt/logs
/ # mount
/dev/root on / type squashfs (ro,relatime,errors=continue)
devtmpfs on /dev type devtmpfs (rw,relatime,size=1892336k,nr_inodes=473084,mode=755)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk1p4 on /perm type ext4 (rw,relatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,relatime)
tmpfs on /etc/cni/net.d type tmpfs (rw,relatime)
/dev/sda4 on /mnt/logs type ext4 (rw,relatime)
/ # cat /etc/gokrazy/mountdevices.json 
[{"Source":"PARTUUID=c995c848-1d29-4f08-a34d-cd811e84a763","Type":"ext4","Target":"/mnt/logs","Options":""}]/ # 
/ # 
```

Let me know if you have any feedback / if it works for you!

Next step: adding documentation, perhaps more supported source devices (currently only `PARTUUID=`  and full device paths).

vilhalmer, 2024-07-04:
This looks really useful, I'm currently working through getting gokrazy running in DigitalOcean and I'd love to be able to mount the metadata which lives on an iso9660 device. The Options don't actually do anything yet, so I'm getting EACCES from the mount call since it's a read only file system.

Rather than trying to implement all of them here, I wonder if pulling in something like https://pkg.go.dev/github.com/moby/sys/mount would be a better option?

stapelberg, 2024-07-05:
> This looks really useful, I'm currently working through getting gokrazy running in DigitalOcean 

Cool!

> and I'd love to be able to mount the metadata which lives on an iso9660 device. The Options don't actually do anything yet, so I'm getting EACCES from the mount call since it's a read only file system.

Huh. I would have expected the file system to automatically fall back to read-only. 

> Rather than trying to implement all of them here, I wonder if pulling in something like https://pkg.go.dev/github.com/moby/sys/mount would be a better option?

I’m hesitant to add new dependencies unless absolutely necessary. In this case, copying the option parsing seems feasible, and it’s not that many lines: https://github.com/moby/sys/blob/35f6c4e79794685251bedd215ca6dcc2bf7860d7/mount/flags_unix.go#L119

vilhalmer, 2024-07-05:
> Huh. I would have expected the file system to automatically fall back to read-only.

I did as well which is why it took me ages to figure out what was going on, turns out that's a feature of `libmount` rather than the syscall itself.

> In this case, copying the option parsing seems feasible

Definitely. My reasoning for adding a dep was that it's unlikely to change much and thus be low-risk, but that's just as good of a reason to simply duplicate it.

I'll take a look at implementing this, and possibly adding support for mounting by `LABEL=` as well. So far running in DO has mostly Just Worked, I'm very impressed with gokrazy! Only other issue I've run into is that dhcp only supports one interface and they use a separate one for the private network, but I've written a tiny workaround to apply the IPs from the cloud-init metadata for the time being.

thirdeyenick, 2024-07-13:
I just tried the new feature and it works really well for me. Thanks for the implementation!

stapelberg, 2024-07-19:
Documentation is now available at https://gokrazy.org/userguide/instance-config/#mountconfig

stapelberg, 2024-07-19:
Okay, we now have support for PARTUUID= and UUID= (file system uuid) matching :)

The only big thing missing is the option parsing now. @vilhalmer did you want to send a PR for it, or should I just go ahead?

vilhalmer, 2024-07-19:
If you're on a roll by all means go for it, but I should be able to work on it this weekend!

vilhalmer, 2024-07-20:
I've got this put together, but I cannot for the life of me figure out how to actually use my local branch with a `replace` directive. It seems like using branch names is no longer supported? I tried [this](https://github.com/gokrazy/gokrazy/issues/164#issuecomment-1407360051) without luck, and haven't been able to guess the right syntax for `go mod edit -replace`. Something like:
```
go mod edit -replace=github.com/gokrazy/gokrazy=/home/vil/code/gokrazy@mount-options
```
Just causes it to try to find a directory including the @:
```
go: github.com/gokrazy/gokrazy@v0.0.0-20240719164423-476f34dea4f4 (replaced by /home/vil/code/gokrazy@mount-options): reading /home/vil/code/gokrazy@mount-options/go.mod: open /home/vil/code/gokrazy@mount-options/go.mod: no such file or directory
```

I'm fairly new to go and have tripped over the module system more than anything so far, can you share how you handle this if things have changed in the last year?

stapelberg, 2024-07-21:
Thanks for your PR!

Regarding the `replace` directive: see https://go.dev/ref/mod#go-mod-file-replace for more details. Specifically this paragraph:

> If the path on the right side of the arrow is an absolute or relative path (beginning with ./ or ../), it is interpreted as the local file path to the replacement module root directory, which must contain a go.mod file. The replacement version must be omitted in this case.

So, yes, as you eventually figured out, when you specify a path, Go uses the files in that path, without any consideration for VCS.

stapelberg, 2024-07-21:
With option support merged as commit https://github.com/gokrazy/gokrazy/commit/75df378463b9f38cdc8f68a10deca774c0470127, this feature is now complete! 🎉

closed 2024-07-21 by stapelberg

vilhalmer, 2024-07-21:
Thanks for your help!