gokrazy/gokrazy #242: How to compile this project on mac

Nuclear6, 2024-01-11:
## Platform
Mac OS M1Pro 16GBRAM

This is a great project. I have a question about how this project is built. When I open this project on my Mac computer, there are so many incompatible errors. What is the reason?
[image]

[image]

stapelberg, 2024-01-11:
You’re browsing code that is written to run on Linux. 

Because you’re using macOS, you’ll need to cross-compile. 

Go uses the GOOS and GOARCH variables for that. 

When using gokrazy, the packer arranges for these variables (and a few build tags) to be set. See: 

https://github.com/gokrazy/tools/blob/80a59f115332a41206487afc6038beaaed48ce70/packer/gotool.go#L23-L68

and:

https://github.com/gokrazy/tools/blob/80a59f115332a41206487afc6038beaaed48ce70/packer/gotool.go#L321-L342

To browse gokrazy’s Linux-specific code, you need to configure your editor to use GOOS=linux. gokrazy builds for GOARCH=arm64 by default, which matches your computer, so you only need to change GOOS, I think.

Nuclear6, 2024-01-12:
Very good, I solved it according to your method. I will learn more about this project later and hope to make some contributions to this project.

closed 2024-01-12 by Nuclear6