gokrazy/gokrazy #243: continuous integration test gap: skip kexec to exercise bootloader

stapelberg, 2024-01-12:
As part of https://github.com/rtr7/router7/issues/83, I noticed that in our continuous integration setup, the MBR bootloader is not actually tested because we’re using kexec to reboot into new kernels on amd64. The result is that a broken bootloader does not block merging the offending change.

I think we should disable kexec in CI, and if we think kexec should be tested in CI, it should be a separate step after the first non-kexec boot.

damdo, 2024-01-12:
+1 one on having two separate tests one with and one without kexec.
LMK if I can help with this.

stapelberg, 2024-01-13:
With the linked commits, kexec should now be skipped in CI.

I re-titled the issue to track adding an explicit kexec test. Not terribly urgent, though.