gokrazy/gokrazy #245: Go web application fails to load template file

flber, 2024-01-25:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior
I'm running a small go web application (published [here](https://git.sr.ht/~flber/gok-upload)) which has one template file. When deployed to gokrazy panics with the error `open index.html: no such file or directory` and the following logs:
```
panic({0x317bc0?, 0x4000069710?})
	/usr/lib/go/src/runtime/panic.go:920 +0x26c
html/template.Must(...)
	/usr/lib/go/src/html/template/template.go:368
main.IndexHandler({0x3f3718, 0x4000156380}, 0x4000064ae8?)
	/home/benh/projects/gok-upload/main.go:20 +0x1ec
net/http.HandlerFunc.ServeHTTP(0x4000064b08?, {0x3f3718?, 0x4000156380?}, 0x2542d8?)
	/usr/lib/go/src/net/http/server.go:2136 +0x38
net/http.(*ServeMux).ServeHTTP(0x658ec0?, {0x3f3718, 0x4000156380}, 0x4000150400)
	/usr/lib/go/src/net/http/server.go:2514 +0x144
net/http.serverHandler.ServeHTTP({0x4000069650?}, {0x3f3718?, 0x4000156380?}, 0x6?)
	/usr/lib/go/src/net/http/server.go:2938 +0xbc
net/http.(*conn).serve(0x400014c7e0, {0x3f3f18, 0x4000068e70})
	/usr/lib/go/src/net/http/server.go:2009 +0x518
created by net/http.(*Server).Serve in goroutine 1
	/usr/lib/go/src/net/http/server.go:3086 +0x4cc
2024/01/25 14:33:57 Index
2024/01/25 14:33:57 http: panic serving 100.66.196.107:54030: open index.html: no such file or directory
```
The layout of the project is very simple:
```
.
├── go.mod
├── index.html
└── main.go
```
Is there a step I'm missing to package static files or something? I'm super interested in this project and would love to learn how to use it more effectively :)

## Expected behavior
When `go install`-ing and running locally, the program has no issues and serves the html correctly.

christian-heusel, 2024-01-26:
This is how you can ship extra files: https://gokrazy.org/userguide/package-config/#extrafiles

flber, 2024-01-26:
Oh perfect, thank you @christian-heusel! The go packaging systems still trips me up sometimes, what *is* the difference between the two situations I was describing without that config step? It doesn't seem like the local version is dependent on that file existing in the fs, and in gokrazy it doesn't panic at compile time?

Also I noticed that I had to delete my program's builddir files in order for it to actually rebuild anything, is that intended behavior? It seems that if a change has been made upstream it should rebuild a package no matter what?

closed 2024-01-26 by flber

stapelberg, 2024-01-26:
> The go packaging systems still trips me up sometimes, what _is_ the difference between the two situations I was describing without that config step? It doesn't seem like the local version is dependent on that file existing in the fs, and in gokrazy it doesn't panic at compile time?

gokrazy will only include the result of `go install`, i.e. an executable program, in the installation.

Any files that your program expects to read from the local file system must be included separately.

This isn’t specific to Go or gokrazy, you will encounter this issue with any program you copy from one computer to another — you need to arrange for the environment to match what your program expects.

An alternative is to embed the files, which will make Go compile them into the executable program: https://pkg.go.dev/embed

> Also I noticed that I had to delete my program's builddir files in order for it to actually rebuild anything, is that intended behavior? It seems that if a change has been made upstream it should rebuild a package no matter what?

This doesn’t sound right to me. I never need to delete the builddir. Changes are picked up using `gok get` to get new upstream versions, or from my local working directory when the replace directive is in use (see also https://gokrazy.org/development/modules/). 

Can you file a separate issue with more details about this, including steps to reproduce?

flber, 2024-01-26:
> An alternative is to embed the files, which will make Go compile them into the executable program: https://pkg.go.dev/embed

Ahhhh, that makes sense! Embedding works perfectly, thank you for the pointer.

> This doesn’t sound right to me. I never need to delete the builddir. Changes are picked up using `gok get` to get new upstream versions, or from my local working directory when the replace directive is in use (see also https://gokrazy.org/development/modules/).

Oh wow, I really don't know how I missed the `gok get` command - yeah that does it! 
Okay thank you so much for the response, that's completely cleared up my misunderstandings.