gokrazy/gokrazy #246: Support installing Intel microcode

FiloSottile, 2024-01-27:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [x] gokrazy/x86-64

## Expected behavior

Regrettably, Intel CPUs constantly need microcode updates to be secure (and sometimes correct!).

These upgrades are distributed as initrd images that need to be loaded before the main system.

It would be nice if gokrazy images natively supported fetching and installing the microcode updates.

stapelberg, 2024-01-27:
> Regrettably, Intel CPUs constantly need microcode updates to be secure (and sometimes correct!).

So do AMD CPUs, no? If we add microcode update support, we should probably ship both. 

Which particular fix are you after, and isn’t that included in your most recent UEFI? I understood most of the CPU issues such that they mostly affect cloud providers (untrusted VMs) or browsers (untrusted JavaScript), but my typical gokrazy usages don’t fall into these categories.

> These upgrades are distributed as initrd images that need to be loaded before the main system.

gokrazy does currently not use an initrd at all. I don’t know if that means that we could just load a microcode initrd easily, or if it would require us to also start using an initrd to begin with (if so, that would be a rather significant change). 

If we need an initrd, I wrote [an initrd generator for distri](https://michael.stapelberg.ch/posts/2020-01-21-initramfs-from-scratch-golang/), which could possibly be reused.

Though, currently I’m not yet sure how important this issue is.

stapelberg, 2024-02-11:
> It's true that the OEM should provide a way to install microcode updates with UEFI updates, but the last time I successfully installed a UEFI update on a 5 years old machine without using Windows and remotely was precisely never, alas.

Huh, that does not match my experience at all. I run my machines exclusively with Linux and have done plenty of UEFI updates by rebooting into UEFI with a USB memory stick attached that contains the update on a FAT partition. This works fine on ASUS mainboards.

On laptops, the ThinkPads and Dells and HPs I’ve used support UEFI capsule updates, so fwupd can be used to update their firmware conveniently from within Linux. 

I’d encourage you to give updating UEFI another try :)

> Actually, microcode can be compiled straight into the kernel, so given the excellent CI of rtr7/kernel, that might be the easiest path. https://www.kernel.org/doc/html/v5.17/x86/microcode.html#builtin-microcode

Thanks for the pointer! Yes, this conceptually sounds easiest, but I have recently tracked down a previously-unknown size limit on kernel images in the MBR bootloader setup that gokrazy currently uses: we’ll need to fix #248 before we can make the kernel substantially larger.