gokrazy/gokrazy #247: WebAuthN authentication

FiloSottile, 2024-01-27:
It would be nice to be able to use FIDO2 tokens (or passkeys) to authenticate with the web interface instead of passwords. It can be made easy to set up with a `gok auth` command that runs the registration flow and adds the necessary data to the config file.

Even CLI authentication could be handled with the WebAuthN protocol, by having the CLI get an authentication response from the FIDO2 token. It might even be able to skip the roundtrip with the server by taking the necessary challenge data from the config.

Related to #225.

stapelberg, 2024-01-27:
Generally I’m a fan of hardware tokens, but my knowledge dates back to when they were still called U2F :) 

I’m a bit wary of the complexity that FIDO/passkey integration entails (and, more generally speaking, of offering more than one authentication mechanism), though maybe I’m not estimating it correctly. 

A (quick & dirty) proof of concept would be helpful if you feel like exploring this further.

stapelberg, 2024-06-05:
> Generally I’m a fan of hardware tokens, but my knowledge dates back to when they were still called U2F :)

By now, I have upgraded my U2F tokens to FIDO2 tokens and set up a few passkeys.

One thing that isn’t quite clear to me yet: The remote origin (e.g. scan2drive.lan) is part of a passkey. When you suggested the `gok auth` flow that adds to `config.json`, where/how does the remote origin get set correctly? Or, asked another way: don’t we need to initiate the passkey flow from the correct remote origin for passkeys to work?

FiloSottile, 2024-06-05:
> The remote origin (e.g. scan2drive.lan) is part of a passkey. When you suggested the `gok auth` flow that adds to `config.json`, where/how does the remote origin get set correctly? Or, asked another way: don’t we need to initiate the passkey flow from the correct remote origin for passkeys to work?

If you generate a passkey from the CLI using the FIDO2 or platform APIs (as opposed from within a web page with WebAuthN), you are as powerful as the browser itself, so you can create it for any origin. You do need to know the origin it will be used from, though, I think.

damdo, 2024-09-06:
+1 for this @FiloSottile :)