gokrazy/gokrazy #255: Documentation how to set sysctl

joneskoo, 2024-03-10:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [x] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

No documentation how to set sysctl.

Using `tailscale up --advertise-routes=192.168.127.0/24`  does not appear to be sufficient (I enabled it using Tailscale SSH).

```
/ #  sysctl -w net.ipv4.ip_forward=1
net.ipv4.ip_forward = 1
/ #  sysctl -w net.ipv6.conf.all.forwarding=1
net.ipv6.conf.all.forwarding = 1
```

After this, routing does work.

## Expected behavior

Documentation how to set sysctls, either generally and/or as part of Tailscale documentation page.

Am I expected to create an app that calls applySysctl like https://github.com/rtr7/router7/blob/95fc74327defd019c54ef4e8f73f21a584ffee3e/internal/netconfig/netconfig.go#L1107-L1126 or should this be built in gokrazy functionality, or part of the tailscale client?

anupcshan, 2024-03-12:
I wrote a [tiny tool](https://github.com/anupcshan/sysctl-apply/blob/main/main.go) to apply sysctls on a Gokrazy instance (to run a Tailscale exit node). Its a very thin wrapper around https://github.com/lorenzosaino/go-sysctl. It would be neat if this could be built directly into Gokrazy.

stapelberg, 2024-03-12:
Applying sysctls via writing text into files (like the linked router7 example) seems fine to me. 

We can think about having gokrazy parse a file and apply the sysctls, but I don’t understand why we’d need the go-sysctl package for that…?

anupcshan, 2024-03-13:
> We can think about having gokrazy parse a file and apply the sysctls, but I don’t understand why we’d need the go-sysctl package for that…?

We don't need that package - its just something I used because it was the most expedient.

stapelberg, 2024-03-13:
> Sysctls in a text files like wifi config makes sense to me. @stapelberg do you mean this should be built in feature then in gokrazy, or does it exist?

It doesn’t exist yet.

> it'd be nice if writing a go module is not required to e.g. use tailscale as router?

Agreed. I actually would have expected Tailscale to configure everything it needs. 

That should work for forwarding (subnet routers), but probably ends at NAT configuration, which is too specific for Tailscale to set up (for exit nodes).

stapelberg, 2024-03-13:
Yeah, creating an issue and/or pull request at Tailscale would be good. It can’t hurt to ask to set the sysctl on all Linux systems, not just gokrazy. In the past, bradfitz has been helpful with such issues, so maybe cc him directly on the issue.

joneskoo, 2024-03-13:
I've created the issue and PR to implement setting sysctl in Tailscale upstream. https://github.com/tailscale/tailscale/pull/11408/

stapelberg, 2025-02-22:
I’ll close this issue as writing sysctls via the OS file API seems good enough.

closed 2025-02-22 by stapelberg