gokrazy/gokrazy #257: Feature request: Use Tailscale for gokrazy API authorization

joneskoo, 2024-04-01:
Currently tailscale uses HTTP basic authentication with username gokrazy and a password to authorize management (e.g. updates) of gokrazy.

The idea proposed here is that we'd use Tailscale in place of username/password for authorization. This would be nicer since browser would not prompt username/password, and more secure as it would prevent password from being phished / exposed via DNS hijacking / public WiFi as gokrazy by default doesn't use TLS. If the only way to access gokrazy is over Tailscale, it is not possible to reach it without being connected through Tailscale. No secrets would be needed.

To get the usability benefits, gokrazy would need to accept access without basic authentication (no username / password). This could be implemented either by running tsnet library and have gokrazy register as a node on Tailscale network (likely Tailscale authorization key for initial connection), or by running tailscale application and using `tailscale serve`.

It would also be possible [to require capabilities in Tailscale ACL, e.g. `gokrazy.org/cap/admin`](https://tailscale.com/kb/1324/acl-grants#app) to do fine grained permissions, although probably most users would be happy with default ACL allowing everything (no capabilities) and it'd still enforce that only access from same Tailnet is allowed.

In summary, options are:

1. Simplest: if user connects via Tailscale, they're implicitly trusted (rely on Tailscale ACL)
2. Intermediate: gokrazy checks that connecting user is explicitly allowed. User owning the connecting device could be identified through [Tailscale WhoIs](https://pkg.go.dev/tailscale.com/client/tailscale#LocalClient.WhoIs).
3. Expert: Rely on Tailscale capabilities to define application level fine-grained permissions

I'd suggest to start with simplest form: "no basic authorization, only listen on Tailscale for management http interface" as that's closest to current basic authentication except better.

stapelberg, 2024-04-02:
Yeah, skipping the basic auth check for requests coming in over the tailnet seems reasonable for personal tailnets. I do wonder if it should be opt-in, though, to not be surprising — if people deploy gokrazy in a company tailnet, they probably don’t want *everyone* in that tailnet to have access without password. 

My general guidelines are that any integration should be done with minimal dependency footprint. 

I think the tailscale client package is relatively light-weight, but maybe we can get somewhere without even pulling in extra dependencies: we already identify addresses as belonging to tailscale in gokrazy/ifaddr/ifaddr.go.

JustinAzoff, 2024-04-06:
If you are using `serve` you don't necessarily need a separate whois call as `serve` can inject the identity headers:

https://tailscale.com/kb/1312/serve#identity-headers

In this sort of deployment there wouldn't be any tailscale specific code in the gocrazy webserver other than a configuration for what header to look for, i.e. `Tailscale-User-Login`

stapelberg, 2025-02-22:
https://github.com/gokrazy/gokrazy/issues/265 is related, and in https://github.com/gokrazy/breakglass/issues/21#issuecomment-2649633830, someone reports they are using that setup to effectively use tailscale for authentication.

Maybe all we need to do is document this better? cc @tho @bradfitz

joneskoo, 2025-02-23:
Tailscale has also been responsive and welcoming to changes in their tailscale cli upstream - we already have special cases for gokrazy. So don't discount that if it simplifies things and can be reasonably justified.

joneskoo, 2025-02-23:
As I see, what we primarily want is authorization: restrict which users can access gokrazy. In my opinion it would make most sense to do this authorization on Tailscale ACL side.

Changes would be needed in:
- gokrazy http endpoint changes? Or auth proxy?
- gok tool to support passwordless
- breakglass
- document how to set this up as example on gokrazy Tailscale page

I think priority should be on safety (mainly misconfiguration potential) and simplicity, but also only have Tailscale specifics in gokrazy if required.

Maybe auth proxy using tailscale.WhoIs is ideal: https://github.com/tailscale/tailscale/blob/f5997b3c570481279990a9dc2cfd75c2f64da63e/cmd/nginx-auth/nginx-auth.go#L52 so you'd run tailscale up, tailscaled and gokrazy-tailscale-auth. That should solve the unix socket problem too, and be safe as it is explicitly configured.

joneskoo, 2025-03-02:
ListenAddress is not the right way to restrict it in case of Tailscale authorization. We need not to tie the listen address but authorized client address. Otherwise e.g. in case of routing, bypass is an easy misconfiguration.

joneskoo, 2025-03-03:
> can you elaborate on the potential routing bypass misconfigurations given how gokrazy is configured? I'm wondering if I'm missing something and if the caddy setup described above is insecure?

This is getting off topic but I should have clarified in the comment.

<detail>
For example, if gokrazy was also a router on WiFi/LAN (think router7) and doing NAT and it was possible to connect to tailscale IP through it from non-Tailscale interface. Depending on routes and firewall rules, a remote host might reach it not via Tailscale. I didn't check if Tailscale explicitly denies this but IMO even that wouldn't make it right.

There could be other attack paths but hopefully this proves the potential to do so which is enough that it's not a solution for this issue (likelihood for misconfiguration).

Binding is useful to run different services on same port, which as you say should be a different issue, and might be valid feature for other use cases if so.
</detail>

stapelberg, 2025-05-30:
Hello everyone, thank you for the continued discussion and sorry that I didn’t have time to participate more.

In the meantime, I have become aware of an exciting new option for how to make available HTTP services on Tailscale: the [tailscale caddy plugin](https://github.com/tailscale/caddy-tailscale)

This plugin makes Caddy tailscale-aware, meaning you can declare (HTTPS) listeners and have Caddy join your tailnet — notably, you can specify as many `bind` directives as you want, meaning Caddy can join your tailnet multiple times. Personally, I use this on my server to easily reverse proxy to each component of the Prometheus stack (prometheus, alertmanager, pushgateway, grafana), but giving each virtual host its own tailnet node, thereby allowing access at https://prometheus.example.ts.net :) 

In the context of gokrazy, this gives us everything we need to make a “tailscale-native experience” possible, which can be set up as follows:

1. Create a new gokrazy instance:

```
% gok -i tail new
% cd ~/gokrazy/tail
```

2. Change `HTTPPassword` in the `Update` section to `"NoPassword": true` like so:

```json
{
    "Hostname": "tail",
    "Update": {
        "NoPassword": true
    },
// …
```

3. Add tailscale-enabled Caddy:

```
% gok add github.com/tailscale/caddy-tailscale/cmd/caddy
% gok add github.com/gokrazy/mkfs
```

(The mkfs package is required because Caddy writes state when obtaining TLS certificates.)

(In case the instance does not reboot after creating the `/perm` partition, try `gok get github.com/gokrazy/mkfs@fbb07f9ec5fd85dbed6db8a3f64ef9c55315cc1d`)

4. Open https://login.tailscale.com/admin/settings/keys and click “Generate auth key”

5. Configure Caddy in `~/gokrazy/tail/config.json` (as documented on https://gokrazy.org/packages/caddy-http-server/):

```json
        "github.com/tailscale/caddy-tailscale/cmd/caddy": {
            "ExtraFilePaths": {
                "/etc/caddy/Caddyfile": "Caddyfile"
            },
            "CommandLineFlags": [
                "run",
                "--config",
                "/etc/caddy/Caddyfile"
            ]
        }
```

With a `Caddyfile` like this:

```caddy
{
  tailscale {
    auth_key tskey-auth-secretsecretredacted-secretsecretredacted
  }
}

https://tail.example.ts.net {
	bind tailscale/tail

	tailscale_auth

	reverse_proxy unix//run/gokrazy-http.sock
}
```

6. Deploy the gokrazy instance (https://gokrazy.org/quickstart/). On first boot, the instance will create the `/perm` file system, reboot, and then join the tailnet (you’ll see it on https://login.tailscale.com/admin/machines)

7. Now you can open https://tail.example.ts.net in your browser to access the gokrazy web interface. The first request will take a while because of the TLS provisioning.

8. In your `~/gokrazy/tail/config.json`, configure the fully qualified domain name in the `Update` section:

```json
{
    "Hostname": "tail",
    "Update": {
	"Hostname": "tail.example.ts.net",
        "NoPassword": true
    },
// …
```

otherwise you will get an error like “checking target partuuid support: Get "https://gokrazy:***@tail/update/features": remote error: tls: internal error.”

9. You can use `gok update` to update the instance as usual.

There are a bunch of nice properties to this setup. For example, you can use Tailscale ACLs to control access to this node.

To serve actual web services, you can add another host and have it join your tailnet under a separate name, thereby allowing you use Tailscale ACLs to share the actual service more widely than admin access to the gokrazy instance.

Let me know what you think / if this works for you. We can get that documented and possibly simplified a little more.

joneskoo, 2025-06-20:
This sounds promising. I have been keeping this in my inbox with intent to try it out.

Before documenting this model, it's important to check no obvious sharp edges result. Have we checked if all gokrazy HTTP endpoints are safe from CSRF? I was just researching as I set up Tailscale auth proxy for another thing and IP based authentication with web is kind of scary. At least (but I'm not expert on this):

1. HTML forms can POST without Javascript, not subject to same origin/Access-Control-Allow-Origin (but not content type application/json?)
2. Browsers do pre-flight requests for GET so if GET can cause state changing action, it'd be a security issue.
3. It might be possible to get information disclosure from same origin rejected requests even if Javascript can't directly read the response, at least timing.

I suspect basic auth would have limited attack surface before.

- [ ] Check for CSRF when disabling basic auth
- [ ] As auth keys have maximum expiration, will described setup need to be updated frequently or does auth key life only affect initial provisioning / adding virtual hosts?

stapelberg, 2025-06-21:
Thanks for flagging this, @joneskoo! 

gokrazy does use XSRF protection (via cookies) for the supervise requests (stop/restart service), but you’re right that we need to ensure all endpoints are protected for the scenario of a tailscale-authenticated gokrazy instance.

Regarding the tailscale auth keys: those are single-use. So once the node has joined your tailnet, the key is automatically invalidated — there’s nothing to manage/update. (But the downside is that for each new node, you need to create an auth key in advance.)

tho, 2026-03-01:
FYI `tailscale serve` can now serve unix sockets (https://github.com/tailscale/tailscale/issues/9771).

I'm successfully using the following config to `gok update` via Tailscale:

```
    "Update": {
        "Hostname": "foo.tailnet.ts.net",
        "HTTPPassword": "redacted"
    },
    "PackageConfig": {
        "github.com/tho/gokrazy-cmdgroup": {
            "CommandLineFlags": [
                "-watch", "1,2",
                "/user/tailscale",
                "--", "up", "--auth-key=file:/etc/tailscale/auth_key", "--ssh",
                "--", "serve", "--http", "80", "unix:///run/gokrazy-http.sock",
                "--", "serve", "--https", "443", "unix:///run/miniflux.sock"
            ],
            "ExtraFilePaths": {
                "/etc/tailscale/auth_key": "tailscale.auth_key"
            }
        }
    }
```

The `gokrazy-cmdgroup` package is not required. I'm just using it to configure Tailscale in one go and start two serves.