gokrazy/gokrazy #260: Feature Request: Provide /sbin/modprobe implementation

dev-zero, 2024-05-17:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [x] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I am trying to get an arm64 RPi kernel to run (basically, I need hifiberry-dac, but that patch seems to require a lot of other things, and also adds support for newer sdxc cards) and while trying to load modules from the serial-busybox, I get this:

```
/lib/modules/6.8.9-v8 # modprobe btbcm
[34646.470585] Invalid ELF header magic: != ELF
```

Checking I see that `/sbin/modprobe` is missing which I assume is the likely culprit.
Any plans on providing a simple implementation in `/gokrazy/...`?
There is this: https://pkg.go.dev/github.com/pmorjan/kmod

stapelberg, 2024-05-18:
> I am trying to get an arm64 RPi kernel to run (basically, I need hifiberry-dac, but that patch seems to require a lot of other things, and also adds support for newer sdxc cards) and while trying to [...]

No, I don’t think that’s the culprit: busybox’s modprobe works for me on gokrazy:

```
          model: Raspberry Pi 4 Model B Rev 1.1
          build: 2024-05-18T07:19:34+02:00

/tmp/breakglass3870960151 # lsmod
/tmp/breakglass3870960151 # modprobe btbcm
/tmp/breakglass3870960151 # lsmod
btbcm 24576 0 - Live 0xffffd4ea679e8000
bluetooth 806912 1 btbcm, Live 0xffffd4ea67a0f000
ecdh_generic 16384 1 bluetooth, Live 0xffffd4ea679e1000
ecc 36864 1 ecdh_generic, Live 0xffffd4ea679fd000
/tmp/breakglass3870960151 # 
```

Are you sure that your kernel modules were compiled for the correct architecture? 

> Any plans on providing a simple implementation in `/gokrazy/...`?
> There is this: https://pkg.go.dev/github.com/pmorjan/kmod

Each extra binary we put into /gokrazy should be something that most installations need.

I would rather recommend having a separate package (e.g. github.com/gokrazy/modprobe) which you can then add if you need it. 

But, stepping back, I don’t see the use for an interactive modprobe command? When debugging, you can use busybox’s modprobe. For your actual program, you can use the kmod package as you point out, or just load the modules directly without dependency resolution ([example in gokrazy/wifi](https://github.com/gokrazy/wifi/blob/f611656338b68faa5f42bc2c8bbf1e0231a74a50/wifi.go#L203-L212)). So I’m not sure why a modprobe command would be needed. 

(Side note: I also have a kmod implementation in distri’s minitrd which is only a few lines: https://github.com/distr1/distri/blob/master/cmd/minitrd/kmod.go — but there’s no particular reason to prefer this one over the kmod package you linked to.)

dev-zero, 2024-05-23:
> No, I don’t think that’s the culprit: busybox’s modprobe works for me on gokrazy:
>
> ```
>           model: Raspberry Pi 4 Model B Rev 1.1
>           build: 2024-05-18T07:19:34+02:00 [...]

Ok, based on this I went back to my configuration and it seems that using `.xz` compressed modules breaks things (which I had set due to switching to a RPi kernel).

> Are you sure that your kernel modules were compiled for the correct architecture?
>
>
> Each extra binary we put into /gokrazy should be something that most installations need.
>
> I would rather recommend [...]

An `/sbin/modprobe` would mostly be there for cases where the kernel would automatically load a module via the `request_module` function call.

stapelberg, 2024-05-24:
> Ok, based on this I went back to my configuration and it seems that using `.xz` compressed modules breaks things (which I had set due to switching to a RPi kernel).

Huh, strange. I read through busybox-1.34.1’s source a little bit and found:

* bb_init_module_24 loads the module into memory and uncompresses it if needed: https://git.busybox.net/busybox/tree/modutils/modutils-24.c?h=1_34_1#n3827
* (the mmap code path is not taken because `CONFIG_FEATURE_INSMOD_TRY_MMAP` is not set in our busybox.config)
* the open_zipped function calls open_transformer: https://git.busybox.net/busybox/tree/archival/libarchive/open_transformer.c?h=1_34_1#n329
* the transformer implementation looks for all sorts of compression format magic values: https://git.busybox.net/busybox/tree/archival/libarchive/open_transformer.c?h=1_34_1#n160
* We set `CONFIG_FEATURE_SEAMLESS_XZ=y` in our busybox.config, so xz compression should be supported. 

Is there a bug in busybox maybe? Have you tried with a newer version? Does the Go modprobe work when busybox’s doesn’t?

> An `/sbin/modprobe` would mostly be there for cases where the kernel would automatically load a module via the `request_module` function call.

Right. Is there a specific kernel module you need which requires that, or is it more of a theoretical concern? :) 

To add this feature to gokrazy, we should probably symlink /sbin/modprobe to /gokrazy/init and implement it multi-call binary style (like busybox).

dev-zero, 2024-05-27:
> Huh, strange. I read through busybox-1.34.1’s source a little bit and found:
>
>     * bb_init_module_24 loads the module into memory and uncompresses it if needed: [...]

I would need some more time to test these things out. But since also gokrazy/wifi does not support `.ko.xz` I turned module compression off for now.

> Right. Is there a specific kernel module you need which requires that, or is it more of a theoretical concern? :)
>
> To add this feature to gokrazy, we should probably symlink /sbin/modprobe to [...]

That would make the most sense, yes. At the moment it is more a theoretical concern. Since I want fast boot times in my project, I stripped the kernel down to bare minimum to reduce the code size being loaded before gokrazy runtime. In that process I also moved iptables and IPv6 to `M`.
While being at it, it seems that Tailscale is using the userland wireguard implementation and the kernel module is not needed.