gokrazy/gokrazy #262: Gokrazy under Hyper-V

jshield, 2024-05-27:
Happy to report that Gokrazy boots under Hyper-V (in Azure) without too many modifications:

[image]

documenting a couple of things in the extremely unlikely case that someone else is trying to do the same:

* Added Hyper-V drivers to the rtr7 kernel build and removed unnecessary drivers from config.addendum.txt, further actions could be switching it out for either the ubuntu azure kernel or the wsl2 kernel as a lot of defconfig is completely unnecessary for hyper-v and azure.
* Bootstrapping is a bit of a challenge, need to convert the raw image to a vhd, create a managed disk for upload, upload via azcopy, then create a V1 generation VM from that, not insurmountable but a bit of a pain for automation, might be suitable to extend gokrazy-machine to natively interact with Azure RM APIs, call qemu-img convert, etc.
* Azure VMs provide a scratch disk that can survive reboots, but not deallocation of the VM, would be good to initialize that as either swap or add a /scratch mount and create an appropriate filesystem on it (possibly as part of an azure management daemon)
* No guest reporting as waagent is a python app, there was a minimal c agent written for openbsd a while back, may be possible to be convert it to golang as a base and atleast report OS metadata to the Azure Management plane.
* Public IP information isn't feed through as Azure doesn't directly attach a public network interface by default, rather it gets routed via the underlying Virtual Network.

stapelberg, 2024-05-27:
Thanks for the report!

I was thinking about providing a generic x86 kernel (independent of router7) with VM-related config options enabled by default for common VM providers. We had someone who recently ran gokrazy on Amazon EC2, for example.

Regarding guest reporting, be sure to have a look at https://github.com/bradfitz/qemu-guest-kragent for QEMU-based virtualization. Maybe it’s helpful as a starting point for other providers, too.

bradfitz, 2024-06-02:
> in the extremely unlikely case that someone else is trying to do the same

Oh, hi! I'm also working on cloud-ifying (AWS, GCP, Azure) Gokrazy appliance images.

I have a tool to build AMIs but hadn't gotten to GCP and Azure yet, so thanks!