gokrazy/gokrazy #263: Tracking issue: Provide generic x86 kernel

stapelberg, 2024-05-27:
Currently, for people who want to use gokrazy on x86, they need to use https://github.com/rtr7/kernel

It would be good to offer a generic x86 kernel independent of router7. Notably, said generic kernel should have all VM-related options enabled by default so that it can be used on Proxmox, Amazon EC2, Microsoft Hyper-V, etc.

* [ ] update kernel config (based on router7 config?)
* [ ] create a separate repository
* [ ] integrate into Continuous Integration

bradfitz, 2024-06-02:
Thanks for filing!

Not just x86, though. We also want to distribute AWS AMI images for their AWS Graviton (arm64) machines.

This is the kernel config we're currently using for our Tailscale-on-AWS-with-Gokrazy prototype: https://github.com/tailscale/gokrazy-kernel/blob/main/cmd/rtr7-build-kernel/config.addendum.txt

stapelberg, 2024-06-03:
Thanks, I updated the first comment to point this out.

bradfitz, 2024-06-07:
Have you ever considered letting a kernel package supply kernels for multiple GOARCHes?

The only way our arm64 and amd64 appliance configs would differ is their KernelPackage lines.

stapelberg, 2024-06-08:
It’s certainly a possibility. The extrafiles mechanism already looks for architecture-specific files, and repositories like gokrazy/mkfs provide binaries for a number of architectures.

However, what I don’t like about this approach is that the module will grow significantly in size. The kernel is already our largest module.

If you’re after using the same configuration file for arm64 and amd64, I would recommend config templating as described at https://gokrazy.org/userguide/automation/#make-config-templating

stapelberg, 2024-08-01:
I created https://github.com/gokrazy/kernel.amd64 with a generic amd64 (x86_64) build. The config is the router7 config for now, but we are free to modify it as we see fit, e.g. move more functionality into modules, enable additional settings.

Feel free to give it a try and report back. I have tested this kernel build only on qemu so far, not real hardware. 

The build process is a little different from the existing kernel repositories in that the build tool no longer lives in the repository itself, but in github.com/gokrazy/autoupdate. See the README.md for instructions.

damdo, 2024-08-01:
Awesome! Thanks Michael :)

closed 2024-08-02 by stapelberg

damdo, 2024-08-02:
Looking at this: https://github.com/gokrazy/kernel.arm64?tab=readme-ov-file#gokrazy-kernel-repository-map

@stapelberg are we planning to never support upstream kernel support for pi5 and greater?
Or do you foresee https://github.com/gokrazy/kernel to also work for pi5+ in the future? Thanks!

stapelberg, 2024-08-02:
I’d love for the upstream kernel to work on the Pi 5, but I have no hope of it happening any time soon. See https://github.com/gokrazy/gokrazy/issues/223#issuecomment-1993707270 for more details.

damdo, 2024-08-04:
I see, thanks! Let's hope they'll somehow manage to in a not too distant future!