gokrazy/gokrazy #264: Tracking issue: Provide Raspberry Pi kernel

stapelberg, 2024-05-27:
It seems like support for the Raspberry Pi 5 won’t land in the upstream linux kernel anytime soon unfortunately, so there’s only one available kernel for that platform :-/

I’m considering creating a new kernel repository for the raspberry pi kernel.

* [ ] create a separate repository (maybe github.com/gokrazy/kernel.rpi?)
* [ ] integrate into Continuous Integration

dev-zero, 2024-05-27:
I would very much welcome this. Also, many of the RPi 3+ features will take time to land in the upstream kernel, such as the recent DMA work, proper sdxc support.
Others, like the GPIO DebugFS (required for periph.io to setup an EdgeDetection GPIO pin via its default mechanism) will be unlikely to make it. On the other hand, the latter can (depending on the device) be worked around by using a polled IRQ Pin like so (for an MFRC522 RFID reader):

```
    var gpioIrqPin gpio.PinIO = gpioreg.ByName(irqPin)
    gpioIrqPin = gpioutil.PollEdge(gpioIrqPin, 100*physic.KiloHertz)
    rfid, _ := mfrc522.NewSPI(p, gpioResetPin, gpioIrqPin)
 ```
 
 (yes, I plan to submit a PR to update https://gokrazy.org/development/gpio/)
 
 In my fork of gokrazy/kernel I have this changeset: https://github.com/dev-zero/gokrazy-kernel/commit/729512f6159e5a0f9d3beb05a6e1cf8373e0b5a1
 So, rather than using the current mechanism, I switched to a `defconfig` mechanism, based on the original RPi kernel. The actual config is tuned to my needs, though. The advantage over the `addendum` method is that the config can be diffed using `diffconfig`. This could then also be generalized a bit more, to permit specifying it on the `gokr-rebuild-kernel` binary, alongside with a config override of a device-tree copying map? On the other hand, I was positively surprised how easily scriptable Go and the current codebase is.

stapelberg, 2024-05-27:
Thanks for your message! It’s good to see what changes you made.

> So, rather than using the current mechanism, I switched to a `defconfig` mechanism, based on the original RPi kernel. The actual config is tuned to my needs, though. The advantage over the `addendum` method is that the config can be diffed using `diffconfig`.

I don’t quite follow. Why is defconfig better for diffing? The addendum approach can also be used for diffing. Do you mean that there is an extra step (producing .config) involved?

What I dislike about creating a defconfig file is that changes need to be synchronized from the default defconfig file. With the current setup, the addendum is always based on the latest defconfig.

stapelberg, 2024-05-27:
I created https://github.com/gokrazy/kernel.rpi and pushed a first build. It boots up for me on the Pi 4 and Pi 5.

Next steps are listed in the first comment (CI + upstream watch automation).

dev-zero, 2024-05-27:
> Thanks for your message! It’s good to see what changes you made.
>
>
> I don’t quite follow. Why is defconfig better for diffing? The addendum approach can also be used for diffing. Do you mean that [...]

Yes, exactly that. To compare I would always need the final `.config`.

> What I dislike about creating a defconfig file is that changes need to be synchronized from the default defconfig file. With the current setup, the addendum is always based on the latest defconfig.

Agreed, but if the RPi kernel devs are changing their setup, you might suddenly be including things you did not want.
So you may occasionally have to update the config again.

dev-zero, 2024-05-27:
I wonder why in the RPi kernel they are still building those `bcm2837-*.dtb`.
Depending on the overlay being used, these will generate errors on boot since the overlay will be requesting symbols the device tree in those does not provide (which blocked me a fairly long time until I saw that in your mainline kernel you are renaming the files).

bradfitz, 2024-06-02:
> I created https://github.com/gokrazy/kernel.rpi and pushed a first build. It boots up for me on the Pi 4 and Pi 5.

Oooh, this might means I can use it on my the Pi 400 jukebox for my kids now! (#188)

stapelberg, 2024-06-23:
Okay, the last remaining items (overlays/ conflict and README) are now also done. I’ll close this issue and if anything else pops up, we can track that in a separate issue.

closed 2024-06-23 by stapelberg