gokrazy/gokrazy #265: Passwordless mode for public images?

bradfitz, 2024-06-02:
I'm working on making Gokrazy based appliance images of Tailscale for public use by anybody who wants to run them.

The bootstrapping method for configuration will be either:

* serial TUI (AWS makes these easy to access via web + CLI)
* cloud config metadata (EC2 user-data, cloud-init, etc).
* letting people download a bespoke image with secrets baked in (Raspberry Pi case, mostly, for optional wifi password, if they don't want to edit a file on a partition)

Whichever method people choose, none involve a well-known password shared by all AWS users.

So we want to remove the `gokr-pw.txt` from the filesystem entirely and have a new "password-less" mode where all web access can only come from localhost. Then userspace programs can tunnel in to localhost for debugging, if users want to see it. Auto updates will also happen from userspace to localhost.

Is this a reasonable feature request? Or is there something else we should be doing instead?

stapelberg, 2024-06-05:
Thanks for filing this issue.

Issue https://github.com/gokrazy/gokrazy/issues/257 (tailscale auth) is related.

Design doc https://docs.google.com/document/d/1a7TXSxNgdR9pkz2KdZxG_gF01DTkneNwtDxLDXh8tkM/edit?usp=sharing (image daemon) is related: that design suggests generating an instance password at customization-time.

In general, I think it’s a reasonable idea to offer other authentication mechanisms aside from passwords.

One open question for me is whether HTTP over localhost is sufficient for authentication. I could imagine security issues in programs resulting in an attacker being able to send requests to localhost. Would it be better (if less convenient) to use UNIX domain sockets? 

Note that there are a few places in gokrazy that read the password file to send API requests to localhost. These would need to be updated (and probably changed from ad-hoc API requests to a central package like https://github.com/gokrazy/gokapi while at it).

bradfitz, 2024-06-06:
> Would it be better (if less convenient) to use UNIX domain sockets?

I'm fine with that. That's what Tailscale does itself on unix systems.

I can send that as a separate change?

stapelberg, 2024-06-07:
> I can send that as a separate change?

Sounds great! I think a PR is the best way to figure out the details :)