gokrazy/gokrazy #267: NVMe boot support (with M.2 hat on Pi 5)

stapelberg, 2024-06-06:
There now is an official M.2 hat for the Pi 5: https://www.raspberrypi.com/products/m2-hat-plus/

Unfortunately, the Pi 5 does not boot off of the NVMe by default. Instead, the EEPROM needs to be updated to tell the boot loader to boot from NVMe: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#nvme-ssd-boot

I successfully booted gokrazy off of an NVMe SSD by:
1. Booting Raspberry Pi OS, running sudo raspi-config, enabling NVMe boot
2. Copying the SD card to NVMe: `dd if=/dev/mmcblk0 of=/dev/nvme0n1 bs=5M`

We should figure out if there is a better way. Maybe gokrazy’s firmware management code could support the configuration that Raspberry Pi OS writes? Here is the log of what raspi-config printed:

<details>
<summary>raspi-config EEPROM update log</summary>

```
Updating bootloader EEPROM
 image: /lib/firmware/raspberrypi/bootloader-2712//default/pieeprom-2024-02-16.bin
config_src: /tmp/tmp.hu1IXBMwWE
config: /tmp/tmp.hu1IXBMwWE
################################################################################
[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf146

################################################################################

*** To cancel this update run 'sudo rpi-eeprom-update -r' ***

*** CREATED UPDATE /tmp/tmpuy3k3_vp/pieeprom.upd  ***

   CURRENT: Fri Feb 16 03:28:41 PM UTC 2024 (1708097321)
    UPDATE: Fri Feb 16 03:28:41 PM UTC 2024 (1708097321)
    BOOTFS: /boot/firmware
'/tmp/tmp.brlM6oWJz5' -> '/boot/firmware/pieeprom.upd'

UPDATING bootloader.

*** WARNING: Do not disconnect the power until the update is complete ***
If a problem occurs then the Raspberry Pi Imager may be used to create
a bootloader rescue SD card image which restores the default bootloader image.

flashrom -p linux_spi:dev=/dev/spidev10.0,spispeed=16000 -w /boot/firmware/pieeprom.upd
UPDATE SUCCESSFUL

```

</details>

BTW, updating a gokrazy instance on NVMe is pretty fast:

```
[…]
Transferred root file system (163 MiB) at 85.77 MiB/s (total: 2s)/s                 
Transferred boot file system (64 MiB) at 85.20 MiB/s (total: 1s)s                 
Triggering reboot
[…]
```

The total update time is about 30 seconds with a Pi 5.

stapelberg, 2025-11-16:
https://gokrazy.org/userguide/nvme-disk/ now shows how to use NVMe disks with gokrazy 🥳

One remaining improvement is to provide some way of applying an updated partition table on the device, which would make Step 3. Option B a lot more useful.

lurch, 2025-11-16:
> One remaining improvement is to provide some way of applying an updated partition table on the device, which would make Step 3. Option B a lot more useful.

I know nothing about how "gokrazy" works (I've never used it, and I've never programmed anything in Go either), but on a "normal Linux distro" I'd used `parted` for that.

Also, as well as the NVMe drive being bigger than the SD card (which will be the common scenario), meaning that there's "unused space" on the NVMe drive; I guess a more serious problem might occur if the SD card is larger than the NVMe drive - I dunno if the filesystem might try writing to sectors beyond the end of the NVMe drive? 😱  🤔