gokrazy/gokrazy #269: gokrazy /perm filesystem corrupted after power outage

joneskoo, 2024-06-21:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [x] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/rpi5b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

After power outage, ext4 /perm is corrupted. I see errors on console:

```
EXT4-fs error (device mmcblk0p4): ext4_lookup:1855: inode #12: comm init: iget: checksum invalid
```

## Expected behavior

When filesystem becomes corrupted, it would be nice if it was possible to recover from it somehow. Maybe automatically?

I'd wish there was a way to clear /perm partition . Now mkfs sees that it's mounted and does nothing.

```
1970/01/01 02:00:03 gokrazy: creating $HOME: mkdir /perm/home: bad message
1970/01/01 02:00:03 gokrazy: attempt 0, starting ["/user/mkfs"]
1970/01/01 02:00:05 Found mountpoint "/"
1970/01/01 02:00:05 Found mountpoint "/dev"
1970/01/01 02:00:05 Found mountpoint "/tmp"
1970/01/01 02:00:05 Found mountpoint "/dev/pts"
1970/01/01 02:00:05 Found mountpoint "/dev/shm"
1970/01/01 02:00:05 Found mountpoint "/run"
1970/01/01 02:00:05 Found mountpoint "/proc"
1970/01/01 02:00:05 Found mountpoint "/sys"
1970/01/01 02:00:05 Found mountpoint "/perm"
1970/01/01 02:00:05 /perm file system already mounted, nothing to do
1970/01/01 02:00:05 gokrazy: process should not be supervised, stopping
1970/01/01 02:00:05 gokrazy: exit status 125
```

Break glass is not usable

```
2024/06/21 22:05:07 gokrazy: exit status 1
2024/06/21 22:05:08 gokrazy: creating $HOME: mkdir /perm/home: bad message
2024/06/21 22:05:08 gokrazy: attempt 526, starting ["/user/breakglass"]
2024/06/21 22:05:08 breakglass.go:203: see https://github.com/gokrazy/breakglass#installation
2024/06/21 22:05:08 breakglass.go:205: open /perm/breakglass.authorized_keys: no such file or directory
2024/06/21 22:05:08 gokrazy: exit status 1
```

I know I can disconnect the SD card and clear the filesystem but for robustness it'd be nicer still if I didn't have to.

Is it too dangerous to clear /perm if it gets corrupted, and would an API for clearing it be appropriate?

I wonder if /perm could use safer mount options to avoid corruption? I suppose fsck is not possible because of C dependency.

joneskoo, 2024-06-21:
I hacked a version of mkfs and run it with gok run.

```go
func resetPerm() error {
	// Clear /perm by writing zeros on it
	dev := rootdev.Partition(rootdev.Perm)
	log.Printf("Zeroing /perm filesystem on %s", dev)

	// Write 1 MB of zero bytes to the beginning of the partition
	zero1MB := bytes.Repeat([]byte{0}, 1024*1024)
	err := os.WriteFile(dev, zero1MB, fs.FileMode(os.O_WRONLY))
	if err != nil {
		return fmt.Errorf("zeroing /perm: %v", err)
	}

	return nil
}
```

Then I reboot the instance and mkfs made a new filesystem. I solved my problem. Perhaps some approach like this could be made into standard feature. I suppose it'd work fine as a package used with gok run?

stapelberg, 2024-06-29:
Hey @joneskoo, sorry to hear your filesystem got corrupted, that’s never fun :-/ 

> 2024/06/21 22:05:08 breakglass.go:205: open /perm/breakglass.authorized_keys: no such file or directory

I recommend installing the breakglass authorized_keys file to /etc instead of managing it yourself on /perm. That will solve this issue and also write-protect it. The `gok new` command will automatically create such a configuration, but I think your gokrazy instance is a bit older :) 

Regarding fsck: I added it to the mkfs package with commit https://github.com/gokrazy/mkfs/commit/4910d43147002f35811f69ae6257bfcc1b769cb2

In a test inside a virtual machine, I could get fsck to work, but I had to stop all services to be able to `umount /perm` first. In your scenario (with an unclean file system), you wouldn’t need to umount anything, so hopefully the now-included fsck tool can help for future corruption. 

Regarding clearing the file system: if you can login with breakglass or via the serial console you can use `dd` to clear the device: `dd if=/dev/zero of=/dev/mmcblk0p4 bs=1M count=1`

I wouldn’t mind adding the code you used to mkfs, so feel free to send a pull request. It should be gated behind a `-overwrite_current_filesystem` flag and that flag should be automatically set to `false` when started by gokrazy (when the environment variable `GOKRAZY_FIRST_START=1` is set). Otherwise, when users accidentally set the flag in config.json, they will end up with a boot-looping installation.

joneskoo, 2024-08-13:
For me the particular problem was that tailscale needs /perm to come online so a remote appliance is tricky to remotely service.

I think the root cause for my problem was bad SD card. Now a month later I got the appliance back for service (family IT 🙈 ) and it was returning mmc timeouts. Let's hope it was just bad SD card and not actual power failure corruption.

Let's see if someone hits the problem again and it doesn't look like a case of bad SD card and then we can think of ways to make gokrazy more robust. I'll close this - I understood you weren't planning on adding more at this point (fair) and were just waiting for me.

closed 2024-08-13 by joneskoo