gokrazy/gokrazy #270: Support Pine64 Rock64

anupcshan, 2024-06-24:
I have Gokrazy running on a [Pine64 Rock64](https://pine64.org/devices/rock64/) and have been using it for the past several months.

This requires a few changes to the Gokrazy suite of tools:
- Handle firmware larger than 4MiB before the first partition (Rock64 u-boot requires SPL and TPL which brings it closer to 9MiB)
- Bakery support for [different serial console speed](https://github.com/gokrazy/bakery/commit/e83f13eb68fcb67d9a0c20d945e0a1bd26c2540e) and to work around weird network hardware (Pine64 does not actually issue a unique MAC address for each device - this has to be faked in firmware)

stapelberg, 2024-06-29:
Thanks for opening this issue and the deviceconfig PR!

Don’t forget to update https://gokrazy.org/userguide/instance-config/#devicetype for the new `rock64` device type please :)

anupcshan, 2024-07-09:
Rock64 is reasonably supported right now. I'll close this task for now. I don't really have a need to get eMMC working (incomplete subtask) - maybe that will change after https://github.com/gokrazy/gokrazy/issues/236 gets resolved.

closed 2024-07-09 by anupcshan