gokrazy/gokrazy #273: Using i2c with periph.io fails

leosunmo, 2024-07-09:
## Platform
I’m using:
* [ ] gokrazy/rpi3b
* [X] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [ ] gokrazy/rpi5b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

I am trying to read a BME280 over i2c using periph.io packages, basically using the example code from https://periph.io/device/bmxx80/.
I understand that periph.io expects raspberry pi's to run rasbian, so this could obviously cause issues as it's not upstream linux.

My understanding of how these things work is lacking, so I can't really give much detail here, but would really appreciate some pointers on what to investigate next. It could be a wiring error, but it's a pretty simple cuircut and I have 3.3V on the correct wires.

Do I need to enable some stuff in the kernel used by Gokrazy? Is it not possible at all? Any pointers are super appreciated! I would love to avoid running plain Go + raspbian as I love what Gokrazy appliances bring in terms of simplicity!

Running this code:
```go
package main

import (
	"context"
	"fmt"
	"log"
	"os"
	"os/signal"
	"time"

	"periph.io/x/conn/v3/i2c"
	"periph.io/x/conn/v3/i2c/i2creg"
	"periph.io/x/conn/v3/physic"
	"periph.io/x/devices/v3/bmxx80"
	"periph.io/x/host/v3"
)

func main() {
	// Load all the drivers:
	if _, err := host.Init(); err != nil {
		log.Fatal(err)
	}

	refs := i2creg.All()
	if len(refs) == 0 {
		log.Print("No I²C buses available\n")
	}

	ctx, cancel := signal.NotifyContext(context.Background(), os.Interrupt)
	defer cancel()
	// Open a handle to the first available I²C bus:
	bus, err := i2creg.Open("")
	if err != nil {
		log.Fatal(err)
	}
	defer bus.Close()

	// Take readings from the sensor every 2 seconds:
	for {
		select {
		case <-time.After(2 * time.Second):
			err := takeReading(bus)
			if err != nil {
				log.Printf("failed to take reading: %s", err)
			}
		case <-ctx.Done():
			return
		}
	}
}

func takeReading(bus i2c.BusCloser) error {
	// Open a handle to a bme280/bmp280 connected on the I²C bus using default
	// settings:
	dev, err := bmxx80.NewI2C(bus, 0x76, &bmxx80.DefaultOpts)
	if err != nil {
		return fmt.Errorf("failed to open BMxx80: %w", err)
	}
	defer dev.Halt()
	// Read temperature from the sensor:
	var env physic.Env
	if err = dev.Sense(&env); err != nil {
		return fmt.Errorf("failed to read environmental data: %w", err)
	}
	log.Printf("Temp: %8s\tHumidity: %9s\n", env.Temperature, env.Humidity)
	return nil
}

```
## Observed behavior
produces
```
2024/07/09 15:32:10 gokrazy: attempt 19, starting ["/user/moist"]
2024/07/09 15:32:10 No I²C buses available
2024/07/09 15:32:10 i2creg: no bus found; did you forget to call Init()?
2024/07/09 15:32:10 gokrazy: exit status 1
```

## Expected behavior
Expected some kind of output.

leosunmo, 2024-07-09:
I just swapped to 
```
    "KernelPackage": "github.com/gokrazy/kernel",
```
without having seen your comment, and it worked.

I assumed it was on `github.com/gokrazy/kernel.rpi` be default now as per docs?

For completeness, I swapped to kernel `github.com/gokrazy/kernel` and specified `i2creg.Open("1")` (using the default blank `""` doesn't work). 

The output of `i2creg.All()` helped me find the correct bus name for `Open`.

stapelberg, 2024-07-10:
> I assumed it was on `github.com/gokrazy/kernel.rpi` be default now as per docs?

That should be the default, yes. Can you check if your `gok` tool is new enough? What does `gok --version` print?

leosunmo, 2024-07-10:
`https://github.com/gokrazy/tools/commit/05d1f9d1a0ef`

It was using `kernel.rpi`, I could tell from the `gok update` output. So something is presumably missing from `kernel.rpi` for i2c that is available in the old `kernel`.

stapelberg, 2024-07-11:
Ah, interesting, so it’s actually the other way around compared to what I initially expected :)

We currently specify `CONFIG_I2C_BCM2835=y` in kernel.rpi/cmd/gokr-build-kernel/config.addendum.txt, but perhaps the Raspberry Pi kernel needs a different option name (sometimes the upstream driver uses slightly different terminology).

mennis, 2024-07-19:
Looks like it might need to be `CONFIG_I2C_BCM2835=m` in gokrazy/kernel.rpi as it is [here](https://github.com/raspberrypi/linux/blame/8457fd6b2d76cd8c8771f09165ce3734b398bc8c/arch/arm/configs/bcmrpi_defconfig#L678) . 

I'm a little confused about this though. I thought that "m" would induce the kernel build tooling to make this a module where "y" would link it into the kernel.  If that were the case though, "y" would have added i2c into the kernel and I would expect it to be seen in printk output in dmesg.

mutantmonkey, 2024-07-23:
Perhaps `CONFIG_BCM_VCIO=y` is the needed option? It's not present in config.addendum.txt, but seems to be set here: https://github.com/raspberrypi/linux/blame/8457fd6b2d76cd8c8771f09165ce3734b398bc8c/arch/arm/configs/bcmrpi_defconfig#L649

stapelberg, 2024-07-25:
I poked at this some more, but it remains a mystery.

On Raspberry Pi OS, I see that the `i2c_dev` and `i2c_brcmstb` modules are loaded, and `/dev/i2c-11` as well as `/dev/i2c-12` show up. 

I tried enabling all of these options, either built-in or as a module:

```
CONFIG_I2C_CHARDEV=m
CONFIG_I2C_BCM2708=m
CONFIG_I2C_BCM2835=m
CONFIG_I2C_BRCMSTB=m
```

…but to no avail. There must be some other difference we’re not aware of yet.

stapelberg, 2024-07-26:
I noticed that on Raspberry Pi OS, the i2c device nodes disappear with an (almost) empty config.txt and re-appear once I add these lines:

```
dtoverlay=vc4-kms-v3d
max_framebuffers=2
```

Unfortunately, even with these lines, they are missing in gokrazy. 

I’ll need to try to swap out more things (entire kernels, entire kernel configs) to narrow this down further.

stapelberg, 2024-07-30:
Finally figured it out! Raspberry Pi introduced another layer of indirection in the boot process that I wasn’t aware of, the overlay_map.dtb file: https://www.raspberrypi.com/documentation/computers/configuration.html#part2.2.10

This file was not included by our kernel build program (nor the packer), but I have fixes for both ready.

Once the overlay_map is included in the image, the Pi firmware will select the right overlays (in our case, i2c0-pi5.dtbo).

Together with `CONFIG_I2C_BRCMSTB` and `CONFIG_I2C_CHARDEV`, this makes the `/dev/i2c*` device nodes appear.

closed 2024-07-30 by stapelberg

leosunmo, 2024-07-31:
I am struggling to get this to work still. 
The `gok` tool is updated to latest, `https://github.com/gokrazy/tools/commit/9f81add3a91e`.

I had issues upgrading to the latest kernel (couldn't find docs that touched on this, would really love that) but ended up clearing out my `gokrazy/<project>/builddir/github.com/gokrazy/kernel.rpi` directory and letting it pull it again. I now get this output from the `gok update` command:
```
Kernel directory: /home/leo/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20240730193703-e59741899e72
```

Still no joy.

What am I missing?

stapelberg, 2024-07-31:
> I had issues upgrading to the latest kernel (couldn't find docs that touched on this, would really love that) but ended up clearing out my `gokrazy/<project>/builddir/github.com/gokrazy/kernel.rpi` directory and letting it pull it again.

Use `gok get github.com/gokrazy/kernel.rpi`, like for any other package :)

> I am struggling to get this to work still. The `gok` tool is updated to latest, `https://github.com/gokrazy/tools/commit/9f81add3a91e`.
> 
> ```
> Kernel directory: /home/leo/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20240730193703-e59741899e72
> ```

The versions look recent enough.

> What am I missing?

Which BootloaderExtraLines did you configure? Can you share your entire config.json?