gokrazy/gokrazy #276: Kernel Panic after `gok update`

denysvitali, 2024-08-17:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [X] gokrazy/rpi4b
* [ ] gokrazy/rpi5b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

`gok overwrite --full /dev/sda` correctly writes the image to the sdcard.

After gokrazy starts, updating the sdcard OTA results in a kernel panic, because most likely the kernel cannot mount the `root_b` partition:
[image]

After the update, the sdcard looks like this:

```
$ sudo fdisk -l /dev/sda
Disk /dev/sda: 58.24 GiB, 62534975488 bytes, 122138624 sectors
Disk model: Storage Device
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 60C24CC1-F3F9-427A-8199-09F7222A0000

Device       Start       End   Sectors  Size Type
/dev/sda1     8192    212991    204800  100M EFI System
/dev/sda2   212992   1236991   1024000  500M Linux root (ARM-64)
/dev/sda3  1236992   2260991   1024000  500M Linux filesystem
/dev/sda4  2260992 122138589 119877598 57.2G Linux filesystem
```

Where the first 32 bytes of each partition are:

```
$ for i in 1 2 3 4; do echo "sda$i"; sudo head -c 32 /dev/sda$i | xxd; done
sda1
00000000: eb3c 9067 6f6b 7261 7a79 2100 0204 0400  .<.gokrazy!.....
00000010: 0180 0000 00f8 7d00 2000 0400 0100 0000  ......}. .......
sda2
00000000: 6873 7173 2601 0000 a7a3 c066 0000 0200  hsqs&......f....
00000010: 0000 0000 0100 1100 1903 0100 0400 0000  ................
sda3
00000000: a133 00d1 a3a3 00d1 e403 0191 e013 40f9  .3............@.
00000010: 82d1 fad0 4250 0191 8802 00b9 a833 00d1  ....BP.......3..
sda4
00000000: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000010: 0000 0000 0000 0000 0000 0000 0000 0000  ................
````

Surprisingly:

- sda3 (`root_b`?) contains data that doesn't make any sense
- sda4 (`perm`?) is empty, despite I'm using `github.com/gokrazy/mkfs`

## Expected behavior

`gok update` correctly writes the update to the inactive partition (`/dev/sda3`) and the kernel doesn't panic

## Additional information

The log of `gok update` doesn't contain anything odd:

```
Updating https://gokrazy:password:8443
Transferred root file system (79 MiB) at 1.33 MiB/s (total: 1m0s)                 
Transferred boot file system (62 MiB) at 2.49 MiB/s (total: 25s)                    
[update boot file system] 0.00% of 62 MiB, uploading at 0 B/s                 Triggering reboot
Updated, waiting 5m0s for the device to become reachable (cancel with Ctrl-C any time)
```

Running with `--testboot` results in the device booting in the old partition (because the OTA failed?), so it all points out to being a problem in how the partition is flashed on the device.

stapelberg, 2024-08-18:
Hey @denysvitali, thanks for the report and sorry for the trouble.

To clarify, is the problem reproducible for you? I just tried it out and for me, `gok -i gokrazy overwrite --full /dev/sda` works, `gok update` will write to sda3, and the device successfully reboots. Afterwards, I see:

```
/ # for i in 1 2 3 4; do echo "sda$i"; head -c 32 /dev/mmcblk0p$i | xxd; done
sda1
00000000: eb3c 9067 6f6b 7261 7a79 2100 0204 0400  .<.gokrazy!.....
00000010: 0180 0000 00f8 7d00 2000 0400 0100 0000  ......}. .......
sda2
00000000: 6873 7173 bd01 0000 50a7 c166 0000 0200  hsqs....P..f....
00000010: 0000 0000 0100 1100 1903 0100 0400 0000  ................
sda3
00000000: 6873 7173 bd01 0000 b3a7 c166 0000 0200  hsqs.......f....
00000010: 0000 0000 0100 1100 1903 0100 0400 0000  ................
sda4
00000000: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000010: 0000 0000 0000 0000 0000 0000 0000 0000  ................
```

Do you have a serial console connected to the device? Can you supply a full boot log from right after the `overwrite`, ideally including the update and failed boot attempt? 

(I have recently also encountered an issue where the device did not boot into the updated version, but I haven’t been able to track it down — it only seems to happen sometimes. Maybe that problem is the same as what you are seeing.)

To get more debugging information, you could apply the following patch to gokrazy/gokrazy:

```diff
diff --git i/gokrazy.go w/gokrazy.go
index fe506b9..b5e0639 100644
--- i/gokrazy.go
+++ w/gokrazy.go
@@ -178,6 +178,8 @@ func maybeSwitchToInactive() {
                return
        }
 
+       log.Printf("cmdline: %s", string(cmdline))
+
        if !strings.Contains(string(cmdline), "gokrazy.try_boot_inactive=1") {
                return
        }
diff --git i/update.go w/update.go
index 23974de..28689c1 100644
--- i/update.go
+++ w/update.go
@@ -307,10 +307,12 @@ func initUpdate() error {
                w.Header().Set("Content-Type", "application/json")
                w.Write(b)
        })
+       inactiveRootPartition := rootdev.InactiveRootPartition()
+       log.Printf("update handler will be called with inactiveRootPartition = %v", inactiveRootPartition)
        http.HandleFunc("/update/mbr", nonConcurrentUpdateHandler(rootdev.BlockDevice()))
-       http.HandleFunc("/update/root", nonConcurrentUpdateHandler(rootdev.Partition(rootdev.InactiveRootPartition())))
-       http.HandleFunc("/update/switch", nonConcurrentSwitchHandler(rootdev.InactiveRootPartition()))
-       http.HandleFunc("/update/testboot", nonConcurrentTestbootHandler(rootdev.InactiveRootPartition()))
+       http.HandleFunc("/update/root", nonConcurrentUpdateHandler(rootdev.Partition(inactiveRootPartition)))
+       http.HandleFunc("/update/switch", nonConcurrentSwitchHandler(inactiveRootPartition))
+       http.HandleFunc("/update/testboot", nonConcurrentTestbootHandler(inactiveRootPartition))
 
        for _, extraUpdateHandler := range extraUpdateHandlers {
                http.HandleFunc(fmt.Sprintf("/update/device-specific/%s", extraUpdateHandler.name), nonConcurrentLimitedUpdateHandler(extraUpdateHandler.device, extraUpdateHandler.offset, extraUpdateH>
```

…and this one to gokrazy/internal:

```diff
diff --git i/rootdev/rootdev.go w/rootdev/rootdev.go
index 70bfb43..ee731e3 100644
--- i/rootdev/rootdev.go
+++ w/rootdev/rootdev.go
@@ -71,6 +71,7 @@ func findGPTPartUUID(uuid string, offset int) (_ string, partition int, _ error)
                        if strings.ToLower(partUUID) != uuid {
                                continue
                        }
+                       log.Printf("found PARTUUID=%s as devname=%s idx=%d offset=%d", uuid, devname, idx, offset)
                        dev = devname
                        partition = (idx + 1) + offset
                        // TODO: abort early with sentinel error code
```

denysvitali, 2024-08-18:
No worries Michael, I love the project and I'm more than happy to help and fix this :)

I can reproduce the issue 100% of the time. I have never encountered such an issue before: I've used `gok update` for other devices before in the past, but admittedly I don't update them often. 
My impression is that something might have changed in the way gokrazy performs its updates (specifically in getting the active partition?) and therefore the B partition is never touched.

Thank you for the debugging steps, I'll help you track down the issue by applying the suggested patches and providing the serial log (currently it's disabled)!

denysvitali, 2024-08-18:
This is the full kernel boot log when the issue happens:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.31-v8 (gokrazy@docker) (aarch64-linux-gnu-gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Wed Mar  1 20:57:29 UTC 2017
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000002ac00000..0x000000002ebfffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef64d00..0x000000003ef64eff (0 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef64f40..0x000000003ef64f75 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 31 pages/cpu s87144 r8192 d31640 u126976
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:E5:81:D0 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 console=ttyS0,115200 root=PARTUUID=60c24cc1-f3f9-427a-8199-09f7222a0001/PARTNROFF=2 init=/gokrazy/init rootwait panic=10 oops=panic
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7912676K/8245248K available (19904K kernel code, 3414K rwdata, 7772K rodata, 7936K init, 1146K bss, 267036K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 63217 entries in 247 pages
[    0.000000] ftrace: allocated 247 pages with 7 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000353] Console: colour dummy device 80x25
[    0.000368] printk: console [tty1] enabled
[    0.001007] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.001039] pid_max: default: 32768 minimum: 301
[    0.001120] LSM: initializing lsm=capability,integrity
[    0.001342] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[... 312 lines cut ...]
[    4.185457]
[    4.191318] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,3)
[    4.201118] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.6.31-v8 #1
[    4.208621] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[    4.215762] Call trace:
[    4.219474]  dump_backtrace+0xa4/0x130
[... cut ...]
[    4.258615]  kernel_init_freeable+0x374/0x3d8
[    4.264286]  kernel_init+0x2c/0x1f8
[    4.269078]  ret_from_fork+0x10/0x20
[    4.273965] SMP: stopping secondary CPUs
[    4.279209] Kernel Offset: 0x46a508800000 from 0xffff800080000000
[    4.286679] PHYS_OFFSET: 0xffffacf440000000
[    4.292217] CPU features: 0x0,80000201,3c020000,0000421b
[    4.298905] Memory Limit: none
[    4.303318] Rebooting in 10 seconds..
```

I'll now apply the patches to gokrazy so that we can debug this further

denysvitali, 2024-08-18:
I've implemented a `/update/debug` route:

```go
http.HandleFunc("/update/debug", func(w http.ResponseWriter, r *http.Request) {
	w.WriteHeader(http.StatusOK)
	fmt.Fprintf(w, "inactiveRootPartition = %v\n", inactiveRootPartition)
	fmt.Fprintf(w, "blockDevice = %v\n", rootdev.BlockDevice())
	fmt.Fprintf(w, "rootPartition = %v\n", rootdev.Partition(inactiveRootPartition))
	fmt.Fprintf(w, "rootPartitionCmdline = %v\n", rootdev.PartitionCmdline(inactiveRootPartition))
	fmt.Fprintf(w, "partUUID: %v\n", rootdev.PARTUUID())
})
```

and it returns:

```
inactiveRootPartition = 3
blockDevice = /dev/mmcblk0
rootPartition = /dev/mmcblk0p3
rootPartitionCmdline = PARTUUID=60c24cc1-f3f9-427a-8199-09f7222a0001/PARTNROFF=2
partUUID: 60c24cc1-f3f9-427a-8199-09f7222a0001
```

Everything looks correct to me - except that the partition table looks odd (this is directly from the device serial console):

```
Disk /dev/mmcblk0: 58 GB, 62534975488 bytes, 122138624 sectors
7602 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1 *  1023,254,63 1023,254,63       8192     212991     204800  100M  c Win95 FAT32 (LBA)
/dev/mmcblk0p2    1023,254,63 1023,254,63          1       8191       8191 4095K ee EFI GPT

Partition table entries are not in disk order
```

But the partitions are there in the filesystem:

```
/ # ls -la /dev/mmcblk0*
brw-------    1 0        0         179,   0 Jan  1  1970 /dev/mmcblk0
brw-------    1 0        0         179,   1 Jan  1  1970 /dev/mmcblk0p1
brw-------    1 0        0         179,   2 Jan  1  1970 /dev/mmcblk0p2
brw-------    1 0        0         179,   3 Jan  1  1970 /dev/mmcblk0p3
brw-------    1 0        0         179,   4 Aug 18 12:09 /dev/mmcblk0p4
```

The cmdline looks good before a reboot:

```
1970/01/01 01:00:04 gokrazy.go:181: cmdline: console=tty1 console=serial0,115200 root=PARTUUID=60c24cc1-f3f9-427a-8199-09f7222a0001/PARTNROFF=1 init=/gokrazy/c
```

I'm still unable to patch the `internal` package - probably my approach of just adding a `replace` directive in `builddir/github.com/gokrazy/gokrazy/go.mod` and one in the `go.mod` of the replaced `gokrazy` module is not enough: the first approach works for modifying gokrazy - but it still looks like `internal` is fetched from the GitHub repo - a bit confusing.

stapelberg, 2024-08-18:
Thanks for all the extra debugging!

To confirm that the root file system is faulty on the host (before being streamed to the target), can you modify gokrazy/tools like so please?

```diff
--- i/internal/packer/packer.go
+++ w/internal/packer/packer.go
@@ -1782,6 +1782,24 @@ func (pack *Pack) logic(programName string) error {
 
        // Start with the root file system because writing to the non-active
        // partition cannot break the currently running system.
+       {
+               dump, err := os.Create("/tmp/root.dump")
+               if err != nil {
+                       return err
+               }
+               if _, err := io.Copy(dump, rootReader); err != nil {
+                       return err
+               }
+               if err := dump.Close(); err != nil {
+                       return err
+               }
+
+               if _, err := tmpRoot.Seek(0, io.SeekStart); err != nil {
+                       return err
+               }
+               rootReader = tmpRoot
+
+       }
        if err := updateWithProgress(prog, rootReader, target, "root file system", "root"); err != nil {
                return err
        }

```

After the next `gok update`, you should have a `/tmp/root.dump` file — does it look similar to what ends up on the SD card? Does it start with the SquashFS header?

> Everything looks correct to me - except that the partition table looks odd (this is directly from the device serial console):
>
> ```
> Disk /dev/mmcblk0: 58 GB, 62534975488 bytes, 122138624 sectors
> 7602 [...]

Don’t worry about this output, it’s expected — the busybox fdisk only reads the MBR (but not the GPT, like other fdisk implementations might) and finds a protective MBR: https://en.wikipedia.org/wiki/GUID_Partition_Table#Protective_MBR_(LBA_0). We need this scheme to keep the Raspberry Pi boot loader working (which only understands MBR/FAT) alongside UEFI-based systems.

> * gokrazy updater.go blindly accepts anything that we ask to write as a `rootfs` (should we at least check for a magic here so that we prevent these things from happening?)

Yeah, sounds like a good extra check. We can think more about measures like this once we have identified the actual issue :)

denysvitali, 2024-08-18:
> After the next `gok update`, you should have a `/tmp/root.dump` file — does it look similar to what ends up on the SD card? Does it start with the SquashFS header?

Surprisingly... yes!

```
$ xxd root.dump | head -n 5
00000000: 6873 7173 2601 0000 9a68 c266 0000 0200  hsqs&....h.f....
00000010: 0000 0000 0100 1100 1903 0100 0400 0000  ................
00000020: 8b19 0220 0000 0000 183c f704 0000 0000  ... .....<......
00000030: 103c f704 0000 0000 ffff ffff ffff ffff  .<..............
00000040: 92e9 f604 0000 0000 4123 f704 0000 0000  ........A#......
```

and to confirm what I'm sending with the client, I've modified the HTTP client to report the TLS key to a file (so that I can intercept the HTTPS traffic with Wireshark) and... surprise:

```
PUT /update/root HTTP/1.1
Host: my-host:8443
User-Agent: Go-http-client/1.1
Transfer-Encoding: chunked
Authorization: Basic Z29rcmF6eTpodW50ZXIy
X-Gokrazy-Update-Hash: crc32
Accept-Encoding: gzip

8000
hsqs&...xn.f....................... .....<.......;.......................#.......;..............x.,..N.0.@w..I]`p..Ab...4C("...r.G6q{..&._.*u|.
o....&.=R8.~7.?.*.a.....'.c.c..p...x..4..d"....P..N...hbPKf.B.s.MY....;....Ak.W.........M.\......H...>_..5.c.}.:...EE.=.s..........3
.:...,.........GSx.L..N.0.E....3.!).........$...
```

It looks like the updater (client) is working fine, but the updater (server) is misinterpreting the incoming bytes (???).

I'll investigate this side now

stapelberg, 2024-08-19:
Huh. So you’re saying with the debugging patch, the issue is reliably not triggering, but without the patch, it reliably triggers? 

(You might want to clear /dev/mmcblk0p3 using something like `dd if=/dev/zero of=/dev/mmcblk0p3 bs=5M count=50` to overwrite the partition when testing.)

denysvitali, 2024-08-19:
Yes. I want to test more to confirm this, but it looks like it.

Looking at the code (and `git blame`) I don't see why this should have any effect - and if this is really the case I think a stdlib function might contain a bug (?).

Before going down the rabbit hole of the stdlib problem, I would like to fine a way to reliably confirm that I can 100% reproduce the issue without my patch and with my patch.

Just for reference, `gok update` was launched with Go 1.23.0 `linux/arm64` (`go 1.23.0-1` on Arch Linux).

stapelberg, 2024-08-19:
That’s really weird. 

I thought a bit about what could be the cause and originally thought that the various fast paths which `io.Copy` can take (`WriteTo`, `ReadFrom`), but then realized that these shouldn’t run because we pass an `io.TeeReader()` to `streamRequestTo`, which passes through none of these optional interfaces.

I think we can be sure that the right number of bytes (and right contents) are read from the network, otherwise the checksum wouldn’t match — but maybe it would be good to print how many bytes `io.Copy()` copied and the checksum, just in case.

Another theory is that writing to disk somehow fails. Are you sure the SD card is in good health? Do you perhaps have another SD card you could use, just to rule out any hardware issues?

BTW, can you provide your entire config.json? I wonder which kernel package you use. If you use gokrazy/kernel, try gokrazy/kernel.rpi and vice-versa. AFAIR, the upstream kernel uses a different SD card driver compared to the Raspberry Pi kernel, so if it’s a driver problem, we could perhaps narrow it down that way.

denysvitali, 2024-08-19:
My `config.json`:

```json
{
    "Hostname": "sv1-helper",
    "Update": {
        "UseTLS": "self-signed",
        "HTTPPort": "8080",
        "HTTPSPort": "8443",
        "HTTPPassword": "hunter2"
    },
    "Packages": [
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/mkfs",
        "github.com/gokrazy/wifi",
        "github.com/coredhcp/coredhcp/cmds/coredhcp",
        "tailscale.com/cmd/tailscale",
        "tailscale.com/cmd/tailscaled"
    ],
    "PackageConfig": {
        "github.com/caddyserver/caddy/v2/cmd/caddy": {
            "CommandLineFlags": [
                "run",
                "--config",
                "/etc/caddy/Caddyfile"
            ],
            "ExtraFilePaths": {
                "/etc/caddy/Caddyfile": "Caddyfile"
            }
        },
        "github.com/coredhcp/coredhcp/cmds/coredhcp": {
            "CommandLineFlags": [
                "-c",
                "/etc/dhcp/config.yaml"
            ],
            "ExtraFilePaths": {
                "/etc/dhcp/config.yaml": "dhcp/config.yaml"
            }
        },
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "/some/path/gokrazy/sv1-helper/breakglass.authorized_keys"
            }
        },
        "github.com/gokrazy/dhcp": {
            "CommandLineFlags": [
                "-interface=eth0",
                "-static_network_config=/etc/ip/static-eth0.json"
            ],
            "ExtraFilePaths": {
                "/etc/ip/static-eth0.json": "static-eth0.json"
            }
        },
        "github.com/gokrazy/gokrazy/cmd/randomd": {
            "ExtraFileContents": {
                "/etc/machine-id": "d05181b112784e50affc09b0f98a60f2\n"
            }
        },
        "github.com/gokrazy/wifi": {
            "ExtraFilePaths": {
                "/etc/wifi.json": "wifi.json"
            }
        },
        "tailscale.com/cmd/tailscale": {
            "CommandLineFlags": [
                "up"
            ]
        }
    },
    "InternalCompatibilityFlags": {}
}
```

### Retrying

I have now retrying to reproduce the issue, and surprisingly... I'm not able to do so anymore! Neither with my patches applied or removed. It just magically started to work again.

I am not able to reproduce this anymore, even with the original `gok` binary and the original `gokrazy` in the `go.mod`.  **The only thing that I have changed in my setup is that my Raspberry Pi is now powered via a power bank instead of a USB-C power supply.** I'd be impressed if this is the reason, but this might explain the weird behavior.

The kernel being used seems to be `github.com/gokrazy/kernel.rpi@v0.0.0-20240802144420-fd363f1112d3` (from the `Kernel Directory` log line).

stapelberg, 2024-08-20:
> I am not able to reproduce this anymore, even with the original `gok` binary and the original `gokrazy` in the `go.mod`. **The only thing that I have changed in my setup is that my Raspberry Pi is [...]

The Raspberry Pi 4 and 5 are known to be quite picky with regards to power supply. The official advice is to only use the official Pi 4 power supply with the Pi 4, and same for the Pi 5. (This was different with the Pi 3, which I have successfully run with all sorts of PSUs…)

The most common power issue is that too many or too power-hungry peripherals cannot be used, but I wouldn’t be surprised if a power supply situation caused the weird “lost writes” we were seeing. 

> Re-using the same power adapter and triggering a couple of updates seems to work fine. I'm really confused.

It’s weird that this issue just disappeared :-/

So, how should we proceed? Close the issue and re-open if it happens again?

denysvitali, 2024-08-20:
Yes, I think it's fair to close the issue and re-open it if someone else notices the issue as well. I thin I'll make a PR to check for the squashfs header though, at least we're able to fail the update in case such an issue happens again to someone else.

Another small improvement would be to reboot to the previous partition if the boot fails - in my case the Raspberry Pi was just looping forever with the kernel panic.

Thanks for the support so far, and sorry I wasn't able to find the real root cause!

stapelberg, 2024-08-20:
> I think I'll make a PR to check for the squashfs header though, at least we're able to fail the update in case such an issue happens again to someone else.

Though in the end, we concluded the squashfs header was being sent, right? It just didn’t make it to disk.

Either way, the check can’t hurt, so I’m not opposed to adding it. Do you want to send a PR?

> Another small improvement would be to reboot to the previous partition if the boot fails - in my case the Raspberry Pi was just looping forever with the kernel panic.

Yes, that is tracked in https://github.com/gokrazy/gokrazy/issues/75

denysvitali, 2024-08-20:
> Though in the end, we concluded the squashfs header was being sent, right? It just didn’t make it to disk.

Oh yes, but we can still read the header from the mmc and prevent a reboot if we know it's bad - but this would only solve this particular problem and doesn't guarantee that the rest of the disk content makes sense.

> Yes, that is tracked in #75

Awesome, thanks!

stapelberg, 2024-08-20:
> Oh yes, but we can still read the header from the mmc and prevent a reboot if we know it's bad - but this would only solve this particular problem and doesn't guarantee that the rest of the disk content makes sense.

That’s a fair point. I was thinking about whether it would make sense to verify the contents by re-reading them from disk and comparing the checksum, but that’d be pretty slow. Re-reading the header as a quick check seems like a good idea, though.