gokrazy/gokrazy #280: shutdown remotely

BrunoTeixeira1996, 2024-09-22:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [X] gokrazy/rpi4b
* [ ] gokrazy/rpi5b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I am trying to figure out a way of remotely or locally shutdown Gokrazy using ssh.

## Expected behavior

I’m currently using Prometheus and AlertManager to monitor my UPS. If the UPS runs on battery for more than five minutes, I receive an alert on a small Go program (running outside of Gokrazy) that triggers a shutdown of all devices connected to the UPS, including Gokrazy.

Currently, I use SSH to execute a halt command on the UPS connected devices, but this approach would not work for Gokrazy. I’m considering an alternative method: Does Gokrazy allow any POST request to a specific endpoint to gracefully shut down, utilizing basic authentication with a username and password? Is this feasible?

I currently have the following entry in the .ssh/authorized_keys file on my other targets:

```
no-pty,no-X11-forwarding,command="sudo /root/off" ssh-rsa ...
```

And then on `/root` I have the `off` file -  `halt -f -f -p`

stapelberg, 2024-09-23:
> Currently, I use SSH to execute a halt command on the UPS connected devices, but this approach would not work for Gokrazy. I’m considering an alternative method: Does Gokrazy allow any POST request [...]

Yes, there is a /poweroff handler you can use: https://github.com/gokrazy/gokrazy/blob/4015d441026fecc3d24cbdecb54779d3f774d3a0/update.go#L366

(We should really document the update protocol / available API better.)

stapelberg, 2024-09-23:
> So the only way to power off Gokrazy right now is to make a POST request to `/poweroff` handler right?

Yes.

Theoretically, you could also call `unix.Reboot(unix.LINUX_REBOOT_CMD_POWER_OFF)` in your program directly, but then your services will not be stopped in an orderly fashion.

BrunoTeixeira1996, 2024-09-23:
Thank you!
Closed this as I was able to poweroff Gokrazy using the following

```
$ curl -X POST http://<PI>/poweroff -u "gokrazy:<password>" -v

All services killed in 1m40.033807051s
Powering off...
```

closed 2024-09-23 by BrunoTeixeira1996