gokrazy/gokrazy #286: Tailscale authorization link not logged as documented

markdrayton, 2024-12-29:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [x] gokrazy/rpi4b
* [ ] gokrazy/rpi5b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

Following the "interactive authentication" instructions at https://gokrazy.org/packages/tailscale/ does not result in an authorization link being logged as described. However, clicking "run once" on `/user/tailscale` *does* result in the link being printed, so I'm assuming there's a dependency that isn't ready when it first runs. I tried adding `WaitForClock` in case there was some time-related problem but that didn't fix it. My config is otherwise as documented in the link above.

Unfortunately I don't have a lot of time to dig into this at the moment. A minimal workaround would be to document that one might need to click run once to produce a login link. (I assume this issue also prevents the initial authorization when using `--auth-key` but I didn't try to confirm this.)

[image]

## Expected behavior

Link printed as documented.

stapelberg, 2025-01-12:
Thanks for your report, I can reproduce the problem.

Commit https://github.com/tailscale/tailscale/commit/b7e48058c8d243adf1ff687e3e92d3fb02b035ea is the problem (cc @bradfitz ): it unconditionally makes the tailscale CLI exit with exit code 0 on first start on gokrazy, regardless of whether any command-line arguments were specified.

I have sent https://github.com/tailscale/tailscale/pull/14611 to revert that commit.

stapelberg, 2025-01-13:
This is fixed with version v1.79.0-pre.0.20250113192735-64ab0ddff14c, to which the following commands should upgrade:

```
gok get tailscale.com/cmd/tailscale@main
gok get tailscale.com/cmd/tailscaled@main
```

closed 2025-01-13 by stapelberg