gokrazy/gokrazy #287: Including loadable kernel modules slow during compile

andig, 2025-01-13:
## Observed behavior

This step in `gok overwrite` writing to local filesystem is very slow (>1min on Mac M1):

```
Including loadable kernel modules from:
/.../go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20250109091720-7437caa083f3/lib/modules
```

## Expected behavior

Writing to local fs being quick, at least when run twice. I didn't look into the code where it spends the time.

stapelberg, 2025-01-13:
Hey @andig and belated happy new year!

When I run `gok overwrite --root /tmp/root.squashfs --trace_file /tmp/gok.trace`, I get a trace whose user regions look like this (see highlight):

[image]

On my MacBook Air M1 from 2020, the timings are:
* 4.91s build
* 72ms validate
* 750ms findbins
* 382ms findextrafiles
* 312ms buildinit
* 376ms sbom
* 2.7s writeroot

On my Intel Core i9-12900K from 2022, the timings **of a different instance** are:
* 5s build
* 8ms validate
* 453ms findbins
* 253ms findextrafiles
* 360ms buildinit
* 614ms sbom
* 2.6s writeroot

You can see how across machines and gokrazy instances, the relative proportions stay somewhat constant.

Can you capture a trace on your machine and attach it here? You can use `go tool trace` to view it, or https://gotraceui.dev/ (that’s what my screenshot above shows).

Is it by chance the SBOM phase that’s slow on your machine? You can use `gok sbom` to run just that.

andig, 2025-01-14:
Another thing learned, thank you for demonstration of the runtime tracer! Indeed SBOM takes 2min. Not sure if there's anything we could do about that?

stapelberg, 2025-01-14:
2 minutes! O_O

That is definitely much longer than expected. I wonder if the problem is with your hardware (worn out disk?) or software (e.g. an Antivirus program scanning every file access).

Can you send a screenshot of the Activity Monitor while `gok sbom` is taking its 2 minutes? Can you try creating a new user account on your Mac and see if the problem reproduces there (without your user settings)?

andig, 2025-01-14:
Ok, I see:

    gok -i evcc overwrite --root=evcc_0.132.1.image --target_storage_bytes=1258299392

works, but slow. But:

```
❯ gok sbom
/Users/andig/go/pkg/mod/github.com/evcc-io/evcc@v0.0.0-20250113133848-4acb4ccfcae5/main.go:15:13: pattern dist: no matching files found
2025/01/14 09:15:04 [go list -mod=mod -tags gokrazy -f {{ .Dir }} github.com/evcc-io/evcc]: exit status 1
```

Our build requires generated files (i.e. the UI assets) that will only be there in the (current) build folder but not in the Go module. I guess we're hitting the sbom error multiple times and they get swallowed?

If that's the case- how can I make `gok` use only the current folder for the main module? It won't work if it tries to `go build` or `go install` the main module.

stapelberg, 2025-01-14:
Can you please provide full steps to reproduce, i.e. how I can re-create your setup from scratch?

I have tried these steps:

```
% gok -i evcc new
% git clone https://github.com/evcc-io/evcc
% cd evcc
% gok -i evcc add .
% gok -i evcc overwrite --root /tmp/root.squashfs
% time gok -i evcc sbom
gok -i evcc sbom  0,42s user 0,17s system 647% cpu 0,092 total
```

What’s the full output of `gok sbom` on your machine?

naltatis, 2025-01-15:
Running the above commands from @stapelberg works on my machine (MacBook Pro M2 Max) and is not slow.

```
gok -i evcc sbom  1.82s user 1.69s system 171% cpu 2.050 total
```

andig, 2025-01-15:
@stapelberg should I open separate issue for the build process (i.e. how/where can the image build include additional non-go files that are required during build)?

stapelberg, 2025-01-16:
> @stapelberg should I open separate issue for the build process (i.e. how/where can the image build include additional non-go files that are required during build)?

Yes, please keep each issue focused on just one thing.

> Only obvious difference is that I have the assets built locally (i.e. /dist folder populated). After checkout:
> 
> ```
> make install
> make ui
> ```

That does not seem to be it. Even after `make install` and `make ui`, my `gok sbom` runs complete in <1s.

> Can you send a screenshot of the Activity Monitor while `gok sbom` is taking its 2 minutes? Can you try creating a new user account on your Mac and see if the problem reproduces there (without your user settings)?

I would still be curious about the screenshot of Activity Manager I asked for…?

> What’s the full output of `gok sbom` on your machine?

This question is also still pending. Please provide these details, otherwise it’s very frustrating to help.

andig, 2025-01-16:
Sorry, my bad:

[image]

Output:

```json
{
    "sbom_hash": "f68ba568d087e9a56c9d213817960b1064d5ce838294ef16dabaa667fdf55764",
    "sbom": {
        "config_hash": {
            "path": "/Users/andig/gokrazy/evcc/config.json",
            "hash": "400f3b271c233708a04269e28d5af6a6d0722b111801d8a67bd009e2e3e1f77b"
        },
        "go_mod_hashes": [
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/evcc-io/evcc/go.mod",
                "hash": "fa257d2fde8d3b30e32748df73c5c26b468606d016670b2b79614522496aece5"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/breakglass/go.mod",
                "hash": "9c1b3eaf195d331d68ccca1686b6584bbe7e863fc659131d4750bd7520a5a724"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/fbstatus/go.mod",
                "hash": "b6a0bb625015e61bab922a62266d64ee6e2b3a0c495fdcdbe436ef6b4b04ddc1"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/firmware/go.mod",
                "hash": "99847ea561e4c4525b6f584384e892f76bfd6f03e2b5f69448f890b1c1e1a4a3"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/gokrazy/cmd/dhcp/go.mod",
                "hash": "2b4d337a1e23469f788e5a0d0b55bfdf9f42ff25f358772998896b685df4eefb"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/gokrazy/cmd/heartbeat/go.mod",
                "hash": "2b4d337a1e23469f788e5a0d0b55bfdf9f42ff25f358772998896b685df4eefb"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/gokrazy/cmd/ntp/go.mod",
                "hash": "2b4d337a1e23469f788e5a0d0b55bfdf9f42ff25f358772998896b685df4eefb"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/gokrazy/cmd/randomd/go.mod",
                "hash": "2b4d337a1e23469f788e5a0d0b55bfdf9f42ff25f358772998896b685df4eefb"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/gokrazy/go.mod",
                "hash": "2b4d337a1e23469f788e5a0d0b55bfdf9f42ff25f358772998896b685df4eefb"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/hello/go.mod",
                "hash": "d2200f4c6014b43d880affed964b5eaa466975c57b60d84e8748f76815304092"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/kernel.rpi/go.mod",
                "hash": "b53a993d37a1ee7646616a737d73974bf74f1afa5b20b21c24f348d10da64aa1"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/rpi-eeprom/go.mod",
                "hash": "4eff2643d034d1bf87c99d0c4b34ce131b4a4c35658acf28c1f27989d1098f7f"
            },
            {
                "path": "/Users/andig/gokrazy/evcc/builddir/github.com/gokrazy/serial-busybox/go.mod",
                "hash": "976a97e6655a51d2e7cd71d385d010e2deec3dd2fc81a6f3e2975ac2cade60d7"
            },
            {
                "path": "/Users/andig/htdocs/evcc",
                "hash": "qh:DfQnqrINOvkBvEJmDCULDw=="
            }
        ],
        "extra_file_hashes": [
            {
                "path": "/Users/andig/gokrazy/evcc/breakglass.authorized_keys",
                "hash": "8829192b1854ee0dbc940ed46f501453e0c0391a1cdd02705d80dbeb7ba0bbbc"
            }
        ]
    }
}
```

`gok sbom` tool 2min again.

andig, 2025-01-17:
Think we can close here. Seems av is the culprit. Thank you!

closed 2025-01-17 by andig

stapelberg, 2025-01-18:
Wow, I didn’t even know that Microsoft made a version of Windows Defender for Mac.

In their [“Troubleshoot performance issues”](https://learn.microsoft.com/en-us/defender-endpoint/mac-support-perf) page, Microsoft recommends [configuring an exclusion](https://learn.microsoft.com/en-us/defender-endpoint/mac-exclusions) for paths that are costly to scan. 

In your case, it sounds like adding an exclusion for paths related to evcc development would improve your life.

---

Looking at your `gok sbom` output, I can see:

```
            {
                "path": "/Users/andig/htdocs/evcc",
                "hash": "qh:DfQnqrINOvkBvEJmDCULDw=="
            }
```

…which does not show up in my setup, so you were not following my steps to reproduce (and you still did not provide your steps to reproduce). 

Maybe `~/htdocs/evcc` contains a lot of files? I’m not sure why/how you are pulling it into your gokrazy instance.

(Based on the `qh:` prefix, you can tell which directories `gok` needed to hash, as opposed to where `gok` could just read the `go.mod` file, which is much quicker.)