gokrazy/gokrazy #288: DHCP when no wired connection

mpl, 2025-01-14:
## Platform

I’m using:

* [ ] gokrazy/rpi3b
* [ ] gokrazy/rpi3b+
* [ ] gokrazy/rpi4b
* [x] gokrazy/rpi5b
* [ ] gokrazy/apu2c4
* [ ] gokrazy/x86-64

## Observed behavior

I do not use any wired connection, and I was observing errors in the DHCP service logs

```
2025/01/14 14:45:40 dhcp.go:430: read packet 2c:cf:67:26:3f:0a: i/o timeout
2025/01/14 14:45:40 gokrazy: exit status 1
2025/01/14 14:45:41 gokrazy: attempt 4338, starting ["/gokrazy/dhcp"]
2025/01/14 14:45:41 dhcp.go:299: waiting indefinitely for eth0 to appear
2025/01/14 14:45:43 dhcp.go:282: lost carrier on interface eth0, de-prioritizing routes
```

And when reading the docs at https://gokrazy.org/userguide/dhcp/

"If you configure gokrazy to [connect to a WiFi network](https://gokrazy.org/userguide/wifi/), the gokrazy/wifi package will [run another instance](https://github.com/gokrazy/wifi/blob/320785ba3e91dac849e28d888c1cf6e6568320d0/wifi.go#L83) of the gokrazy DHCP client with the -interface=wlan0 flag set."

So I figured I might as well remove the default DHCP package, since I have no use for it. Which I did with:

```
	"GokrazyPackages": [
		"github.com/gokrazy/gokrazy/cmd/ntp",
		"github.com/gokrazy/gokrazy/cmd/randomd",
		"github.com/gokrazy/gokrazy/cmd/heartbeat"
    ],
```

## Expected behavior

Now at the moment my wifi connection seems to still be working, however I'm observing that in the logs of /user/wifi:

```
2025/01/14 15:56:20 connected to 70:fc:8f:3c:c5:9e for 11m41s, signal -29
2025/01/14 15:56:20 starting [/gokrazy/dhcp -interface=wlan0]
2025/01/14 15:56:20 dhcp process failed: exec: not started
```

So should I re-enable the gokrazy/cmd/dhcp package then, or is the above log harmless?

stapelberg, 2025-01-14:
> Now at the moment my wifi connection seems to still be working

This confuses me. Do you not need DHCP for your wifi connection to work? How does your device get an IP address? Are you using IPv6 with stateless auto-config?

The wifi program starts /gokrazy/dhcp, so if you remove github.com/gokrazy/gokrazy/cmd/dhcp from your gokrazy instance, that means the wifi program can no longer start a DHCP client and configure your IP address via DHCP.

If you do not want to start dhcp, but still include it in your system and make the wifi program start it, you can set https://gokrazy.org/userguide/instance-config/#packagedontstart for the dhcp package.

mpl, 2025-01-14:
> This confuses me. Do you not need DHCP for your wifi connection to work? How does your device get an IP address? Are you using IPv6 with stateless auto-config?
>
> The wifi program starts [...]

Ok, I get it now, thanks.
As I said, the phrase in the user guide led me to believe that /user/wifi would somehow take care of importing and using cmd/dhcp all by itself.

> If you do not want to start dhcp, but still include it in your system and make the wifi program start it, you can set https://gokrazy.org/userguide/instance-config/#packagedontstart for the dhcp package.

Ok. To be clear, what I would like to avoid is to start dhcp for eth0, since I have no need for it. And there's no point in starting it manually for wlan0 either, because /user/wifi will do that by itself, right?
If I set DontStart on cmd/dhcp, will that interfere with /user/wifi needing dhcp?

stapelberg, 2025-01-15:
> As I said, the phrase in the user guide led me to believe that /user/wifi would somehow take care of importing and using cmd/dhcp all by itself.

I’m not sure why you read the sentence like that. The user guide says “another instance” (so there must be an instance to begin with), and even links to the code in case there are any remaining doubts. If you have any suggestions for how to rephrase it to be more clear, let me know.

> Ok. To be clear, what I would like to avoid is to start dhcp for eth0, since I have no need for it.

Yes.

> And there's no point in starting it manually for wlan0 either, because /user/wifi will do that by itself, right?

Yes.

> If I set DontStart on cmd/dhcp, will that interfere with /user/wifi needing dhcp?

No.

closed 2025-01-15 by stapelberg