gokrazy/gokrazy #289: gok overwrite doesn't pick up extra files

andig, 2025-01-19:
I'm working on https://github.com/evcc-io/evcc/pull/18209, trying to create a build process that does not touch user's instances by using a different `--parent_dir`. `config.json` is generated in `packaging/gokrazy/evcc`:

```json
{
    "Hostname": "evcc",
    "Update": {
        "HTTPPort": "8080",
        "HTTPPassword": "HjJv6aBX51G4wT8EueCc"
    },
    "Packages": [
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/mkfs",
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/wifi",
        "github.com/evcc-io/evcc"
    ],
    "PackageConfig": {
        "github.com/evcc-io/evcc": {
            "GoBuildFlags": [
                "-tags=release -trimpath -ldflags='-X github.com/evcc-io/evcc/server.Version=0.132.1 -X github.com/evcc-io/evcc/server.Commit=3dba9ab26 -s -w'"
            ],
            "Environment": [
                "EVCC_NETWORK_PORT=80",
                "EVCC_DATABASE_DSN=/perm/evcc.db"
            ]
        },
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys",
                "-forward=private-network"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "breakglass.authorized_keys"
            }
        },
        "tailscale.com/cmd/tailscale": {
            "CommandLineFlags": [
                "up",
                "--ssh=true"
            ]
        }
    },
    "SerialConsole": "disabled",
    "InternalCompatibilityFlags": {}
}
```

`breakglass.authorized_keys` is in the same folder:

```
❯ ls packaging/gokrazy/evcc
breakglass.authorized_keys builddir                   config.json
```

Yet build doesn't find it:

```
❯ make gokrazy

sed 's!"GoBuildFlags": null!"GoBuildFlags": ["-tags=release -trimpath -ldflags='"'"'-X github.com/evcc-io/evcc/server.Version=0.132.1 -X github.com/evcc-io/evcc/server.Commit=3dba9ab26 -s -w'"'"'"]!g' packaging/gokrazy/config.tmpl.json > packaging/gokrazy/evcc/config.json
gok -i evcc --parent_dir packaging/gokrazy add .
2025/01/19 14:29:18 Adding the following package to gokrazy instance "evcc":
  Go package  : github.com/evcc-io/evcc
  in Go module: github.com/evcc-io/evcc
  in local dir: /Users/andig/htdocs/evcc
2025/01/19 14:29:18 Adding replace directive to existing go.mod
2025/01/19 14:29:18 Adding package to gokrazy config
2025/01/19 14:29:18 All done! Next, use 'gok overwrite' (first deployment), 'gok update' (following deployments) or 'gok run' (run on running instance temporarily)
# gok -i evcc --parent_dir packaging/gokrazy add tailscale.com/cmd/tailscaled
# gok -i evcc --parent_dir packaging/gokrazy add tailscale.com/cmd/tailscale
gok -i evcc --parent_dir packaging/gokrazy overwrite --full=evcc_0.132.1.img --target_storage_bytes=1258299392
gokrazy gok gca8464 on GOARCH=arm64 GOOS=darwin

Build target: CGO_ENABLED=0 GOARCH=arm64 GOOS=linux
Build timestamp: 2025-01-19T14:29:18+01:00
Loading system CA certificates from bundled Mozilla CA list
Building 6 Go packages:

  github.com/gokrazy/breakglass
    will be started with command-line flags
      from packaging/gokrazy/evcc/config.json
      last modified: 2025-01-19T14:29:18+01:00 (0s ago)

  github.com/gokrazy/mkfs

  github.com/gokrazy/fbstatus

  github.com/gokrazy/serial-busybox

  github.com/gokrazy/wifi

  github.com/evcc-io/evcc
    will be compiled with build flags
      from packaging/gokrazy/evcc/config.json
      last modified: 2025-01-19T14:29:18+01:00 (0s ago)
    will be started with environment variables
      from packaging/gokrazy/evcc/config.json
      last modified: 2025-01-19T14:29:18+01:00 (0s ago)

[done] in 7.31s                        

Including extra files for Go packages:

  github.com/gokrazy/breakglass
    will include extra files in the root file system
      from breakglass.authorized_keys
      last modified: 2025-01-19T14:21:54+01:00 (7m32s ago)

  github.com/gokrazy/mkfs
    will include extra files in the root file system
      from /Users/andig/go/pkg/mod/github.com/gokrazy/mkfs@v0.0.0-20240629161431-4910d4314700/_gokrazy/extrafiles_arm64.tar
      last modified: 2024-06-29T18:08:21+02:00 (4893h21m6s ago)

  github.com/gokrazy/serial-busybox
    will include extra files in the root file system
      from /Users/andig/go/pkg/mod/github.com/gokrazy/serial-busybox@v0.0.0-20241231085202-bf9bc192356e/_gokrazy/extrafiles_arm64.tar
      last modified: 2024-12-31T09:20:09+01:00 (461h9m18s ago)

  github.com/gokrazy/wifi
    will include extra files in the root file system
      from /Users/andig/go/pkg/mod/github.com/gokrazy/wifi@v0.0.0-20240603195309-320785ba3e91/_gokrazy/extrafiles.tar
      last modified: 2024-06-03T21:35:27+02:00 (5513h54m0s ago)

Including loadable kernel modules from:
/Users/andig/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20250109091720-7437caa083f3/lib/modules
2025/01/19 14:29:27 open /Users/andig/htdocs/evcc/packaging/gokrazy/evcc/builddir/github.com/gokrazy/breakglass/breakglass.authorized_keys: no such file or directory
make: *** [gokrazy] Error 1
```

andig, 2025-01-19:
**UPDATE** the only difference I can see compared to a new instance created by `gok -i evcc new` is that that will create a json containing an absolute path:

            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "/Users/andig/gokrazy/evcc/breakglass.authorized_keys"
            }

It would be great (since the config.json is created from template) if relative paths would work- but relative to which root?

andig, 2025-01-19:
When I try to

    gok -i evcc new --parent_dir packaging/gokrazy/

this will create `packaging/gokrazy/evcc/config.json`:

```json
{
    "Hostname": "evcc",
    "Update": {
        "HTTPPassword": "WDRY9PDXEXXXCKGFgvzf"
    },
    "Packages": [
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/hello",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/breakglass"
    ],
    "PackageConfig": {
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "packaging/gokrazy/evcc/breakglass.authorized_keys"
            }
        },
        "github.com/gokrazy/gokrazy/cmd/randomd": {
            "ExtraFileContents": {
                "/etc/machine-id": "0d15c6aad89844ba892bda190d27ddcf\n"
            }
        }
    },
    "SerialConsole": "disabled"
}
```

However, this folder doesn't seem to work either:

    2025/01/19 14:56:12 ExtraFilePaths of github.com/gokrazy/breakglass: stat packaging/gokrazy/evcc/breakglass.authorized_keys: no such file or directory

stapelberg, 2025-01-20:
Hey @andig 

Seems like you found a bug, but I can’t quite reproduce it yet.

From your first message:

> 2025/01/19 14:29:27 open /Users/andig/htdocs/evcc/packaging/gokrazy/evcc/builddir/github.com/gokrazy/breakglass/breakglass.authorized_keys: no such file or directory

This means the code is still working in the build directory of breakglass when it should have switched back its working directory to the instance config. My first guess was that https://github.com/gokrazy/tools/blob/8edbabf66841e9f5539876b1a38e9cd9779e6d14/internal/packer/sbom.go#L55 contains a bug where it returns without calling `os.Chdir()`, but I don’t see how that could be the case just reading the code…

Can you verify if running `gok -i evcc sbom` also prints the same error? (I would like to track down where the error originates.)

For me, following your steps does not result in an error. Can you provide an archive (zip or tar) containing your `config.json` and `builddir` please? That should allow me to build exactly the same instance as you attempted.

> It would be great (since the config.json is created from template) if relative paths would work- but relative to which root?

Relative paths work in general, and they are relative to the instance directory (the same directory containing `config.json`).

From your last message:

> ```
> gok -i evcc new --parent_dir packaging/gokrazy/
> ```
> ---
> "ExtraFilePaths": {
>                 "/etc/breakglass.authorized_keys": "packaging/gokrazy/evcc/breakglass.authorized_keys"
>             }

This cannot work. You are specifying a path relative to the working directory in which you run `gok`, but your path needs to be relative to the instance directory, i.e. just `breakglass.authorized_keys` in this case.

andig, 2025-01-20:
> your path needs to be relative to the instance directory, i.e. just breakglass.authorized_keys in this case.

> Can you verify if running gok -i evcc sbom also prints the same error?

See below:

```
❯ pwd
/Users/andig/htdocs/evcc

❯ cat packaging/gokrazy/evcc/config.json
{
    "Hostname": "evcc",
    "Update": {
        "HTTPPort": "8080",
        "HTTPPassword": "HjJv6aBX51G4wT8EueCc"
    },
    "Packages": [
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/mkfs",
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/wifi",
        "github.com/evcc-io/evcc"
    ],
    "PackageConfig": {
        "github.com/evcc-io/evcc": {
            "GoBuildFlags": [
                "-tags=release -trimpath -ldflags='-X github.com/evcc-io/evcc/server.Version=0.132.1 -X github.com/evcc-io/evcc/server.Commit=3dba9ab26 -s -w'"
            ],
            "Environment": [
                "EVCC_NETWORK_PORT=80",
                "EVCC_DATABASE_DSN=/perm/evcc.db"
            ]
        },
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys",
                "-forward=private-network"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "breakglass.authorized_keys"
            }
        },
        "tailscale.com/cmd/tailscale": {
            "CommandLineFlags": [
                "up",
                "--ssh=true"
            ]
        }
    },
    "SerialConsole": "disabled",
    "InternalCompatibilityFlags": {}
}%                                                                                                                                                                

❯ gok -i evcc --parent_dir packaging/gokrazy sbom

2025/01/20 09:39:49 open /Users/andig/htdocs/evcc/packaging/gokrazy/evcc/builddir/github.com/gokrazy/breakglass/breakglass.authorized_keys: no such file or directory
```

> For me, following your steps does not result in an error. Can you provide an archive (zip or tar) containing your config.json and builddir please? That should allow me to build exactly the same instance as you attempted.

The current state is in https://github.com/evcc-io/evcc/pull/18209/files#diff-f429495d389eb99c3dcd00a87bdade799b4fdfcfc1beab73b5b8f7cba2b501da. If you run

    make gokrazy

that should you bring to the `overwrite` error:

```
Including loadable kernel modules from:
/Users/andig/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20250109091720-7437caa083f3/lib/modules
2025/01/20 09:43:46 open /Users/andig/htdocs/evcc/packaging/gokrazy/evcc/builddir/github.com/gokrazy/breakglass/breakglass.authorized_keys: no such file or directory

```

One final observation :`breakglass.authorized_keys` is currently an empty file- would that change anything?

stapelberg, 2025-01-22:
Thanks for the details, I have submitted two changes which should fix the issue you were seeing. Please update your `gok` tool and try again.

closed 2025-01-22 by stapelberg

andig, 2025-01-23:
Confirmed working, thank you!