gokrazy/gokrazy #291: Apptainer

ben-auo, 2025-01-20:
Similar to your Podman package, would it be possible to support Apptainer, a Linux Foundation container engine that's built in Go and [can be statically compiled](https://github.com/apptainer/apptainer/pull/2445)?

stapelberg, 2025-01-21:
Hey @ben-auo! 

I haven’t heard of Apptainer before.

The reason we package podman is that it gives us access to a rich ecosystem of pre-built (and maintained, often by upstream) containers available on the docker hub.

What additional use-cases would Apptainer make possible that podman doesn’t? Or, asked another way, what’s your motivation for asking for this piece of software?

ben-auo, 2025-01-23:
The single-file, cyryptographically signable [sif format](https://github.com/apptainer/sif) is ideal for sharing containers securely without the need for a third-party OCI hub. In addition, it makes A/B OTA upgrades trivial.

ben-auo, 2025-01-24:
In particular, my use case is embedded IoT devices. These tend to be hardware devices with minimal RAM and CPU resources so a Kubernetes cluster via something like Talos Linux would be overkill and commercial solutions like JFrog / qbee seem unnecessary. Ideally the solution would have a bare minimum OS that runs a container engine to allow for over-the-air upgrades. I looked at solutions like [Bare Minimal Linux](https://github.com/sapcc/bare-minimal-linux) but it's not very active, projects like [LinuxKit](https://github.com/linuxkit/linuxkit) and [void-mklive](https://github.com/void-linux/void-mklive) create images that are hundreds of megabytes in size, [BalenaOS](https://github.com/balena-io/open-balena) was buggy, [Mender](https://github.com/mendersoftware/mender)/[Yocto](https://www.yoctoproject.org/)/[buildroot](https://github.com/buildroot/buildroot) get way too technical with partitions & bootloaders, and Alpine Linux' musl base sometimes has [compatibility issues](https://docs.voidlinux.org/installation/musl.html#incompatible-software). 

Ideally the solution would be a distroless or [extremely small base OS](https://www.siderolabs.com/blog/there-are-only-12-binaries-in-talos-linux/) with just enough to run Apptainer and an executable that runs regularly to check a remote endpoint for an updated sif file to replace the existing (backing up the original for rollback if needed).

All that said, [Wolfi](https://github.com/wolfi-dev) and [Pantavisor](https://github.com/pantavisor/pantavisor) seem closest to GoKrazy so that might be an option if you feel this isn't a good fit.

ben-auo, 2025-02-01:
P.S. I was targeting amd64 Intel NUC devices but your rtr7/kernel isn't compatible so you can ignore this request. Thanks for considering it, though!

closed 2025-02-01 by ben-auo

stapelberg, 2025-02-07:
Thanks for the details, that helps me better understand where you’re coming from.

gokrazy was developed for (primarily) running Go appliances, with containers being available mostly as a prototyping / escape-hatch offering, whereas your goal seems to be to run containers, with no special interest in Go.

Of course, you could package apptainer (like we package podman) for gokrazy if you wanted to. But it would be up to you, and not something the gokrazy project officially supports.

> P.S. I was targeting amd64 Intel NUC devices but your rtr7/kernel isn't compatible so you can ignore this request. Thanks for considering it, though!

Out of curiosity, what about the rtr7/kernel kernel isn’t compatible? 

Note that https://github.com/gokrazy/kernel.amd64 is another gokrazy kernel that works on amd64 PCs. The rtr7/kernel is from my router7.org project, so it’s meant to be used for network routers, whereas gokrazy/kernel.amd64 is a general purpose kernel that should work for anything.

ben-auo, 2025-02-09:
Here's the result when I tried the kernel.amd64:

```
$ gok new
$ gok edit
# add this line >> "KernelPackage": "github.com/gokrazy/kernel.amd64",
$ GOARCH=amd64 gok overwrite --full /dev/sda
# Build complete!
```
See screenshot (displays, then bootloops)

[image]

stapelberg, 2025-02-09:
Thank you for the details. I would like to better understand the problem and ideally fix it :) Let’s continue this discussion over in https://github.com/gokrazy/kernel.amd64/issues/34 please